awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #157: Pipeline error in RTSP stream example

nyska, 2018-12-14:
I'm trying to stream from an RTSP camera source with the foloowing command:

`$ gst-launch-1.0 rtspsrc location=rtsp://YourCameraRtspUrl short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name=YourStreamName storage-size=512`

after, obviously, changing my stream name and RTSP URL and exporting the access credentials.

But this is the error I'm getting:

`Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ./kvs_log_configuration
INFO - Initializing curl.
2018-12-14 11:45:20 [139747905300224] INFO - 
createKinesisVideoClient(): Creating Kinesis Video Client
2018-12-14 11:45:20 [139747905300224] INFO - 
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
2018-12-14 11:45:20 [139747905300224] INFO - 
heapInitialize(): Creating AIV heap.
2018-12-14 11:45:20 [139747905300224] INFO - 
heapInitialize(): Heap is initialized OK
2018-12-14 11:45:20 [139747905300224] INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1544768120896627750 Expiration: 0
2018-12-14 11:45:20 [139747905300224] INFO - New credentials expiration is 1544770520
2018-12-14 11:45:20 [139747905300224] DEBUG - createDeviceHandler invoked
2018-12-14 11:45:20 [139747905300224] INFO - 
createDeviceResultEvent(): Create device result event.
2018-12-14 11:45:20 [139747905300224] DEBUG - clientReadyHandler invoked
2018-12-14 11:45:20 [139747905300224] DEBUG - Awaiting for the producer to become ready...
2018-12-14 11:45:20 [139747905300224] DEBUG - Kinesis Video producer is Ready.
2018-12-14 11:45:20 [139747905300224] INFO - try creating stream
2018-12-14 11:45:20 [139747905300224] INFO - Creating Kinesis Video Stream vnex
2018-12-14 11:45:20 [139747905300224] INFO - 
createKinesisVideoStream(): Creating Kinesis Video Stream.
2018-12-14 11:45:20 [139747905300224] DEBUG - describeStreamHandler invoked
2018-12-14 11:45:20 [139747905300224] DEBUG - Awaiting for the stream to become ready...
2018-12-14 11:45:20 [139747905300224] INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1544768120896828871 Expiration: 0
2018-12-14 11:45:20 [139747251971840] DEBUG - describeStream response: {"StreamInfo":{"CreationTime":1.544765892261E9,"DataRetentionInHours":24,"DeviceName":null,"KmsKeyId":"arn:aws:kms:us-west-2:354387791117:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:354387791117:stream/vnex/1544765892261","StreamName":"vnex","Version":"aXjbGB5w0xnX4BRghlGW"}}
2018-12-14 11:45:22 [139747251971840] INFO - Discovered existing Kinesis Video stream: arn:aws:kinesisvideo:us-west-2:354387791117:stream/vnex/1544765892261
2018-12-14 11:45:22 [139747251971840] INFO - stream arn in stream_info struct: arn:aws:kinesisvideo:us-west-2:354387791117:stream/vnex/1544765892261
2018-12-14 11:45:22 [139747251971840] INFO - 
describeStreamResultEvent(): Describe stream result event.
2018-12-14 11:45:22 [139747251971840] DEBUG - streamingEndpointHandler invoked
2018-12-14 11:45:22 [139747251971840] INFO - Refreshing credentials. Force refreshing: 0 Now time is: 1544768122242899328 Expiration: 0
2018-12-14 11:45:22 [139747243579136] DEBUG - getStreamingEndpoint response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
2018-12-14 11:45:23 [139747243579136] INFO - streaming to endpoint: https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com
2018-12-14 11:45:23 [139747243579136] INFO - 
getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2018-12-14 11:45:23 [139747243579136] DEBUG - streamingTokenHandler invoked
2018-12-14 11:45:23 [139747243579136] INFO - Refreshing credentials. Force refreshing: 1 Now time is: 1544768123532750987 Expiration: 1544770520
2018-12-14 11:45:23 [139747243579136] INFO - New credentials expiration is 1544770523
2018-12-14 11:45:23 [139747243579136] INFO - 
getStreamingTokenResultEvent(): Get streaming token result event.
2018-12-14 11:45:23 [139747243579136] DEBUG - streamReadyHandler invoked
2018-12-14 11:45:23 [139747243579136] DEBUG - Kinesis Video stream vnex is Ready.
2018-12-14 11:45:23 [139747905300224] Stream is ready
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
Progress: (connect) Connecting to rtsp://admin:admin@123@192.168.1.58:554/LiveMedia/ch1/Media1
ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not open resource for reading and writing.
Additional debug info:
gstrtspsrc.c(6775): gst_rtspsrc_retrieve_sdp (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
Failed to connect. (Generic error)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
`

Any help/advice would be appreciated. :)

chehefen, 2018-12-14:
your rtsp url looks weird. Can you double check it's playable on some other players like VLC?

nyska, 2018-12-14:
Yeah I checked. It's playable on vlc.

nyska, 2018-12-14:
Okay so I just tried it with a sample rtsp stream from [https://stackoverflow.com/questions/35067817/sample-rtsp-url-for-video-streaming](url) and it works.

I guess it's a problem with my rtsp url then. But it works in vlc. Any ideas why this is happening?

chehefen, 2018-12-17:
can you try remove the @ from your password?

unicornss, 2018-12-21:
Closing the issue for now as the issue has been answered.

closed 2018-12-21 by unicornss