awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #160: Curl: error setting certificate verify locations

davidgatti, 2018-12-17:
Since building the producer on a RPi Zero takes 12h. I decided to just move the content of the `lib` and `bin` folder to another RPi, and change the paths for the following env variables:

- LD_LIBRARY_PATH
- PATH
- GST_PLUGIN_PATH
- GST_SYSTEM_PLUGIN_PATH

When I did try to stream I got `error setting certificate verify locations`, because Curl is looking for the cert in a fixed location like so:

```
CAfile: /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/certs/cert.pem
```

a. Had no idea that you do that :)
b. can I specify the location of the cert with an env variable? 
c. I know you explain how to recompile Curl in this [readme](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md) file at the end of it, but I rather not do that if possible.

davidgatti, 2018-12-21:
Just a friendly ping :)

unicornss, 2018-12-21:
@davidgatti  sorry for the delayed response. 

curl requires the certs for sending streams to cloud. If your concern is build time, please note that in this case you are recompiling only the `libcurl` which should be faster.

On another note, are you able to add the certs in the /etc/ssl/ folder ? does that help to resolve the issue?
> Please download the PEM file from SFSRootCAG2.pem to /etc/ssl/cert.pem. If the file already exists /etc/ssl/cert.pem then you can append it by running sudo cat SFSRootCAG2.pem >> /etc/ssl/cert.pem.

Thanks

unicornss, 2018-12-28:
Closing the issue for now as the questions have been answered. Please reopen if required or create a new ticket.
Thanks

closed 2018-12-28 by unicornss