awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #162: If I do a revamp of your README will you accepted it?

davidgatti, 2018-12-20:
Hello Stream Masters!

I've been working with your producer for a while now and the biggest issues that I ran in to are due to the documentation. Based on my experience I know that there is room for improvement. 

Little bit about me:

- I did work as a Developer Relations Manger at two hardware companies.
- Now I'm the CEO of a software company called 0x4447.
- This is how i present my company GitHub account to the world https://github.com/0x4447.

Based on people feedback I know that the way I structure ideas and knowledge is appealing to people. And since I respect your work, and use it, I was wondering if you'd be OK me trying to revamp your READMEs to make them easier to understand. 

Best.

MushMal, 2018-12-20:
Hi @davidgatti. Thanks a lot for reaching out. We do accept pull requests! Thanks again!

closed 2018-12-20 by davidgatti

garishsood, 2018-12-20:
Hi David, 
My name is Girish and I am the product manager for Kinesis VIdeo Streams. We would love to learn more about your experience with the product and the challenges that you have faced with our documentation. Can you drop me a note at garisood@amazon.com and if you are open to the idea, I would like to schedule a brief call for us to discuss the improvements that we can make to our documentation?
Best,
-G