awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #163: "Failed to submit frame to Kinesis Video client. status: 0x32000002" on RTSP IP camera

aroca, 2018-12-27:
Hi guys, been checking out the related issues and ned a hand here. 
From reading the other posts i "think" the issue might be in how the camera encodes, but really not sure.

OS: MAC OS Sierra
CAM: UnionCam Q5

## Tests i did.

- Run the sample_app (embedded mac camera)
./kinesis_video_gstreamer_sample_app -w 640 -h 480 stream_name

This works perfectly. I can see the Preview in the AWS console KVideo. (so i assume no networking issues)

- Run the sample_rtsp
./kinesis_video_gstreamer_sample_rtsp_app rtsp://user:password@192.168.100.15 same_stream
That rtsp URL works fine; tested it from VLC and streams perfect to my Mac.

Here is where i got the errors. And the console Preview shows the no fragments error...

I chedked the create_kinesis_video_frame and modified it to 20 as it was in 10 just to test and same error.

Its worth mentioning that i have another error while trying the RTSP client app that is random:
`[INFO ][2018-12-27 13:58:20] Initializing curl.
[INFO ][2018-12-27 13:58:20]
createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2018-12-27 13:58:20]
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
[INFO ][2018-12-27 13:58:20]
heapInitialize(): Creating AIV heap.
[INFO ][2018-12-27 13:58:20]
heapInitialize(): Heap is initialized OK
[INFO ][2018-12-27 13:58:20] Refreshing credentials. Force refreshing: 0 Now time is: 1545919100 Expiration: 0
[INFO ][2018-12-27 13:58:20] New credentials expiration is 1545921500
[DEBUG][2018-12-27 13:58:20] createDeviceHandler invoked
[INFO ][2018-12-27 13:58:20]
createDeviceResultEvent(): Create device result event.
[DEBUG][2018-12-27 13:58:20] clientReadyHandler invoked
[DEBUG][2018-12-27 13:58:20] Awaiting for the producer to become ready...
[DEBUG][2018-12-27 13:58:20] Kinesis Video producer is Ready.
[DEBUG][2018-12-27 13:58:20] Client is ready
[INFO ][2018-12-27 13:58:20] Creating Kinesis Video Stream kvideo-customer-1
[INFO ][2018-12-27 13:58:20]
createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2018-12-27 13:58:20] describeStreamHandler invoked
[DEBUG][2018-12-27 13:58:20] Awaiting for the stream to become ready...
[INFO ][2018-12-27 13:58:20] Refreshing credentials. Force refreshing: 0 Now time is: 1545919100 Expiration: 0
[DEBUG][2018-12-27 13:58:21] describeStream response: {"StreamInfo":{"CreationTime":1.545876464939E9,"DataRetentionInHours":2,"DeviceName":null,"KmsKeyId":"arn:aws:kms:us-east-1:******:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:******:stream/kvideo-customer-1/1545876464939","StreamName":"kvideo-customer-1","Version":"msEYvwcXski1Bu2iN6tI"}}
[INFO ][2018-12-27 13:58:21] Discovered existing Kinesis Video stream: arn:aws:kinesisvideo:us-east-1:******:stream/kvideo-customer-1/1545876464939
[INFO ][2018-12-27 13:58:21] stream arn in stream_info struct: arn:aws:kinesisvideo:us-east-1:******:stream/kvideo-customer-1/1545876464939
[INFO ][2018-12-27 13:58:21]
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2018-12-27 13:58:21] streamingEndpointHandler invoked
[INFO ][2018-12-27 13:58:21] Refreshing credentials. Force refreshing: 0 Now time is: 1545919101 Expiration: 0
[DEBUG][2018-12-27 13:58:21] getStreamingEndpoint response: {"DataEndpoint":"https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com"}
[INFO ][2018-12-27 13:58:21] streaming to endpoint: https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com
[INFO ][2018-12-27 13:58:21]
getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2018-12-27 13:58:21] streamingTokenHandler invoked
[INFO ][2018-12-27 13:58:21] Refreshing credentials. Force refreshing: 1 Now time is: 1545919101 Expiration: 1545921500
[INFO ][2018-12-27 13:58:21] New credentials expiration is 1545921501
[INFO ][2018-12-27 13:58:21]
getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2018-12-27 13:58:21] streamReadyHandler invoked
[DEBUG][2018-12-27 13:58:21] Kinesis Video stream kvideo-customer-1 is Ready.
[DEBUG][2018-12-27 13:58:21] Stream is ready
[DEBUG][2018-12-27 13:58:21] Pipeline OK
[DEBUG][2018-12-27 13:58:21] rtsp://user:password@192.168.100.15
[DEBUG][2018-12-27 13:58:21] appsink configured
[DEBUG][2018-12-27 13:58:21] Pipeline built, elements *NOT* yet linked
[DEBUG][2018-12-27 13:58:21] Pipeline playing
Error received from element source: Could not read from resource.
Debugging information: gstrtspsrc.c(5464): gst_rtspsrc_try_send (): /GstPipeline:rtsp-kinesis-pipeline/GstRTSPSrc:source:
Could not receive message. (Received end-of-file)
[INFO ][2018-12-27 13:58:27] Freeing Kinesis Video Stream kvideo-customer-1
[INFO ][2018-12-27 13:58:27]
freeKinesisVideoStream(): Freeing Kinesis Video stream.
[INFO ][2018-12-27 13:58:27] Freeing Kinesis Video Stream kvideo-customer-1
[INFO ][2018-12-27 13:58:27]
freeKinesisVideoClient(): Freeing Kinesis Video Client
[INFO ][2018-12-27 13:58:27]
commonHeapDebugCheckAllocator(): Heap is initialized
[INFO ][2018-12-27 13:58:27]
commonHeapDebugCheckAllocator(): Heap limit:         536870912
[INFO ][2018-12-27 13:58:27]
commonHeapDebugCheckAllocator(): Heap size:         0
[INFO ][2018-12-27 13:58:27]
commonHeapDebugCheckAllocator(): Number of allocations:         0
[INFO ][2018-12-27 13:58:27]
aivHeapDebugCheckAllocator(): Allocated blocks pointer:         0x0
[INFO ][2018-12-27 13:58:27]
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2018-12-27 13:58:27]
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2018-12-27 13:58:27]
aivHeapDebugCheckAllocator(): Free blocks pointer:         0x112b83000
[INFO ][2018-12-27 13:58:27]
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2018-12-27 13:58:27]
aivHeapDebugCheckAllocator(): Block:  0x112b83000    size:  536870872
[INFO ][2018-12-27 13:58:27]
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2018-12-27 13:58:27]
heapRelease(): Freeing native heap.`

I will be uploading the logs when it DOES starts and shows the error in the subject in a few minutes.

aroca, 2018-12-27:
Ok. seems like an improvement. I updated and rebuild and now i see this error in the console.

```
Fragment decoding error
There was an error decoding the video data. Verify that the stream contains valid H.264 content.
RetryHide details
Codec ID: avc1.64002a (Profile: High, Level 4.2)

Fragment number: 91343852333182011748620448174858511767407398063

Producer timestamp: December 27, 2018 12:09 PM -03

Server timestamp: December 27, 2018 12:09 PM -03

Size: 527.1 KB

This may be an intermittent issue. Retry the same action, or try refreshing your browser tab.
```

unicornss, 2018-12-28:
Hi @aroca  

`avc1.64002a (Profile: High, Level 4.2)`  the playback behaviour suggests[ B frames](https://en.wikipedia.org/wiki/Video_compression_picture_types) in the H264 stream.  Is it possible for you to set the Encoding to be Baseline profile in the camera and try?

Thanks

aroca, 2018-12-28:
Hi @unicornss ! Thanks for the answer.. I dont know if i can change the Encoding in the IP Camera. Not really well documented, nor a decent app. Let me check.

chehefen, 2018-12-28:
Hi, which version of the sdk are you using?

unicornss, 2018-12-28:
Hi @aroca  

Unfortunately we don't have the same model to reproduce the issue locally. However, there are additional details on how the playback from the Kinesis Media Streams works for your reference:

HLS Documentation for Kinesis Video Streams:
https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-hls.html 

Simple player sample:
https://github.com/aws-samples/amazon-kinesis-video-streams-hls-viewer/blob/master/index.html

Thanks

aroca, 2018-12-28:
> Hi, which version of the sdk are you using?

I just pulled from master yesterday.. as i put in the comment.. Previous version form some months ago could not even show the Preview. Some definitely some improvemnts :)

Will try to check when i am at home to see if i can see the Encoding. Will read those. but way above me right now.. hehe

Thanks

chehefen, 2018-12-28:
I just read your debug_sdk_rtsp.log and I saw lots of MAX_FRAGMENT_DURATION_REACHED error. Can you try go to your camera's configuration page and in the video settings look for something like "GOP size" or "I frame interval" and change the value to something lower or equal to your frame rate.

aroca, 2018-12-29:
hey @chehefen ! Unfortunately the soft and web for the camera really suck. No settings for video at all! :( i am googling to see if i find some hidden url or something, but no luck yet.
Any other thing i can try from the sample app or other, just let me know please.
Thanks guys!

aroca, 2018-12-29:
Guys, i am just trying to use some Android app from my cell, that seems to be more configurable. Do you guys have some app in mind to recommend? I am installing IP Webcam and RTSP Camera Server.
First tests, give me same errors, like dropping a lot of packets. Will continue to test. Thanks

aroca, 2018-12-31:
Both Apps have issues with RTSP. will try to play a bit with configuration. but basically the same drop fragments. Cheers! and have a Happy New Year's Eve!

MushMal, 2019-07-24:
Resolving this for now.

closed 2019-07-24 by MushMal