awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #164: Width and hight parameters failing

dave-malone, 2018-12-28:
After successfully building the latest version of the kinesis_video_gstreamer_sample_app, I am able to stream to a KVS endpoint as long as I do not pass any other parameters to the sample app. As soon as I try to pass any other parameters to the app, I get the following error:

```
pi@raspberrypi:~/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build $ ./kinesis_video_gstreamer_sample_app myteststream -w 640 -h 480 -f 10 -b 128
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::substr: __pos (which is 4294967295) > this->size() (which is 2)
Aborted
```

Raspberry Pi configuration is as follows:
* Raspberry Pi Zero W
* Linux raspberrypi 4.14.79+ #1159 Sun Nov 4 17:28:08 GMT 2018 armv6l GNU/Linux
* NoIR Camera V2

chehefen, 2018-12-28:
Hi, can you checkout the latest code and see if the issue persists? thanks

unicornss, 2019-04-02:
Closing the issue for now as the  question has been answered.
Please re-open or create a new one if additional support is required.

Thanks

closed 2019-04-02 by unicornss