awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #167: GstRTSPSrc Unauthorized

francoisdutoit, 2019-01-20:
Hi - I last worked with Kinesis about a year agao and maged to stream from my IP Camera - that was from ubuntu. I now did a fresh build on Raspberry and getting an error when trying to stream using gst-launch. ( I tested the rtsp url from another program and it works fine ) 

Below are my steps and the console output.

1.) cd:/home/pi/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp-master/kinesis-video-native-build/
2.) export AWS_ACCESS_KEY_ID='MYKEYID'
3.) export AWS_SECRET_ACCESS_KEY='MYSECRET'
4.) ./set_kvs_sdk_env.sh
5.) export PATH=/home/pi/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp-master/kinesis-video-native-build/downloads/local/bin:$PATH
6.) export GST_PLUGIN_PATH=/home/pi/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp-master/kinesis-video-native-build/downloads/local/lib:$GST_PLUGIN_PATH

pi@raspberrypi:~/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp-master/kinesis-video-native-build $ gst-launch-1.0 rtspsrc location=rtsp://192.168.88.108:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name=MyOnvifStream1 storage-size=512
[1] 14270
[2] 14271
[3] 14272
bash: short-header=TRUE: command not found
[2]-  Done                    subtype=0
[3]+  Done                    unicast=true
pi@raspberrypi:~/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp-master/kinesis-video-native-build $ Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
Progress: (connect) Connecting to rtsp://192.168.88.108:554/cam/realmonitor?channel=1
Progress: (open) Retrieving server options
ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Unauthorized
Additional debug info:
gstrtspsrc.c(5580): gst_rtspsrc_send (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
Unauthorized (401)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

chehefen, 2019-01-22:
did you check if your rtsp camera requires username and password to view the stream?

francoisdutoit, 2019-01-22:
Yes I have. I have streamed to aws previously using this camera. But with the new build I'm having this issue...

chehefen, 2019-01-22:
can you try wrap the url in double quotes? like this: `gst-launch-1.0 rtspsrc location="rtsp://192.168.88.108:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif" short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name=MyOnvifStream1 storage-size=512`

francoisdutoit, 2019-01-24:
That worked. Thanks 👍

closed 2019-01-24 by francoisdutoit