awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #168: dll error

juanlu78, 2019-01-25:
I´ve installed the Msys2 to install this client and when I try to run the gst-launch-1.0  I get two dll errors (libgstfdkaac.dll, and libgstopencv.dll) and the message is "cannot find specified module" but the path that shows is the correct one and the file is in there.

Is there any way to fix it.
I triiied with pacman -Syu and pacman -Su and everything is up to date
Can you help me with this?? 
Thanks
Juan Luis

chehefen, 2019-01-28:
can you show us your gst-launch-1.0 command? Was it able to run despite the two error messages?

MushMal, 2019-04-23:
@juanlu78 I am resolving this issue for now. Please re-open if the build issue persists with the GStreamer.

closed 2019-04-23 by MushMal