awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #169: gstkvssink.h:43:10: fatal error: 'gst/gst.h' file not found

arasub, 2019-01-27:
Below with the error, while running install-script

mac$ ./install-script 
 ===========================Starting to build ===========================
              Kinesis Video Stream Producer SDK and Demo Application
       ========================================================================
       Following open source libraries and tools will be downloaded and built
       before building the producer sdk.
       ========================================================================
       # ------------------------- BUILD-TOOLS -------------------------------#
        autoconf   automake
        bison      curl
        flex       gettext
        googletest iconv
        libffi     libpcre
        libtools   log4cplus
        m4         openssl
        zlib

        # ------------------------- GSTREAMER - for demo application --------#
        GStreamer, GStreamer plugins, yasm, gst-libav and x264.
        ------------------------------------------------------------------------
        *** NOTE *** for MacOS High Sierra: Use
            brew install autoconf
            brew install bison before running this script
        ------------------------------------------------------------------------
        ========================================================================
        This install script downloads and builds the dependent libraries from
        source in the ./kinesis-video-native-build/downloads directory only.
        ========================================================================
 

      Y - Confirm to proceed with Producer SDK build
      N - Exit Producer SDK build
Enter selection: Y

Include building GStreamer extra plugins= Y
Additional curl options=
cmake version 3.11.4 CMake suite maintained and supported by Kitware (kitware.com/cmake).
Checking openssl at /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libssl.a
Checking curl at /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libcurl.a
Checking log4cplus at /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/liblog4cplus.dylib/.so
Checking gettext at /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libgettextlib.dylib/.so
Checking libffi at /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libffi.a
Checking pcre at /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libpcre.dylib/.so
Checking zlib at /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libz.dylib/.so
Checking iconv at /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libiconv.dylib/.so
Checking glib at /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libglib-2.0.dylib/.so
Checking googletest at /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libgtest.a
Checking gstreamer at /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libgstreamer-1.0.dylib/.so
Checking gst-plugin-base at /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libgstvideo-1.0.dylib/.so
Checking gst-plugin-good at /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/gstreamer-1.0/libgstavi.la
Checking gst-plugin-bad at /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libgstbadvideo-1.0.dylib/.so
Checking for yasm
Checking gst-libav in /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/
Checking x264 at /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libx264.dylib/.so
Checking gst-plugin-ugly at /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/gstreamer-1.0/libgstx264.dylib/.so
-- System name is Darwin
-- Kinesis Video PIC path is ../kinesis-video-pic
-- Kinesis Video Producer path is ../kinesis-video-producer
-- Kinesis Video JNI path is ../kinesis-video-producer-jni
-- Kinesis Video Gstreamer demp app path is ../kinesis-video-gst-demo
-- Kinesis Video Open Source path is /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads
-- Configuration Unix Makefiles
-- Bitness 64 bits
-- exe libraries Threads::Threads
-- Configuring done
-- Generating done
-- Build files have been written to: /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build
[  1%] Building CXX object CMakeFiles/gstkvsplugintest.dir/opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/plugin-src/gstkvstest.cpp.o
[  1%] Building CXX object CMakeFiles/producer.dir/opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/AwsV4Signer.cpp.o
In file included from /opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/plugin-src/gstkvstest.cpp:1:
/opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/plugin-src/gstkvssink.h:43:10: fatal error: 'gst/gst.h' file not found
#include <gst/gst.h>
         ^~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/gstkvsplugintest.dir/opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/plugin-src/gstkvstest.cpp.o] Error 1
make[1]: *** [CMakeFiles/gstkvsplugintest.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  2%] Building CXX object CMakeFiles/producer.dir/opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/OpenSSLThreadCallbacks.cpp.o
/opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/AwsV4Signer.cpp:4:10: fatal error: 'openssl/hmac.h' file not found
#include <openssl/hmac.h>
         ^~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/producer.dir/opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/AwsV4Signer.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/OpenSSLThreadCallbacks.cpp:3:10: fatal error: 'openssl/crypto.h' file not found
#include <openssl/crypto.h>
         ^~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/producer.dir/opt/aws_demos/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/OpenSSLThreadCallbacks.cpp.o] Error 1
make[1]: *** [CMakeFiles/producer.dir/all] Error 2
make: *** [all] Error 2

MushMal, 2019-04-23:
@arasub Any update on this issue?

unicornss, 2019-05-13:
Closing the issue as the question has been answered.

closed 2019-07-24 by MushMal

rraghav476, 2020-07-23:
on ubuntu i am also not find found gst/gst.h file 
it return fatal error: 'gst/gst.h' file not found

lherman-cs, 2020-07-23:
@rraghav476 did you follow the install steps? You need to install the following:

```sh
sudo apt-get install libssl-dev libcurl4-openssl-dev liblog4cplus-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-base-apps gstreamer1.0-plugins-bad gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly gstreamer1.0-tools
```