awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #170: no fragments found in kinesis video stream

sujithaperumal, 2019-02-06:
I tried streaming video from my laptop webcam to KVS using MinGW(c++ producer SDK). But it shows no fragments found.  I have a lenovo ideapad y700 laptop. When run in MinGW, the following log was found.

(gst-launch-1.0:12932): GStreamer-WARNING **: 01:10:21.774: Failed to load plugin 'C:\msys64\mingw64\lib\gstreamer-1.0\libgstx265.dll': 'C:\msys64\mingw64\lib\gstreamer-1.0\libgstx265.dll': The specified procedure could not be found.
Setting pipeline to PAUSED ...
[INFO ][2019-02-06 19:40:21] Initializing curl.
[INFO ][2019-02-06 19:40:21]
createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2019-02-06 19:40:21]
heapInitialize(): Initializing native heap with limit size 10485760, spill ratio 0% and flags 0x00000001
[INFO ][2019-02-06 19:40:21]
heapInitialize(): Creating AIV heap.
[INFO ][2019-02-06 19:40:21]
heapInitialize(): Heap is initialized OK
[INFO ][2019-02-06 19:40:21] Refreshing credentials. Force refreshing: 0 Now time is: 1549482021798189800 Expiration: 0
[INFO ][2019-02-06 19:40:21] New credentials expiration is 1549484421
[DEBUG][2019-02-06 19:40:21] createDeviceHandler invoked
[INFO ][2019-02-06 19:40:21]
createDeviceResultEvent(): Create device result event.
[DEBUG][2019-02-06 19:40:21] clientReadyHandler invoked
[DEBUG][2019-02-06 19:40:21] Awaiting for the producer to become ready...
[DEBUG][2019-02-06 19:40:21] Kinesis Video producer is Ready.
[INFO ][2019-02-06 19:40:21] try creating stream
[INFO ][2019-02-06 19:40:21] Creating Kinesis Video Stream ExampleStream
[INFO ][2019-02-06 19:40:21]
createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2019-02-06 19:40:21] describeStreamHandler invoked
[DEBUG][2019-02-06 19:40:21] Awaiting for the stream to become ready...
[INFO ][2019-02-06 19:40:21] Refreshing credentials. Force refreshing: 0 Now time is: 1549482021799832400 Expiration: 0
[DEBUG][2019-02-06 19:40:23] describeStream response: {"StreamInfo":{"CreationTime":1.549478491599E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-west-2:073686542274:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:073686542274:stream/ExampleStream/1549478491599","StreamName":"ExampleStream","Version":"vv3HSwA86BBEqZZyCimR"}}
[INFO ][2019-02-06 19:40:23] Discovered existing Kinesis Video stream: arn:aws:kinesisvideo:us-west-2:073686542274:stream/ExampleStream/1549478491599
[INFO ][2019-02-06 19:40:23] stream arn in stream_info struct: arn:aws:kinesisvideo:us-west-2:073686542274:stream/ExampleStream/1549478491599
[INFO ][2019-02-06 19:40:23]
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2019-02-06 19:40:23] streamingEndpointHandler invoked
[INFO ][2019-02-06 19:40:23] Refreshing credentials. Force refreshing: 0 Now time is: 1549482023699084600 Expiration: 0
[DEBUG][2019-02-06 19:40:25] getStreamingEndpoint response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ][2019-02-06 19:40:25] streaming to endpoint: https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com
[INFO ][2019-02-06 19:40:25]
getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2019-02-06 19:40:25] streamingTokenHandler invoked
[INFO ][2019-02-06 19:40:25] Refreshing credentials. Force refreshing: 1 Now time is: 1549482025045338000 Expiration: 1549484421
[INFO ][2019-02-06 19:40:25] New credentials expiration is 1549484425
[INFO ][2019-02-06 19:40:25]
getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2019-02-06 19:40:25] streamReadyHandler invoked
[DEBUG][2019-02-06 19:40:25] Kinesis Video stream ExampleStream is Ready.
Stream is ready
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Redistribute latency...
[INFO ][2019-02-06 19:40:28] try starting stream. Stream handle: 53458096
[INFO ][2019-02-06 19:40:28]
kinesisVideoStreamFormatChanged(): Stream format changed.
[DEBUG][2019-02-06 19:40:28] putStreamHandler invoked
[DEBUG][2019-02-06 19:40:28] Created a new PutMedia request with stream upload handle: 0
[INFO ][2019-02-06 19:40:28]
putStreamResultEvent(): Put stream result event.
[INFO ][2019-02-06 19:40:28]
getKinesisVideoStreamMetrics(): Get stream metrics for Stream 00000000032fb4b0.
[INFO ][2019-02-06 19:40:28]
getKinesisVideoMetrics(): Get the memory metrics size.
[INFO ][2019-02-06 19:40:28] Creating new connection for Kinesis Video stream: ExampleStream
[INFO ][2019-02-06 19:40:28] Refreshing credentials. Force refreshing: 0 Now time is: 1549482028871138300 Expiration: 0
[DEBUG][2019-02-06 19:40:28] Kinesis Video client and stream metrics
        >> Overall storage byte size: 10485760
        >> Available storage byte size: 10478760
        >> Allocated storage byte size: 7000
        >> Total view allocation byte size: 144072
        >> Total streams frame rate (fps): 25
[... 139 lines cut ...]
[DEBUG][2019-02-06 19:40:39] Wrote 10363 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-02-06 19:40:39] Pausing CURL read for upload handle: 0
[INFO ][2019-02-06 19:40:39]
getKinesisVideoStreamMetrics(): Get stream metrics for Stream 00000000032fb4b0.
[INFO ][2019-02-06 19:40:39]
getKinesisVideoMetrics(): Get the memory metrics size.
[DEBUG][2019-02-06 19:40:39] Kinesis Video client and stream metrics
        >> Overall storage byte size: 10485760
        >> Available storage byte size: 10426526
        >> Allocated storage byte size: 59234
        >> Total view allocation byte size: 144072
        >> Total streams frame rate (fps): 15
        >> Total streams transfer rate (bps): 19165400 (18716 Kbps)
        >> Current view duration (ms): 530
        >> Overall view duration (ms): 2994
        >> Current view byte size: 18799
        >> Overall view byte size: 57348
        >> Current frame rate (fps): 15.2386
        >> Current transfer rate (bps): 19165400 (18716 Kbps)
[DEBUG][2019-02-06 19:40:40] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-02-06 19:40:40] Wrote 7160 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-02-06 19:40:40] Pausing CURL read for upload handle: 0
[INFO ][2019-02-06 19:40:40] Curl post body write function for stream: ExampleStream and upload handle: 0 returned: {"EventType":"RECEIVED","FragmentTimecode":8111,"FragmentNumber":"91343852333181447262817814513832989291304493906"}

I don't know what is happening here. Can anyone please help me? Thanks in advance!

zhiyua-git, 2019-02-06:
Can you check if you are trying to view the video in us-west-2(Oregon) region?

Thanks,
-ZH

sujithaperumal, 2019-02-06:
@zhiyua-git I did just now. It works now. Thanks!

closed 2019-02-06 by zhiyua-git