awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #176: openssl install issue in windows using mvsc

sujithaperumal, 2019-02-15:
when trying to run install-script.bat batch file, in the openssl step, i get a dialog box as follows:
[image]
If I ignore this and continue installing, I get this below error in the building producer sdk step.
1>c:\users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer\src\request.h(7): fatal erro
       r C1083: Cannot open include file: 'curl/curl.h': No such file or directory [C:\Users\psidd\amazon-kinesis-video
       -streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]
         KinesisVideoStream.cpp
         You are not using DLL C run time library. You must call log4cplus::initialize() once before you use any other
         log4cplus API.
     1>c:\users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer\src\request.h(7): fatal erro
       r C1083: Cannot open include file: 'curl/curl.h': No such file or directory [C:\Users\psidd\amazon-kinesis-video
       -streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]
         Auth.cpp
     1>c:\users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer\src\request.h(7): fatal erro
       r C1083: Cannot open include file: 'curl/curl.h': No such file or directory [C:\Users\psidd\amazon-kinesis-video
       -streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]
         AwsV4Signer.cpp
     1>c:\users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer\src\request.h(7): fatal erro
       r C1083: Cannot open include file: 'curl/curl.h': No such file or directory [C:\Users\psidd\amazon-kinesis-video
       -streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]
         CurlCallManager.cpp
     1>c:\users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer\src\request.h(7): fatal erro
       r C1083: Cannot open include file: 'curl/curl.h': No such file or directory [C:\Users\psidd\amazon-kinesis-video
       -streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]
         DefaultCallbackProvider.cpp
     1>c:\users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer\src\request.h(7): fatal erro
       r C1083: Cannot open include file: 'curl/curl.h': No such file or directory [C:\Users\psidd\amazon-kinesis-video
       -streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]
         OngoingStreamState.cpp
     1>c:\users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer\src\ongoingstreamstate.h(10)
       : fatal error C1083: Cannot open include file: 'curl/curl.h': No such file or directory [C:\Users\psidd\amazon-k
       inesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]
         CachingEndpointOnlyCallbackProvider.cpp
     1>c:\users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer\src\request.h(7): fatal erro
       r C1083: Cannot open include file: 'curl/curl.h': No such file or directory [C:\Users\psidd\amazon-kinesis-video
       -streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]
         OpenSSLThreadCallbacks.cpp
     1>c:\users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer\src\opensslthreadcallbacks.c
       pp(3): fatal error C1083: Cannot open include file: 'openssl/crypto.h': No such file or directory [C:\Users\psid
       d\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]
         Request.cpp
     1>c:\users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer\src\request.h(7): fatal erro
       r C1083: Cannot open include file: 'curl/curl.h': No such file or directory [C:\Users\psidd\amazon-kinesis-video
       -streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]
         Response.cpp
     1>c:\users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer\src\response.h(4): fatal err
       or C1083: Cannot open include file: 'curl/curl.h': No such file or directory [C:\Users\psidd\amazon-kinesis-vide
       o-streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]
         IotCertCredentialProvider.cpp
     1>c:\users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer\src\request.h(7): fatal erro
       r C1083: Cannot open include file: 'curl/curl.h': No such file or directory [C:\Users\psidd\amazon-kinesis-video
       -streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]
         RotatingCredentialProvider.cpp
     1>c:\users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer\src\request.h(7): fatal erro
       r C1083: Cannot open include file: 'curl/curl.h': No such file or directory [C:\Users\psidd\amazon-kinesis-video
       -streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]
         RotatingStaticCredentialProvider.cpp
     1>c:\users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer\src\request.h(7): fatal erro
       r C1083: Cannot open include file: 'curl/curl.h': No such file or directory [C:\Users\psidd\amazon-kinesis-video
       -streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]
         PutFrameHelper.cpp
     1>c:\users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer\src\request.h(7): fatal erro
       r C1083: Cannot open include file: 'curl/curl.h': No such file or directory [C:\Users\psidd\amazon-kinesis-video
       -streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]
         Generating Code...
     1>Done Building Project "C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\p
       roducer.vcxproj" (default targets) -- FAILED.

Build FAILED.

[... 37 lines cut ...]
       ror C1083: Cannot open include file: 'curl/curl.h': No such file or directory [C:\Users\psidd\amazon-kinesis-vid
       eo-streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]
         c:\users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer\src\request.h(7): fatal er
       ror C1083: Cannot open include file: 'curl/curl.h': No such file or directory [C:\Users\psidd\amazon-kinesis-vid
       eo-streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]
         c:\users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer\src\request.h(7): fatal er
       ror C1083: Cannot open include file: 'curl/curl.h': No such file or directory [C:\Users\psidd\amazon-kinesis-vid
       eo-streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]
         c:\users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer\src\request.h(7): fatal er
       ror C1083: Cannot open include file: 'curl/curl.h': No such file or directory [C:\Users\psidd\amazon-kinesis-vid
       eo-streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxproj]

    0 Warning(s)
    15 Error(s)

Time Elapsed 00:00:08.33

C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>if ERRORLEVEL 1 goto :showerror

C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>cd "C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\..\"

C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>echo Build error occurred! Please delete related directories in downloads\ to start a clean build.
Build error occurred! Please delete related directories in downloads\ to start a clean build.
 
Any help would be appreciated!Thanks much!

zhiyua-git, 2019-02-15:
Hi @sujithaperumal ,

What is your Windows version and bitwise (32 bit or 64 bit)? Which install instruction are you following?

Thanks,
-Zh

sujithaperumal, 2019-02-15:
Hi @zhiyua-git 
I have a windows 10 version and it is 64-bit. I'm following the install instructions for windows using microsoft visual c++

zhiyua-git, 2019-02-15:
Hi @sujithaperumal ,

Can you try to download this msi and see if it can be run on your Win10? http://strawberryperl.com/download/5.28.1.1/strawberry-perl-5.28.1.1-64bit.msi 

This is the msi for perl needed for building openssl on windows. Seems you had issue when install script try to do this.

Thanks,
-ZH

sujithaperumal, 2019-02-15:
@zhiyua-git The site isn't loading!

zhiyua-git, 2019-02-15:
@sujithaperumal  What about http://strawberryperl.com ?

sujithaperumal, 2019-02-15:
@zhiyua-git No. The site isn't loading!

zhiyua-git, 2019-02-15:
@sujithaperumal What about https://www.activestate.com/products/activeperl/ ?

sujithaperumal, 2019-02-15:
@zhiyua-git  Yup! That works. Thanks!

zhiyua-git, 2019-02-15:
If that ActivePerl works for you, make sure you add the perl to your PATH environment variable. 

Delete your kinesis-video-native-build/downloads directory.
Open CMD with administrator permission.
Run vs-buildtools-install.bat.
Download [windows-install-script-with-perl.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/2870412/windows-install-script-with-perl.txt)
 and Rename it to windows-install-script-with-perl.bat.
Run windows-install-script-with-perl.bat64.

@sujithaperumal Let me know if these steps work for you.

sujithaperumal, 2019-02-15:
@zhiyua-git Thanks! It works. But why do I get this below error?

C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>if ERRORLEVEL 1 goto :showerror

C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>msbuild kinesis_video_gstreamer_sample_rtsp_app.vcxproj  /p:Configuration=Release /p:Platform=x64 /m
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: kinesis_video_gstreamer_sample_rtsp_app.vcxproj

C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>if ERRORLEVEL 1 goto :showerror

C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>cd "C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\..\"

C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>echo Build error occurred! Please delete related directories in downloads\ to start a clean build.
Build error occurred! Please delete related directories in downloads\ to start a clean build.

zhiyua-git, 2019-02-15:
@sujithaperumal Can you put a complete log here? It is not clear what is happening according to this. Thanks.

sujithaperumal, 2019-02-15:
C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads>echo "Building Producer SDK"
"Building Producer SDK"

C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads>cd "C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads"

C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads>cd ..

C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>IF 64 == 32 (cmake -G "Visual Studio 15 2017" )  ELSE (cmake -G "Visual Studio 15 2017 Win64" )
-- The C compiler identification is MSVC 19.16.27027.1
-- The CXX compiler identification is MSVC 19.16.27027.1
-- Check for working C compiler: C:/BuildTools/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe
-- Check for working C compiler: C:/BuildTools/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/BuildTools/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe
-- Check for working CXX compiler: C:/BuildTools/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- System name is Windows
-- Kinesis Video PIC path is ../kinesis-video-pic
-- Kinesis Video Producer path is ../kinesis-video-producer
-- Kinesis Video JNI path is ../kinesis-video-producer-jni
-- Kinesis Video Gstreamer demp app path is ../kinesis-video-gst-demo
-- Kinesis Video Open Source path is C:/Users/psidd/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads
-- Configuration Visual Studio 15 2017 Win64
-- Bitness 64 bits
-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE
-- exe libraries Threads::Threads
-- Using static CRT
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/psidd/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build

C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>msbuild producer.vcxproj  /p:Configuration=Release /p:Platform=x64 /m
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

Build started 2/16/2019 2:00:25 AM.
     1>Project "C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxpro
       j" on node 1 (default targets).
     1>Project "C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\producer.vcxpro
       j" (1) is building "C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\ZERO
       _CHECK.vcxproj" (2) on node 1 (default targets).
     2>PrepareForBuild:
         Creating directory "x64\Release\ZERO_CHECK\".
         Creating directory "x64\Release\ZERO_CHECK\ZERO_CHECK.tlog\".
       InitializeBuildStatus:
         Creating "x64\Release\ZERO_CHECK\ZERO_CHECK.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
       CustomBuild:
         Checking Build System
         CMake does not need to re-run because C:/Users/psidd/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-vid
         eo-native-build/CMakeFiles/generate.stamp is up-to-date.
       FinalizeBuildStatus:
         Deleting file "x64\Release\ZERO_CHECK\ZERO_CHECK.tlog\unsuccessfulbuild".
         Touching "x64\Release\ZERO_CHECK\ZERO_CHECK.tlog\ZERO_CHECK.lastbuildstate".
     2>Done Building Project "C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\Z
       ERO_CHECK.vcxproj" (default targets).
     1>PrepareForBuild:
         Creating directory "producer.dir\Release\".
         Creating directory "C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\Re
         lease\".
         Creating directory "producer.dir\Release\producer.tlog\".
       InitializeBuildStatus:
         Creating "producer.dir\Release\producer.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
[... 890 lines cut ...]
       mple_app.cpp(1097): warning C4101: 'err': unreferenced local variable [C:\Users\psidd\amazon-kinesis-video-strea
       ms-producer-sdk-cpp\kinesis-video-native-build\kinesis_video_gstreamer_sample_app.vcxproj]

    3 Warning(s)
    0 Error(s)

Time Elapsed 00:00:14.91

C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>if ERRORLEVEL 1 goto :showerror

C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>msbuild kinesis_video_gstreamer_sample_rtsp_app.vcxproj  /p:Configuration=Release /p:Platform=x64 /m
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: kinesis_video_gstreamer_sample_rtsp_app.vcxproj

C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>if ERRORLEVEL 1 goto :showerror

C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>cd "C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\downloads\..\"

C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>echo Build error occurred! Please delete related directories in downloads\ to start a clean build.
Build error occurred! Please delete related directories in downloads\ to start a clean build.

C:\Users\psidd\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>exit /b1

zhiyua-git, 2019-02-15:
Hi @sujithaperumal ,

Can you remove these two lines from windows-install-script-with-perl.bat and rerun? We will update the install script soon. Thanks.

**msbuild kinesis_video_gstreamer_sample_rtsp_app.vcxproj  /p:Configuration=Release /p:Platform=%BUILD_BITNESS_TYPE% /m
if ERRORLEVEL 1 goto :showerror**

sujithaperumal, 2019-02-15:
@zhiyua-git It worked!Thanks much!

sujithaperumal, 2019-02-16:
Hi @zhiyua-git,
My video streams successfully to the kinesis video stream. Is it possible to store the streaming video in S3?

sujithaperumal, 2019-02-17:
@zhiyua-git ,
when I try to run the gst-launch-1.0 command in order to send video to the stream, I'm getting this below error 
'gst-launch-1.0' is not recognized as an internal or external command,
operable program or batch file.

zhiyua-git, 2019-02-17:
@sujithaperumal Have you set environment variables as mentioned in step 1 here? https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-windows-msvc.md#running-the-sample-demo-applications-to-stream-video-to-kinesis-video-streams

For your question about storing video to S3, can you provide more information about your use case? Kinesis Video provides video storage itself and you can retrieve the video in retention window through our consumption APIs.

sujithaperumal, 2019-02-18:
[image]
Yes I did and this is what I get when I try to run the gst-launch-1.0 command. The command worked perfectly well the day before.

sujithaperumal, 2019-02-18:
As for storing video to s3, I want to detect objects in a video. I didn't know how to detect objects in a live streaming video. So if I could store the video in s3, I can then use rekognition to detect objects. You also told me that you could use Amazon sagemaker to detect objects and when I tried that option, my sagemaker endpoint gets failed to create due to exec: "serve": executable file not found in $PATH error.

vijaygj, 2019-02-18:
> I didn't know how to detect objects in a live streaming video.

We have two offerings here,

1. https://docs.aws.amazon.com/rekognition/latest/dg/streaming-video.html
2. You can also use your own ML model via SageMaker - https://aws.amazon.com/blogs/machine-learning/analyze-live-video-at-scale-in-real-time-using-amazon-kinesis-video-streams-and-amazon-sagemaker/

MushMal, 2019-02-18:
@sujithaperumal , regarding the gst-launch failure. This simply means that Windows command prompt couldn't find a built-in command or an executable by that name in its standard path. Have you followed the following: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-windows-msvc.md#running-the-sample-demo-applications-to-stream-video-to-kinesis-video-streams

You might need to add the path to your %PATH% variable

sujithaperumal, 2019-02-19:
@MushMal Thanks! It worked!

sujithaperumal, 2019-02-19:
@vijaygj I tried the second option and I get exec: "serve": executable file not found in $PATH error when trying to create the sagemaker endpoint

qqia, 2019-03-01:
@sujithaperumal 
for error
```
"serve": executable file not found in $PATH error when trying to create the sagemaker endpoint
```
Where did you get this error? If SageMaker endpoint creation failing, we would suggest you consulting with SageMaker team.

unicornss, 2019-04-01:
Closing the issue as the question  has been answered. Please reopen or create new issue as needed. Thanks

closed 2019-04-01 by unicornss