awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #178: Error while installing yasm-1.3.0.tar.gz

demoV, 2019-02-18:
as http://www.tortall.net/projects/yasm/releases/yasm-1.3.0.tar.gz got expired. install-script is still using the URL. which is leading to the timeout error. 

I tried with the https://github.com/yasm/yasm/archive/v1.3.0.tar.gz. But it says `./install-script: line 584: ./configure: No such file or directory`. 

Please help me in building the producer.

unicornss, 2019-02-18:
Hi  @demoV, 

Thanks for notifying us on the intermittent availability of the `yasm` open source dependency release. The link http://www.tortall.net/projects/yasm/releases/yasm-1.3.0.tar.gz seems to work now. 

Regarding your other question, could you please elaborate on 

> I tried with the https://github.com/yasm/yasm/archive/v1.3.0.tar.gz. But it says ./install-script: line 584: ./configure: No such file or directory.

Did you update the link in the install-script? or downloaded it manually in the   `amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/` directory?

You can check if the ` amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/` directory has the yasm-1.3.0.tar.gz file downloaded successfully.

Sending the logs earlier to the `./configure: No such file or directory` could help us diagnose the issue.

Thanks

demoV, 2019-02-19:
Hi @unicornss,

Thanks for the early reply. The link http://www.tortall.net/projects/yasm/releases/yasm-1.3.0.tar.gz is still not working for me. 

Regarding my other question
`I tried with the https://github.com/yasm/yasm/archive/v1.3.0.tar.gz. But it says ./install-script: line 584: ./configure: No such file or directory.`

Yes, I updated the install-script. and it is downloading the yasm-1.3.0.tar.gz file. 

There are no prior logs for `./configure: No such file or directory`. As after `tar -xvf yasm-1.3.0.tar.gz`,  below code is not getting called on the yasm-1.3.0 folder, downloaded from the https://github.com/yasm/yasm/archive/v1.3.0.tar.gz. 
```for d in ./*/ ; do
     if [ -f "$d/configure.ac" ]; then
       (cd "$d" && ./"${DOWNLOADS}"/local/bin/autoreconf -f -i);
     fi
   done```

Looking for help.
Thanks

MushMal, 2019-02-19:
@demoV I am not entirely sure what's going on, however, the install-script is exceptionally simple. I just double checked and I am able to access the location of the yasm. This said, it's irrelevant where you get the archive from. If you've manually downloaded the yasm-1.3.0.tar.gz file and placed in the downloads directory, you can simply remove yasm-1.3.0 directory from the downloads and re-run the install-script. If it sees that the .gz file is available it will not attempt to download it again but will simply unpack it and try to build.

Please let us know how that goes.

demoV, 2019-02-19:
@MushMal, 
I am also not sure why http://www.tortall.net/projects/yasm/releases/yasm-1.3.0.tar.gz link is not working for me. 

On other hand I have tried the same thing you mentioned above. tar is in downloaded directory.
when I ran ./install_script it is unpacking the tar and logs are below.

```
... Unpack logs
..
x yasm-1.3.0/tools/re2c/re.h
x yasm-1.3.0/tools/re2c/re2c.1
x yasm-1.3.0/tools/re2c/scanner.c
x yasm-1.3.0/tools/re2c/scanner.h
x yasm-1.3.0/tools/re2c/scanner.re
x yasm-1.3.0/tools/re2c/substr.c
x yasm-1.3.0/tools/re2c/substr.h
x yasm-1.3.0/tools/re2c/token.h
x yasm-1.3.0/tools/re2c/translate.c
x yasm-1.3.0/tools/xdf/
x yasm-1.3.0/tools/xdf/xdf.h
x yasm-1.3.0/tools/xdf/xdfdump.c
x yasm-1.3.0/util.h
x yasm-1.3.0/yasm.1
x yasm-1.3.0/yasm_arch.7
x yasm-1.3.0/yasm_dbgfmts.7
x yasm-1.3.0/yasm_objfmts.7
x yasm-1.3.0/yasm_parsers.7
./install-script: line 584: ./configure: No such file or directory
```

MushMal, 2019-02-19:
@demoV could you check whether in the unpacked directory structure at amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/yasm-1.3.0 contains configure.ac and configure file? Running autoreconf in lines 579-583 would set parameters for the configure script.

Are you running install-script from amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build directory?

P.S. Not sure why the path is inaccessible from your network.

demoV, 2019-02-19:
@MushMal, 
Checked `amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/yasm-1.3.0`. And it has `configure.ac` file but does not have `configure` file. 

Lines 579-583 are not getting executed as  below code is running on `amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/yasm-1.3.0` directory.
```
for d in ./*/ ; do
     if [ -f "$d/configure.ac" ]; then
       (cd "$d" && ./"${DOWNLOADS}"/local/bin/autoreconf -f -i);
     fi
   done
```
here we are trying to find `configure.ac` in subdirectory. But configure.ac file is in yasm-1.3.0 directory not in subdirectory. that is why condition  ` if [ -f "$d/configure.ac" ]; then` is false. 

Attaching the tar [yasm-1.3.0.tar.gz](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/2878759/yasm-1.3.0.tar.gz)

MushMal, 2019-02-19:
Odd, LN: 577 cd $DOWNLOADS/yasm-1.3.0 will chande dir into the yasm directory. Just out of curiosity, could you manually cd to amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/yasm-1.3.0
and run

../local/bin/autoreconf -f -i

it should output configure in yasm directory

demoV, 2019-02-19:
@MushMal, Have tried to run `../local/bin/autoreconf -f -i`. it says `no such file or directory: ../local/bin/autoreconf`. 
Tried with `autoreconf -f -i` and it worked. 

Here looks like `autoreconf` is not present in `../local/bin`. 

What would be the clean solution here?

MushMal, 2019-02-19:
@demoV autoreconf is built and installed as part of the autoconf toolset LN: 439. Something must have gone wrong pulling down and installing the toolset.

Try to remove $DOWNLOADS/local/bin/autoconf and $DOWNLOADS/autoconf-2.69.tar.gz and re-run the install-script.

unicornss, 2019-04-02:
KVS uses the following open source dependencies and if the packaging is different between 
http://www.tortall.net/projects/yasm/releases/yasm-1.3.0.tar.gz and
https://github.com/yasm/yasm/archive/v1.3.0.tar.gz., the maintainer of the yasm project needs to be informed.

Closing the issue for now as the tortall.net site is functional.
Please re-open or create a new one if additional support is required.

Thanks

closed 2019-04-02 by unicornss