awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #179: How can I stream from a plugged-in usb camera on Mac?

voidcenter, 2019-02-21:
Hi I have a USB cam (Microsoft LifeCam) plugged in Mac which works well in Photo Booth. I wonder how can I stream it to Kinesis? I tried several method (looking at _sample_app.cpp, stream to rtsp from VLC and then stream from rtsp) without much luck. I wonder what's the recommended way here? 

(/dev/video* doesn't exist, vlc rtp stream also don't let me set IP address, I'm not sure if I got the rtsp url wrong there.)

Thanks!

MushMal, 2019-02-21:
@voidcenter you don't need to produce an rtsp stream in order to stream to KVS. The USB camera can be used directly. For that purpose you can use GStreamer plugin and use gst-launch command or try using the sample application: 
`Usage: AWS_ACCESS_KEY_ID=YourAccessKeyId AWS_SECRET_ACCESS_KEY=YourSecretAccessKey ./kinesis_video_gstreamer_sample_app <my_stream_name> -w <width> -h <height> -f <framerate> -b <bitrateInKBPS>`

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-macos.md

Let us know whether you are able to run this OK.

voidcenter, 2019-02-21:
hi @MushMal , thanks for your reply. when I ran `kinesis_video_gstreamer_sample_app` it would always use my built-in cam on the laptop instead of using the plugged in cam. I have two cameras on the laptop, wonder if there is any way I can specify which camera it should stream from?  (sorry if i missed it, but from the command you give I don't see where it specify the streaming source.)

chehefen, 2019-02-21:
Hi @voidcenter, can you replace the this line in the sample app https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/135e04361f68636d82594497ad12426b93d7bd32/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L683 with
```
source = gst_element_factory_make("avfvideosrc", "source");
g_object_set(G_OBJECT (source), "device-index", 1, NULL);
```

Also, if you have built libgstkvssink.so in the kinesis-video-native-build folder, you can quickly test things out by doing
```
GST_PLUGIN_PATH=$PWD downloads/local/bin/gst-launch-1.0 avfvideosrc device-index=1 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! vtenc_h264_hw allow-frame-reordering=FALSE realtime=TRUE max-keyframe-interval=45 bitrate=500 ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name=YourStreamName storage-size=128 access-key="access-key" secret-key="secret-key"
```

voidcenter, 2019-02-21:
Hi @chehefen , cool, will try that!  thanks!

voidcenter, 2019-02-22:
Cool @chehefen , those work wonderfully. Thanks!

closed 2019-02-22 by MushMal