awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #182: Getting stream metrics when using kvssink gstreamer element

ib1368, 2019-02-27:
It seems that there is no way getting the current status when using kvssink gstreamer element.
Some good metrics got printed when the logging is set to DEBUG. Is there any way to get them via callback or something like this?

```
[DEBUG][2019-02-27 08:16:08] Kinesis Video client and stream metrics
	>> Overall storage byte size: 536870912
	>> Available storage byte size: 536768996
	>> Allocated storage byte size: 101916
	>> Total view allocation byte size: 144072
	>> Total streams frame rate (fps): 8
	>> Total streams transfer rate (bps): 22297200 (21774 Kbps)
	>> Current view duration (ms): 2
	>> Overall view duration (ms): 6005
	>> Current view byte size: 9765
	>> Overall view byte size: 100030
	>> Current frame rate (fps): 8.87216
	>> Current transfer rate (bps): 22297200 (21774 Kbps)
```

MushMal, 2019-02-27:
@ib1368 what is your scenario and what exactly are you trying to achieve? In general, we are not exposing any of the low-level callbacks in the kvssink element. It's an interesting option but I would like to learn a little more about your scenario. Perhaps, you need more information about the metrics instead of the callbacks? Are you building some custom logic on top of it?

On the other hand, if you are integrating with the SDK directly (similar to what the kvssink does) you can have full control over the callbacks.

ebrahimree, 2019-02-27:
@MushMal Yes. I need to expose these stream metrics to another module.

ib1368, 2019-02-27:
@MushMal Yes. I need to expose these stream metrics to another module.

MushMal, 2019-02-27:
Thanks. As I mentioned, we currently do not expose these metrics through the kvssink - this is something we will look into.

Meanwhile, if you can modify the source code of the kvssink then you could expose these metrics to your downstream consumers. There are two public APIs which expose these metrics:

Producer:
KinesisVideoProducerMetrics getMetrics() const;

Stream:
`KinesisVideoStreamMetrics getMetrics() const;`

In your code you could call these to get the metrics and expose these to the downstream. You could do something similar to the code in KinesisVideoStream::putFrame LN 50

The area where you can insert this in the sample processing in the kvssink just before this line:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gstreamer-plugin/plugin-src/gstkvssink.cpp#L1041

Please let us know if you need any help.

ebrahimree, 2019-03-12:
I investigated it more and saw that in the `gst-inspect-1.0 kvssink` output we have:
```
  qos                 : Generate Quality-of-Service events upstream
                        flags: readable, writable
                        Boolean. Default: false
```

So How can I used this field and what will be the outcome?
there are some ways of waiting for qos messages on gstreamer bus. But as I tested it nothing happened.

ib1368, 2019-03-12:
I investigated it more and saw that in the `gst-inspect-1.0 kvssink` output we have:
```
  qos                 : Generate Quality-of-Service events upstream
                        flags: readable, writable
                        Boolean. Default: false
```

So how can I use this field and what will be the outcome?
there are some ways of waiting for qos messages on gstreamer bus. But as I tested it nothing happened.

ib1368, 2019-03-12:
Also wanted to know if you have any plan to support it, in near future?

unicornss, 2019-04-02:
Hi @ib1368 , As @MushMal  has mentioned, the metrics are not exposed through the kvssink; we will update the Github release when this new feature is supported.  Thanks

closed 2019-04-02 by unicornss