awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #183: Kinesis error using kvssink element on gstreamer

felipe-menelau, 2019-02-28:
So, I've been trying to consume some video coming from a camera through RTSP using gstreamer and the kvssink element, my pipeline is as follows:

`gst-launch-1.0 rtspsrc location="#{creation_params[:camera_url]}" short-header=TRUE ! decodebin ! x264enc bframes=0 key-int-max=5 bitrate=220 ! video/x-h264, stream-format=avc, alignment=au, profile=baseline ! kvssink stream-name="#{creation_params[:stream_name]}" storage-size=512 access-key="#{ENV['AWS_ACCESS_KEY_ID']}" secret-key="#{ENV['AWS_SECRET_ACCESS_KEY']}" aws-region="us-east-1" retention-period=#{creation_params[:retention_period]}`

Thing is, eventually (sometimes more then others) kinesis will reject my fragment, saying that 
FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS, which immediately crashes the whole pipeline, causing some gaps in the video consumption (until I restart it).

Is there any way to make kvssink continue trying to stream fragments to kinesis even if some of them are wrong? I don't mind if I miss some fragments, but the fact that it crashes the whole pipeline makes the gaps pretty big.

unicornss, 2019-03-01:
Hi @felipe-menelau,

Please send us the details of the camera model and the log output when running the gst-launch-1.0 command. Also, could you tell us more about how you are planning to use the Kinesis Video Streams?

Also, you could try running the sample application to send streams from a RTSP camera. 
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-macos.md#running-the-gstreamer-rtsp-sample-application

Thanks

chehefen, 2019-03-01:
can you do `export DEBUG_DUMP_FRAME_INFO=TRUE` and run the pipeline up to the point where you got the FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS error and send us the log? It helps us debug how the error occurred

felipe-menelau, 2019-03-07:
Hey guys, sorry for taking so long to follow-up, we just had Carnival here in Brazil.

@unicornss 
I can't send the video directly from the cameras because I don't have access to them, all I have is the RTSP URI and auth info. My intention is to send that data to kinesis and persist it there so my users can watch specific moments in the future.
This particular camera I'm testing is a Intelbras HDCVI

@unicornss  and @chehefen  about the log output, is there any place I can send it to you guys privately? It's full of what I presume are secret keys and stuff like that, so I don't think here is a good place.

Thank you for the quick responses!

chehefen, 2019-03-13:
sorry for the late reply, you can send the log to this email: kinesis-video-support@amazon.com. Thanks

MushMal, 2019-12-04:
@felipe-menelau did this issue get resolved?

MushMal, 2020-03-17:
Closing as stale

closed 2020-03-17 by MushMal