awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #185: FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS ErrorId":4004

xbwei, 2019-03-07:
Hi, I was able to launch the skd with `./kinesis_video_gstreamer_sample_app test`, but continuously get the following warnings and errors, and the video is not playing on Kinesis video streams.

I am using Raspberry Pi and the default Pi camera v2 

[DEBUG][2019-03-07 02:01:59] Connection for Kinesis Video stream: test closed.
[DEBUG][2019-03-07 02:01:59] Network thread for Kinesis Video stream: test with upload handle: 180 exited. http status: 0
[DEBUG][2019-03-07 02:01:59] describeStream response: {"StreamInfo":{"CreationTime":1.551924009774E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-east-1:400720404473:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:400720404473:stream/test/1551924009774","StreamName":"test","Version":"cWNvNr5iIQ1AA8LBuDFE"}}
[DEBUG][2019-03-07 02:01:59] Discovered existing Kinesis Video stream: arn:aws:kinesisvideo:us-east-1:400720404473:stream/test/1551924009774
[DEBUG][2019-03-07 02:01:59] stream arn in stream_info struct: arn:aws:kinesisvideo:us-east-1:400720404473:stream/test/1551924009774
[INFO ][2019-03-07 02:01:59]
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2019-03-07 02:01:59] streamingEndpointHandler invoked
[DEBUG][2019-03-07 02:01:59] Refreshing credentials. Force refreshing: 0 Now time is: 1551924119580822905 Expiration: 0
[DEBUG][2019-03-07 02:01:59] getStreamingEndpoint response: {"DataEndpoint":"https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com"}
[INFO ][2019-03-07 02:01:59] streaming to endpoint: https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com
[INFO ][2019-03-07 02:01:59]
getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2019-03-07 02:01:59] streamingTokenHandler invoked
[DEBUG][2019-03-07 02:01:59] Refreshing credentials. Force refreshing: 1 Now time is: 1551924119690868297 Expiration: 1551924239
[INFO ][2019-03-07 02:01:59] New credentials expiration is 1551924239
[INFO ][2019-03-07 02:01:59]
getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2019-03-07 02:01:59] streamReadyHandler invoked
[DEBUG][2019-03-07 02:01:59] putStreamHandler invoked
[DEBUG][2019-03-07 02:01:59] Created a new PutMedia request with stream upload handle: 272
[INFO ][2019-03-07 02:01:59]
putStreamResultEvent(): Put stream result event.
[INFO ][2019-03-07 02:01:59] Creating new connection for Kinesis Video stream: test
[DEBUG][2019-03-07 02:01:59] Refreshing credentials. Force refreshing: 0 Now time is: 1551924119693383757 Expiration: 0
[DEBUG][2019-03-07 02:01:59] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 272
[INFO ][2019-03-07 02:01:59] Curl post header write function returned:x-amzn-RequestId: fc6c6fa5-c413-0b76-a896-1272951ea442

[DEBUG][2019-03-07 02:01:59] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 272
[DEBUG][2019-03-07 02:01:59] Wrote 11907 bytes to Kinesis Video. Upload stream handle: 272
[DEBUG][2019-03-07 02:01:59] Pausing CURL read for upload handle: 272
[DEBUG][2019-03-07 02:01:59] Curl post body write function for stream: test and upload handle: 272 returned: {"EventType":"ERROR","FragmentTimecode":0,"FragmentNumber":"91343852333182779276396565269381279958750066125","ErrorCode":"FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS","ErrorId":4004}

[DEBUG][2019-03-07 02:01:59] describeStreamHandler invoked
[DEBUG][2019-03-07 02:01:59] streamErrorHandler invoked
[ERROR][2019-03-07 02:01:59] Reporting stream error. Errored timecode: 3990000 Status: 1375731808
[DEBUG][2019-03-07 02:01:59] Refreshing credentials. Force refreshing: 0 Now time is: 1551924119843715829 Expiration: 0
[DEBUG][2019-03-07 02:01:59] fragmentAckReceivedHandler invoked
[WARN ][2019-03-07 02:01:59] Reporting fragment ack received. Ack timecode 0
[ERROR][2019-03-07 02:01:59] curl perform failed for url https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
[WARN ][2019-03-07 02:01:59] HTTP Error 0: Response:
Request URL: https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com/putMedia
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAI2PQ4O4HI3KE2WCQ/20190307/us-east-1/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowledgment-required;x-amzn-fragment-timecode-type;x-amzn-producer-start-timestamp;x-amzn-stream-name, Signature=1847b89191509a0caf10a132ba55f2174b2c964f2cd188927879b4ffb3d55c76
    connection: keep-alive
    host: s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com
    transfer-encoding: chunked
    user-agent: AWS-SDK-KVS/1.7.7 GCC/6.3.0 Linux/4.14.98-v7+ armv7l
    X-Amz-Date: 20190307T020123Z
    x-amzn-fragment-acknowledgment-required: 1
    x-amzn-fragment-timecode-type: RELATIVE
    x-amzn-producer-start-timestamp: 1551924083.236
    x-amzn-stream-name: test

And here is the screen on AWS kinesis:
[image]

danielbfr, 2019-03-08:
I also find this exact error when running the Docker container on my Macbook with a standard RTSP camera

kendog, 2019-03-10:
Same here...  I've scoured the internet and have found nothing...
I am using Raspberry Pi 3 and the default Pi camera v2

kendog, 2019-03-10:
Hey folks,

I was able to resolve my issue by passing the width, height, and framerate:
./kinesis_video_gstreamer_sample_app test -w 640 -h 480 -f 15

Hope this works for you guys.

xbwei, 2019-03-10:
> 
> 
> Hey folks,
> 
> I was able to resolve my issue by passing the width, height, and framerate:
> ./kinesis_video_gstreamer_sample_app test -w 640 -h 480 -f 15
> 
> Hope this works for you guys.

Thanks, @kendog , mine is also working now with the same parameters, cheers!

danielbfr, 2019-03-12:
I was able to get going just by not using Docker for the same script. Seems like something doesn't quite work with the startup script maybe, or something with Docker's networking for an RTSP camera

chehefen, 2019-03-12:
Hi @danielbfr, can you do `export DEBUG_DUMP_FRAME_INFO=TRUE` before running the application and send us the log where the error occurs? thanks.

msjoshi, 2019-03-19:
@chehefen 

I have added the logs with `DEBUG_DUMP_FRAME_INFO=TRUE` in #193

chehefen, 2019-07-25:
#193 
As mention by @msjoshi, configuring rtspsrc to use tcp only resolves the issue
```
rtspsrc location=$4 short-header=TRUE protocols=GST_RTSP_LOWER_TRANS_TCP
```

closed 2019-07-25 by chehefen