awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #187: Docker build fails due to install-script flag

boomshadow, 2019-03-07:
Using the `Dockerfile` provided in this project (https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docker_native_scripts/Dockerfile), I receive the following error:

```
Step 10/16 : RUN ./install-script -a -j 4 -d
 ---> Running in 72e6dff3f9d6
 ===========================Starting to build ===========================
              Kinesis Video Stream Producer SDK and Demo Application
       ========================================================================
       Following open source libraries and tools will be downloaded and built
       before building the producer sdk.
       ========================================================================
       # ------------------------- BUILD-TOOLS -------------------------------#
        autoconf   automake
        bison      curl
        flex       gettext
        googletest iconv
        libffi     libpcre
        libtools   log4cplus
        m4         openssl
        zlib

        # ------------------------- GSTREAMER - for demo application --------#
        GStreamer, GStreamer plugins, yasm, gst-libav and x264.
        ------------------------------------------------------------------------
        *** NOTE *** for MacOS High Sierra: Use
            brew install autoconf
            brew install bison before running this script
        ------------------------------------------------------------------------
        ========================================================================
        This install script downloads and builds the dependent libraries from
        source in the ./kinesis-video-native-build/downloads directory only.
        ========================================================================
 
Unknown option -a
The command '/bin/sh -c ./install-script -a -j 4 -d' returned a non-zero code: 1
ERROR: Job failed: exit code 1
```

It looks like that ['install-script'](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-native-build/install-script) file was edited in the last 8 hours with the `1.7.7` release.

chehefen, 2019-03-07:
The `-a` command line arg should be removed. Can you pull latest and try again? Thanks

boomshadow, 2019-03-07:
All good now. Thank you so much!

closed 2019-03-07 by boomshadow