awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #189: Docker image no longer builds

danielbfr, 2019-03-08:
The command line flag '-a' was removed from ./kinesis-video-native-build/install-script, and now the Docker build fails because it's looking for command input. 
Step 9/16 : RUN ./install-script -j 4 -d
 ---> Running in 237865880a41
 ===========================Starting to build ===========================
              Kinesis Video Stream Producer SDK and Demo Application
       ========================================================================
       Following open source libraries and tools will be downloaded and built
       before building the producer sdk.
       ========================================================================
       # ------------------------- BUILD-TOOLS -------------------------------#
        autoconf   automake
        bison      curl
        flex       gettext
        googletest iconv
        libffi     libpcre
        libtools   log4cplus
        m4         openssl
        zlib

        # ------------------------- GSTREAMER - for demo application --------#
        GStreamer, GStreamer plugins, yasm, gst-libav and x264.
        ------------------------------------------------------------------------
        *** NOTE *** for MacOS High Sierra: Use
            brew install autoconf
            brew install bison before running this script
        ------------------------------------------------------------------------
        ========================================================================
        This install script downloads and builds the dependent libraries from
        source in the ./kinesis-video-native-build/downloads directory only.
        ========================================================================

Unknown option -a
The command '/bin/sh -c ./install-script -a -j 4 -d' returned a non-zero code: 1

chehefen, 2019-03-08:
can you try disable docker cache by doing `docker build --no-cache -t test /path/to/Dockerfile`?

danielbfr, 2019-03-08:
Yeah seems good, I didn't realize that part would be cached

closed 2019-03-08 by danielbfr