awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #193: Internal data stream error when streaming from RTSP source.

msjoshi, 2019-03-14:
I'm using the producer library in a Docker container to stream from RTSP source. The RTSP source is Amcrest cameras (model # AMNV20M8-4B-B).

I'm using the following startup script to trigger the stream. 

```
#!/bin/bash

AWS_ACCESS_KEY_ID=$1 AWS_SECRET_ACCESS_KEY=$2 AWS_DEFAULT_REGION="us-east-1" gst-launch-1.0 rtspsrc location=rtsp://$3 short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name=$4 storage-size=512
```

There are 2 issues. It tries to receive UDP packets for 5 seconds and then switches to TCP. On my Mac, I have switched off firewall to ensure that's not a problem. 

After it switches to TCP it gives warnings about could not transform video/x-h264 and eventually fails with internal data stream error. 

Below are the logs after the pipeline starts playing. 

```
Progress: (open) Retrieving server options

Progress: (open) Retrieving media info

Progress: (request) SETUP stream 0

Progress: (open) Opened Stream

Setting pipeline to PLAYING ...

0:00:00.373601700   267      0x1baaef0 WARN                     bin gstbin.c:2772:gst_bin_do_latency_func:<pipeline0> did not really configure latency of 0:00:00.000000000

New clock: GstSystemClock

Progress: (request) Sending PLAY request

Progress: (request) Sending PLAY request

0:00:00.375093800   267 0x7f416803e450 FIXME                default gstutils.c:3902:gst_pad_create_stream_id_internal:<fakesrc0:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id

Progress: (request) Sent PLAY request

0:00:05.572830600   267      0x1bab450 WARN                 rtspsrc gstrtspsrc.c:4936:gst_rtspsrc_reconnect:<rtspsrc0> warning: Could not receive any UDP packets for 5.0000 seconds, maybe your firewall is blocking it. Retrying using a tcp connection.

WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not read from resource.

Additional debug info:

gstrtspsrc.c(4936): gst_rtspsrc_reconnect (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
Could not receive any UDP packets for 5.0000 seconds, maybe your firewall is blocking it. Retrying using a tcp connection.

0:00:08.287093000   267 0x7f416803cb70 WARN           basetransform gstbasetransform.c:1346:gst_base_transform_setcaps:<capsfilter0> transform could not transform video/x-h264, stream-format=(string)avc, alignment=(string)au, codec_data=(buffer)01640028ffe1000d27640028ac1ad00f0044bca80001000628ee0462c000, level=(string)4, profile=(string)high in anything we support

0:00:08.288605800   267 0x7f416803cb70 WARN           basetransform gstbasetransform.c:1346:gst_base_transform_setcaps:<capsfilter0> transform could not transform video/x-h264, stream-format=(string)avc, alignment=(string)au, codec_data=(buffer)01640028ffe1000d27640028ac1ad00f0044bca80001000628ee0462c000, level=(string)4, profile=(string)high in anything we support

0:00:08.288699500   267 0x7f416803cb70 WARN           basetransform gstbasetransform.c:1346:gst_base_transform_setcaps:<capsfilter0> transform could not transform video/x-h264, stream-format=(string)avc, alignment=(string)au, codec_data=(buffer)01640028ffe1000d27640028ac1ad00f0044bca80001000628ee0462c000, level=(string)4, profile=(string)high in anything we support

0:00:08.388455700   267      0x1bab450 WARN                 rtspsrc gstrtspsrc.c:5162:gst_rtspsrc_loop:<rtspsrc0> error: Internal data stream error.

0:00:08.388699000   267      0x1bab450 WARN                 rtspsrc gstrtspsrc.c:5162:gst_rtspsrc_loop:<rtspsrc0> error: streaming stopped, reason not-negotiated (-4)

ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Internal data stream error.

Additional debug info:

gstrtspsrc.c(5162): gst_rtspsrc_loop (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
streaming stopped, reason not-negotiated (-4)

Execution ended after 0:00:08.015984100

Setting pipeline to PAUSED ...

Setting pipeline to READY ...

Setting pipeline to NULL ...

0:00:08.391575500   267      0x1bab450 WARN                 rtspsrc gstrtspsrc.c:5466:gst_rtspsrc_try_send:<rtspsrc0> receive interrupted

0:00:08.391630000   267      0x1bab450 WARN                 rtspsrc gstrtspsrc.c:6974:gst_rtspsrc_close:<rtspsrc0> TEARDOWN interrupted

Freeing pipeline ...

INFO - Freeing Kinesis Video Stream rtspdockertest

2019-03-13 17:34:51 [139919806072640] INFO -
freeKinesisVideoStream(): Freeing Kinesis Video stream.

2019-03-13 17:34:51 [139919806072640] DEBUG - streamClosedHandler invoked for upload handle: 18446744073709551615
```

msjoshi, 2019-03-14:
Update: 

I tried a public RTSP url and it does not work either. 

rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov

igoticecream, 2019-03-14:
Came here looking for a solution for the same issue

chehefen, 2019-03-18:
Hi, can you add `GST_DEBUG=4` to the pipeline like such: 
```
GST_DEBUG=4 AWS_ACCESS_KEY_ID=$1 AWS_SECRET_ACCESS_KEY=$2 AWS_DEFAULT_REGION="us-east-1" gst-launch-1.0 rtspsrc location=rtsp://$3 short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name=$4 storage-size=512
```
Then send us the log? 
Thanks

msjoshi, 2019-03-18:
@chehefen please find the attached log with debug level 4. 
[stream.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/2979922/stream.log)

msjoshi, 2019-03-18:
@chehefen 

Please find the attached logs with debug level 4. 

[stream.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/2979932/stream.log)

chehefen, 2019-03-18:
instead of `format=avc`, can you try `stream-format=avc`? You can actually remove the entire `video/x-h264, format=avc,alignment=au` section as it has already been coded into the kvssink.

msjoshi, 2019-03-18:
@chehefen 

No difference if I changed `format=avc` to `stream-format=avc`.

However, if I remove the entire `video/x-h264` from the pipeline, it does not give a transform error like before but fails with internal data stream error, reason not negotiated (-4). I have attached those logs below. 

[stream.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/2980179/stream.log)

chehefen, 2019-03-18:
can you try this pipeline?
```
AWS_ACCESS_KEY_ID=$1 AWS_SECRET_ACCESS_KEY=$2 AWS_DEFAULT_REGION="us-east-1" gst-launch-1.0 rtspsrc location=rtsp://$3 ! rtph264depay ! h264parse ! kvssink stream-name=$4 storage-size=512
```

msjoshi, 2019-03-18:
@chehefen 

Adding `h264parse` in pipeline worked. Thanks!

msjoshi, 2019-03-19:
@chehefen 

After some more testing, I found one more issue. After about 4 minutes viewing the live camera I get an error `FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS`. Below is the log snippet. Could this be related to https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/185

```
2019-03-19 14:35:27 [139997058459392] DEBUG - fragmentAckReceivedHandler invoked
2019-03-19 14:35:27 [139997058459392] DEBUG - Curl post body write function for stream: rtspdockertest and upload handle: 0 returned: {"EventType":"ERROR","FragmentTimecode":238526,"FragmentNumber":"91343852333181442850799514501683680165321662322","ErrorCode":"FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS","ErrorId":4004}

2019-03-19 14:35:27 [139997058459392] DEBUG - describeStreamHandler invoked
2019-03-19 14:35:27 [139997058459392] DEBUG - streamErrorHandler invoked
2019-03-19 14:35:27 [139997058459392] ERROR - Reporting stream error. Errored timecode: 176130000 Status: 0x52000060
2019-03-19 14:35:27 [139997058459392] DEBUG - fragmentAckReceivedHandler invoked
2019-03-19 14:35:27 [139997058459392] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1553006127454698500 Expiration: 0
2019-03-19 14:35:27 [139997066852096] INFO - Closing connection for upload stream handle: 0
2019-03-19 14:35:27 [139997058459392] DEBUG - Wrote 0 bytes to Kinesis Video. Upload stream handle: 0
2019-03-19 14:35:27 [139997058459392] DEBUG - terminate curl for handle 0
2019-03-19 14:35:27 [139997058459392] DEBUG - Connection for Kinesis Video stream: rtspdockertest closed.
2019-03-19 14:35:27 [139997058459392] DEBUG - Network thread for Kinesis Video stream: rtspdockertest with upload handle: 0 exited. http status: 200
0:04:04.606060600    65 0x7f539c048d90 WARN                 kvssink gstkvssink.cpp:1095:gst_kvs_sink_handle_buffer:<kvssink0> error: Stream error occurred. Status: 0x52000060
0:04:04.606152300    65 0x7f539c048d90 INFO        GST_ERROR_SYSTEM gstelement.c:1938:gst_element_message_full_with_details:<kvssink0> posting message: GStreamer encountered a general stream error.
0:04:04.606210700    65 0x7f539c048d90 INFO        GST_ERROR_SYSTEM gstelement.c:1965:gst_element_message_full_with_details:<kvssink0> posted error message: GStreamer encountered a general stream error.
0:04:04.606289100    65 0x7f539c048d90 INFO                    task gsttask.c:319:gst_task_func:<rtpjitterbuffer0:src> Task going to paused
ERROR: from element /GstPipeline:pipeline0/GstKvsSink:kvssink0: GStreamer encountered a general stream error.
Additional debug info:
/opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/plugin-src/gstkvssink.cpp(1095): gst_kvs_sink_handle_buffer (): /GstPipeline:pipeline0/GstKvsSink:kvssink0:
Stream error occurred. Status: 0x52000060
2019-03-19 14:35:27 [139997058459392] Execution ended after 0:04:04.314214100
Setting pipeline to PAUSED ...
```

msjoshi, 2019-03-19:
@chehefen 

Please find the attached logs with `DEBUG_DUMP_FRAME_INFO=true`. Please note this time the error occurred after 48 minutes of live video and the logs are about 21MB. 

[stream.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/2984361/stream.log)

MushMal, 2019-03-19:
@msjoshi looking at the logs I see the following:

- Most of the streaming is going on OK with very little in the buffer
- There are bursts of frame rates hundreds of frames per second - this is very odd.
- Initial stream is in tens of Mbps but it stabilizes at around 6-7Mbps

Closer to the end the following error is returned from the backend.
`
2019-03-19 15:45:04 [140333919778560] DEBUG - Curl post body write function for stream: rtspdockertest and upload handle: 1 returned: {"EventType":"ERROR","FragmentTimecode":524611,"FragmentNumber":"91343852333181458364664086826070406453078970563","ErrorCode":"FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS","ErrorId":4004}
`

I am still not sure yet why the media pipeline issues a frame timecode that's less. There is very little in the GST logs. Needs more investigation.

chehefen, 2019-03-21:
The issue is that the frame rate is not stable, can you try adding frame rate setting in the capsfilter like this
```
AWS_ACCESS_KEY_ID=$1 AWS_SECRET_ACCESS_KEY=$2 AWS_DEFAULT_REGION="us-east-1" gst-launch-1.0 rtspsrc location=rtsp://$3 ! rtph264depay ! h264parse ! video/x-h264,framerate=30/1 ! kvssink stream-name=$4 storage-size=512
```
Also, check your camera settings to make sure there is nothing making the frame rate unstable.

msjoshi, 2019-03-21:
@chehefen 

Adding the frame rate in caps filter did not work. It fails instantly. See the logs below. 

```
Setting pipeline to PLAYING ...
0:00:01.073514400   236      0x1ab1900 WARN                     bin gstbin.c:2772:gst_bin_do_latency_func:<pipeline0> did not really configure latency of 0:00:00.000000000
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
0:00:01.076400100   236 0x7f4b88048450 FIXME                default gstutils.c:3902:gst_pad_create_stream_id_internal:<fakesrc0:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
Progress: (request) Sent PLAY request
0:00:06.264090500   236      0x1a6f1e0 WARN                 rtspsrc gstrtspsrc.c:4936:gst_rtspsrc_reconnect:<rtspsrc0> warning: Could not receive any UDP packets for 5.0000 seconds, maybe your firewall is blocking it. Retrying using a tcp connection.
WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not read from resource.
Additional debug info:
gstrtspsrc.c(4936): gst_rtspsrc_reconnect (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
Could not receive any UDP packets for 5.0000 seconds, maybe your firewall is blocking it. Retrying using a tcp connection.
0:00:08.386078100   236      0x1a6f1e0 WARN                 rtspsrc gstrtspsrc.c:5162:gst_rtspsrc_loop:<rtspsrc0> error: Internal data stream error.
0:00:08.386147500   236      0x1a6f1e0 WARN                 rtspsrc gstrtspsrc.c:5162:gst_rtspsrc_loop:<rtspsrc0> error: streaming stopped, reason not-linked (-1)
ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Internal data stream error.
Additional debug info:
gstrtspsrc.c(5162): gst_rtspsrc_loop (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
streaming stopped, reason not-linked (-1)
Execution ended after 0:00:07.312176900
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
0:00:08.387111200   236      0x1a6f1e0 WARN                 rtspsrc gstrtspsrc.c:5466:gst_rtspsrc_try_send:<rtspsrc0> receive interrupted
0:00:08.387155500   236      0x1a6f1e0 WARN                 rtspsrc gstrtspsrc.c:7519:gst_rtspsrc_pause:<rtspsrc0> PAUSE interrupted
Freeing pipeline ...
INFO - Freeing Kinesis Video Stream rtspdockertest
2019-03-21 18:18:21 [139963131811648] INFO -
freeKinesisVideoStream(): Freeing Kinesis Video stream.
2019-03-21 18:18:21 [139963131811648] DEBUG - streamClosedHandler invoked for upload handle: 18446744073709551615
```

MushMal, 2019-03-21:
Looks more like a network issue

msjoshi, 2019-03-21:
@MushMal 

The below works without frame rate caps filter. Randomly, I get the `FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS` error. 

```
GST_DEBUG=3 GST_DEBUG_NO_COLOR=1 AWS_ACCESS_KEY_ID=$1 AWS_SECRET_ACCESS_KEY=$2 AWS_DEFAULT_REGION="us-n=rtsp://user:password@192.168.1.197:554 short-header=TRUE ! rtph264depay ! h264parse ! kvssink stream-name=rtspdockertest storage-size=512
```

MushMal, 2019-03-21:
@msjoshi as @chehefen mentioned, the original issue stems from the fact that your media pipeline is very "bursty" - it can produce hundreds of frames in a second and the frame timestamps are very close to each other. In this case, they fall within a millisecond which is the default timecode scale the SDK uses for MKV packaging which later causes the issue on the backend.

I will let @chehefen comment if he can think of why the capsfilter didn't work or some sort of frame rate limiter.

There are other remedies, you could potentially decrease the millisecond timecode scale to 100micros but that might have other consequences like limiting the fragment duration (MKV specification uses 16bit signed value to represent the relative timecodes of the frames).

msjoshi, 2019-03-21:
@MushMal 

I agree that frame rate is too high. I don't have access to camera settings but will get there soon and report the findings here.

@chehefen 

In the meanwhile, appreciate if you can let me know there is way to stabilize the frame rate with caps filter.

msjoshi, 2019-03-21:
@chehefen 

The following command worked. 

`GST_DEBUG=3 GST_DEBUG_NO_COLOR=1 AWS_ACCESS_KEY_ID=$1 AWS_SECRET_ACCESS_KEY=$2 AWS_DEFAULT_REGION="us-east-1" gst-launch-1.0 rtspsrc location=rtsp://user:password@192.168.1.197:554 short-header=TRUE ! rtph264depay ! video/x-h264,framerate=25/1 ! h264parse ! kvssink stream-name=rtspdockertest storage-size=512`

However, it did not stabilize the frame rate. Below are the sample metrics.

```
getKinesisVideoMetrics(): Get the memory metrics size.
2019-03-21 19:46:34 [140141360314112] DEBUG - Kinesis Video client and stream metrics
	>> Overall storage byte size: 536870912
	>> Available storage byte size: 535596807
	>> Allocated storage byte size: 1274105
	>> Total view allocation byte size: 144072
	>> Total streams frame rate (fps): 407
	>> Total streams transfer rate (bps): 6124976 (5981 Kbps)
	>> Current view duration (ms): 400
	>> Overall view duration (ms): 6000
	>> Current view byte size: 340697
	>> Overall view byte size: 1271686
	>> Current frame rate (fps): 407.17
	>> Current transfer rate (bps): 6124976 (5981 Kbps)
```

It varies a lot but always in hundreds.

chehefen, 2019-03-21:
Can you try this pipeline:
```
GST_DEBUG=3 GST_DEBUG_NO_COLOR=1 AWS_ACCESS_KEY_ID=$1 AWS_SECRET_ACCESS_KEY=$2 AWS_DEFAULT_REGION="us-east-1" gst-launch-1.0 rtspsrc location=rtsp://user:password@192.168.1.197:554 short-header=TRUE ! rtph264depay ! video/x-h264,framerate=25/1 ! decodebin ! fpsdisplaysink
```
and let us know what fps you observe? 
Thanks

msjoshi, 2019-03-21:
@chehefen 

With the above command, I'm observing an average of 24 frames/sec. I have attached the entire log of few seconds of video. 

[stream.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/2994017/stream.log)

msjoshi, 2019-04-02:
@unicornss I see a `successful streaming` label has been added to this issue but that is not the case. This issue is still pending. 

My latest log shows the result when using `fpsdisplaysink` where it shows the correct frame rate. However, when I use `kvssink` the frame rate is too high and randomly causes dropped frame failure and streaming ends.

unicornss, 2019-04-02:
Hi @msjoshi 

Sorry for the incorrect label; removed it now.  
Are you still seeing same stream error `"FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS"` after eight  minutes?

From previous log:

> 2019-03-19 14:57:07 [140333928171264] DEBUG - Curl post body write function for stream:"BUFFERING","FragmentTimecode":0,"FragmentNumber":"91343852333181447247962534043354765710711107616"}
\: 
\:
> 2019-03-19 15:45:04 [140333928171264] DEBUG - Curl post body write function for stream:"RECEIVED","FragmentTimecode":522616,"FragmentNumber":"91343852333181458359712326668928884772144606955"}

> 2019-03-19 15:45:04 [140333919778560] DEBUG - Curl post body write function for stream:"ERROR","FragmentTimecode":524611,"FragmentNumber":"91343852333181458364664086826070406453078970563","ErrorCode":"FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS","ErrorId"

chehefen, 2019-04-02:
Hi @msjoshi ,

If you dont mind, can you add `add_definitions(-DKVS_DEBUG_DUMP_DATA_FILE)` at https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-native-build/CMakeLists.txt#L95. Then rebuild the the plugin and stream until the error happens. That line will cause a couple mkv files to be dumped out. If you can send that mkv to us via kinesis-video-support@amazon.com, it will help us analyze how to fix the issue.

Thanks

msjoshi, 2019-04-03:
@chehefen 

I emailed the mkv file as requested.

msjoshi, 2019-04-18:
@chehefen 

I sent the mkv file as requested couple weeks ago. Any update?

chehefen, 2019-04-18:
can you try this pipeline? 
```
GST_PLUGIN_PATH=$PWD AWS_ACCESS_KEY_ID=$1 AWS_SECRET_ACCESS_KEY=$2 AWS_DEFAULT_REGION="us-east-1" gst-launch-1.0 rtspsrc location=rtsp://$3 ! rtpjitterbuffer mode=none ! rtph264depay ! h264parse ! kvssink stream-name=$4 storage-size=512
```
Try to play with rtpjitterbuffer's mode. it can be either none, slave, buffer or synced.

MushMal, 2019-04-23:
@msjoshi any update on this issue?

msjoshi, 2019-04-23:
@chehefen @MushMal 

I'll try it soon and update here.

Raghava248, 2019-05-02:
Hi,
Iam using ubuntu 16.04
am trying to run this
gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name="" storage-size=512 access-key="" secret-key="" aws-region=""
WARNING: erroneous pipeline: no element "h264parse"
after specfying all the keys and stream name it is giving me this error
basically Iam trying to stream video using amazon kinesis

MushMal, 2019-05-03:
@Raghava248 it seems like the issue you are referring to is not related to the issues highlighted in this topic. Do you mind posting this to a new issue? From the error it seems that you are missing h264parse element. Can you describe whether you are using install script and if it succeeds building all the open source dependencies including GStreamer plugins?

Raghava248, 2019-05-07:
Thank you I cleared the issue by exporting paths
but I am facing the new problem Iam unable to see the video in kinesis window
[image]
[image]

msjoshi, 2019-05-07:
@Raghava248 

This issue is un-related to the issue you are seeing. Please open a separate issue. By mixing multiple issues we lose track of the original issue.

msjoshi, 2019-05-12:
@MushMal @chehefen 

I tried different modes in `rtpjitterbuffer` and did not see any difference in frame rates. When I tried `fpsdisplaysink` earlier the frame rates are shown correctly. Check my post from April 2. 

I'm trying to understand why there is such a large difference between KVS metrics for frame rate and `fpsdisplaysink`. 

I must also mention that ever since I set the protocol in `rtspsrc` to TCP only, the errors and streaming failure went away. Below is the command I'm using in my start streaming script. 

```
#!/bin/bash
GST_DEBUG=$1 AWS_ACCESS_KEY_ID=$2 AWS_SECRET_ACCESS_KEY=$3 AWS_DEFAULT_REGION="us-east-1" gst-launch-1.0 rtspsrc location=$4 short-header=TRUE protocols=GST_RTSP_LOWER_TRANS_TCP ! rtph264depay ! video/x-h264,framerate=25/1 ! h264parse ! kvssink stream-name=$5 storage-size=512
```

Notice `rtspsrc location=$4 short-header=TRUE protocols=GST_RTSP_LOWER_TRANS_TCP`. 

I'm ready to close this ticket as it has  fixed one problem but I do have a latency issue with specific  RTSP source. I'll open a new issue so it can be focussed on latency problem.

MushMal, 2019-05-13:
@msjoshi while I am not an expert in the rtsp gstreamer plugins, I believe the reason for the observed frame rate is due to "burstiness" of the media pipeline. The KVS core calculates the metrics by adding some "inertia" by deploying EMA (Exponential Mean Averaging) which works really well for steady real-time encoder outputs but is thrown off by bursty sources. I believe that fpsdisplaysink smoothes out the source burstiness and this is the reason why you can see the metrics reflecting the observed FPS.

MushMal, 2019-07-24:
Resolving this issue for now. Please create a new one as this thread got too long

closed 2019-07-24 by MushMal

sumanghosh12, 2020-05-01:
@MushMal  

Here is the GST Debug log.

Some useful command to capture the GST Debug logs and redirect them to a file. 
export GST_DEBUG_NO_COLOR=1
export GST_DEBUG_FILE=test.txt

GST Debug log below.

0:00:00.000207127  3744      0x218ee00 INFO                GST_INIT gst.c:510:init_pre: Initializing GStreamer Core Library version 1.10.4
0:00:00.000271080  3744      0x218ee00 INFO                GST_INIT gst.c:511:init_pre: Using library installed in /usr/lib64
0:00:00.000281493  3744      0x218ee00 INFO                GST_INIT gst.c:522:init_pre: Linux lvnwdkpsdk001 3.10.0-1062.18.1.el7.x86_64 #1 SMP Tue Mar 17 23:49:17 UTC 2020 x86_64
0:00:00.000557502  3744      0x218ee00 INFO                GST_INIT gstmessage.c:126:_priv_gst_message_initialize: init messages
0:00:00.001024545  3744      0x218ee00 INFO                GST_INIT gstcontext.c:83:_priv_gst_context_initialize: init contexts
0:00:00.001248278  3744      0x218ee00 INFO      GST_PLUGIN_LOADING gstplugin.c:316:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.001345318  3744      0x218ee00 INFO      GST_PLUGIN_LOADING gstplugin.c:224:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.001352756  3744      0x218ee00 INFO      GST_PLUGIN_LOADING gstplugin.c:226:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.001376931  3744      0x218ee00 INFO            GST_REGISTRY gstregistry.c:1738:ensure_current_registry: reading registry cache: /root/.cache/gstreamer-1.0/registry.x86_64.bin
0:00:00.009359766  3744      0x218ee00 INFO            GST_REGISTRY gstregistrybinary.c:619:priv_gst_registry_binary_read_cache: loaded /root/.cache/gstreamer-1.0/registry.x86_64.bin in 0.007963 seconds
0:00:00.009416437  3744      0x218ee00 INFO            GST_REGISTRY gstregistry.c:1594:scan_and_update_registry: Validating plugins from registry cache: /root/.cache/gstreamer-1.0/registry.x86_64.bin
0:00:00.012678650  3744      0x218ee00 INFO            GST_REGISTRY gstregistry.c:1696:scan_and_update_registry: Registry cache has not changed
0:00:00.012691955  3744      0x218ee00 INFO            GST_REGISTRY gstregistry.c:1773:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.012711366  3744      0x218ee00 INFO                GST_INIT gst.c:720:init_post: GLib runtime version: 2.56.1
0:00:00.012716992  3744      0x218ee00 INFO                GST_INIT gst.c:722:init_post: GLib headers version: 2.46.2
0:00:00.012721027  3744      0x218ee00 INFO                GST_INIT gst.c:723:init_post: initialized GStreamer successfully
0:00:00.012749595  3744      0x218ee00 INFO            GST_PIPELINE gstparse.c:325:gst_parse_launch_full: parsing pipeline description 'rtspsrc location=rtsp://XXXXXXXXXXXXXXX short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name=XXXXXXXXXXXXXXX storage-size=512 access-key=XXXXXXXXXXXXXXX secret-key=XXXXXXXXXXXXXXX aws-region=XXXXXXXXXXXXXXX '
0:00:00.015271405  3744      0x218ee00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib64/gstreamer-1.0/libgstrtsp.so" loaded
0:00:00.015287465  3744      0x218ee00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "rtspsrc"
0:00:00.017173891  3744      0x218ee00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib64/gstreamer-1.0/libgstrtp.so" loaded
0:00:00.017191083  3744      0x218ee00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "rtph264depay"
0:00:00.017266956  3744      0x218ee00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstRTPBaseDepayload@0x22ca090> adding pad 'sink'
0:00:00.017283548  3744      0x218ee00 INFO        GST_ELEMENT_PADS gstelement.c:668:gst_element_add_pad:<GstRTPBaseDepayload@0x22ca090> adding pad 'src'
0:00:00.029857396  3744      0x218ee00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "XXXXXXXXXXXXXXX/FaceRekognition/amazon-kinesis-video-streams-producer-sdk-cpp/build/libgstkvssink.so" loaded
0:00:00.029878351  3744      0x218ee00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "kvssink"
0:00:00.030189832  3744      0x218ee00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:364:gst_element_factory_create: creating element "pipeline"
0:00:00.030919354  3744      0x218ee00 INFO            GST_PIPELINE grammar.y:596:gst_parse_perform_link: linking some pad of GstRTSPSrc named rtspsrc0 to some pad of GstRtpH264Depay named rtph264depay0 (0/0) with caps "(NULL)"
0:00:00.030941924  3744      0x218ee00 INFO        GST_ELEMENT_PADS gstutils.c:1633:gst_element_link_pads_full: trying to link element rtspsrc0:(any) to element rtph264depay0:(any)
0:00:00.030978627  3744      0x218ee00 INFO        GST_ELEMENT_PADS gstelement.c:913:gst_element_get_static_pad: no such pad 'stream_%u' in element "rtspsrc0"
0:00:00.030990780  3744      0x218ee00 INFO        GST_ELEMENT_PADS gstutils.c:1125:gst_element_get_compatible_pad:<rtspsrc0> Could not find a compatible pad to link to rtph264depay0:sink
0:00:00.030997839  3744      0x218ee00 INFO                 default gstutils.c:2022:gst_element_link_pads_filtered: Could not link pads: rtspsrc0:(null) - rtph264depay0:(null)
[... cut ...]
0:00:03.809631488  3744      0x218ee00 INFO        GST_ELEMENT_PADS gstelement.c:785:gst_element_remove_pad:<rtph264depay0> removing pad 'sink'
0:00:03.809640296  3744      0x218ee00 INFO        GST_ELEMENT_PADS gstelement.c:785:gst_element_remove_pad:<rtph264depay0> removing pad 'src'
0:00:03.809646286  3744      0x218ee00 INFO         GST_REFCOUNTING gstelement.c:3040:gst_element_dispose:<rtph264depay0> 0x22ca090 parent class dispose
0:00:03.809659486  3744      0x218ee00 INFO         GST_REFCOUNTING gstelement.c:3071:gst_element_finalize:<rtph264depay0> 0x22ca090 finalize
0:00:03.809663753  3744      0x218ee00 INFO