awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #194: kinesis video stream producer test tool Segmentation fault

jiangxiaorong, 2019-03-15:
Dear:
I am trying to port kvs sdk into an embeded IP camera. 

When I run the test tool which is output of kvs sdk producer library, I meet a segmentation fault, before run the test tool my camera have 30Mbytes free memory and I have set TEST_STORAG_SIZE_IN_BYTES to 8Mbytes.

Below is test log, can U help me find out root cause?
[kinesis_video_stream.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/2970705/kinesis_video_stream.txt)

MushMal, 2019-03-15:
@jiangxiaorong is there a way you could include the crash dump or stack during the crash? Even tough the stream is not available in your case this shouldn't lead to a crash. Also, could you please include some more information about the following:

* Type of the hardware and whether unaligned memory access is allowed.
* OS version
* Used compiler
* Whether other tests are passing or not 

You could list the test using:

`./start --gtest_list_tests`

and execute a test using

`./start --gtest_break_on_failure --gtest_filter=ProducerApiTest.create_produce_stream`

Make sure you specify the access key and secret key env variables like

```
export AWS_ACCESS_KEY_ID=_<access_key>_
export AWS_SECRET_ACCESS_KEY=_<secret key>_
```

jiangxiaorong, 2019-03-16:
@MushMal Thanks for your response. 
I am trying to get crush dump from my camera.

- Type of the hardware and whether unaligned memory access is allowed.
My processor is ARM926EJ-S rev 5, CPU architecture is 5TEJ, and unaligned memory access is **not** allowed.
- OS versio
Linux 3.2.7
- Used compiler
arm-buildroot-linux-uclibcgnueabi-g++ (Buildroot) 4.9.2
- Whether other tests are passing or not
Only exceed_max_track_count can pass, segment fault happen on all other tests.

MushMal, 2019-03-18:
@jiangxiaorong thanks for the response. The core of the codebase is tightly packed using #pragma pack. This shouldn't be an issue as the platform specific compiler/generator will properly simulate the aligned access or simply drop the packing.

We had seen issues like these when dependent libraries were stale. Could you please ensure a clean compile and try to capture the stack trace so we can pinpoint the issue.

jiangxiaorong, 2019-03-20:
@MushMal Thank U! I will try to capture the stack trace. I can share some debug log to you firstly, I am debuging on sdk version 1.5.0 (our platform use this version). Actually, I meet 2 errors on this version:

First one is "ERROR - Unable to create Kinesis Video stream. test error status: 0x4", after debuging I found this error happened when running statement **CHK(*pSize >= packagedSize, STATUS_NOT_ENOUGH_MEMORY)** of function **mkvgenGenerateTag** defined in file **MkvGenerator.cpp**. 

The root cause is eosTracker.size is assigned to 535 in function generateEosMetadata which defined in file Stream.cpp, but it turn to 23 in function mkvgenGenerateTag. Below is my debug log:
    "2019-03-19 20:14:28 [1073894208] [DEBUG]: eosTracker.size = 535, address of eosTracker.size = 
    b1623
    [DEBUG]: pSize = 23, address is b1623
    ERROR -  Unable to create Kinesis Video stream. test Error status: 0x4
    2019-03-19 20:14:28 [1073894208] INFO - Freeing Kinesis Video Stream test
    2019-03-19 20:14:28 [1073894208] INFO - "

The type of eosTracker.size is UINT32, it's address should be multiple of 4, but from the debug log it's not, so I think memory is not aligned. After adding "#pragma pack(push, 4)" at the begining where define _CurrentEosTracker and structure _Kinesis_Video_Stream in file Stream.h the issue is fixed

Second one is segment fault. I do not have debug log on this issue.

MushMal, 2019-03-20:
Thanks for the information @jiangxiaorong ! I am looking at the codebase (the new one) and it's not immediately clear why the alignment is an issue in the function - perhaps I am missing a point.

535=0x217 and 23=0x17 so there is certainly a pointer mismatch or something calling the mkvgenGenerateTag function. Unless I am not understanding the issue the first issue looks more like a packing mismatch..? 

You could try to comment
`#pragma pack(push, include, 1) // for byte alignment`
and the corresponding pragma pop
in the main Include.h file in src/client/include/com/amazon..../Include.h

jiangxiaorong, 2019-03-25:
@MushMal Thank you, after commenting #paragma pack(push, include, 1) in all source code files, I can run the test tool, but after a while, the segment fault happened, attached is gdb debug information after analysis core dump
[gdb-core-debug.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3001467/gdb-core-debug.txt)

MushMal, 2019-03-25:
@jiangxiaorong, it seems that your platform does allow unaligned access - the crash you've captured is not due to the processor unaligned memory access but rather a mis-match of packing order.

The content store which is the backing storage for the buffered data on the device assumes tight packing. 

I think the original issue you had is due to the mismatch of packing rather than unaligned access. Could you please perhaps include how you are integrating your code? If you are integrating using the C++ client it's possible than the `DefaultCallbackProvider::describeStreamHandler` in DefaultCallbackProvider doesn't package the structures tightly.

Could you please try to run the software under the GDB without changing the packing?

jiangxiaorong, 2019-03-26:
@MushMal  Thank you! The root cause is our CPU do not support unaligned memory access, after set CPU to support unaligned memory access, KVS SDK can run on my platform, but CPU performance is very bad, it need about 60% CPU resource. Can you tell me why you must pack the structures tightly?

MushMal, 2019-03-26:
@jiangxiaorong thanks for the update. The impact on the aligned/unaligned access from the perspective of the SDK is minimal even when we are adapting Annex-B to AvCC format. The reason for tight packing is due to the fact that the SDK code is very optimized and doesn't access memory much using non-mem copy methods. The tight packing is useful in this case so we can store more data more optimally.
You can try to capture the perf traces to see what other piece of code runs the CPU hot - if you are doing encoding then indeed it might be CPU intensive.
***Not tried*** you could also change client/src/Include_i.h LN: 122 from
`#define MEMORY_BASED_HEAP_FLAGS     FLAGS_USE_AIV_HEAP`
to
`#define MEMORY_BASED_HEAP_FLAGS     FLAGS_USE_SYSTEM_HEAP`

This will use processes system heap which in this case will map to MEMALLOC which is by default mapped to malloc. You could then try to remove the tight packing.

MushMal, 2019-04-23:
Resolving for now

closed 2019-04-23 by MushMal