awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #197: pipeline could not be constructed: no element "rtmpsrc"

cdebost, 2019-03-27:
`rtmpsrc` is a pipeline element included in the "bad" gstreamer plugins.

Normally, a setup as follows in enough to get gstreamer to recognize the plugin:
```
apt-get install libgstreamer1.0-0 gstreamer1.0-plugins-base gstreamer1.0-plugins-bad gstreamer1.0-tools gstreamer1.0-x
gst-launch-1.0 rtmpsrc  # rtmprc is recognized as a pipeline element
```

But when doing a `min-install-script` installation (where libraries like gstreamer and its plugins are installed beforehand w/ `apt-get`), `gst-launch-1.0 rtmpsrc` reports that "rtmpsrc" does not exist.

Even with the min-install-script, all of the gst plugins appear to be moved into kinesis-video-native-build/downloads, and following that gstreamer is no longer able to recognize rtmpsrc.

I noticed that the libgstrmp library files are missing from kinesis-video-native-build/downloads/local/lib, but manually compiling that plugin and moving build artifacts into the local lib directory doesn't fix the issue. My understanding is that the min-install-script should not be installing any plugins manually, as they were already installed with apt-get.

This is a very similar to https://forums.aws.amazon.com/message.jspa?messageID=859593, where the OP was looking to stream RTMP into Kinesis and was advised to use the C++ producer and the rtmp plugin, but it seems they hit a dead end.

unicornss, 2019-04-01:
Hi @cdebost ,

You can use the Dockerfile and install-script (minimal) [custom-gstreamer-pipelines-with-kvs.tar.gz](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3029136/custom-gstreamer-pipelines-with-kvs.tar.gz) as a _reference_  for building the GStreamer plugin RTMPSRC and your custom pipelines.

The **Dockerfile** installs all the GStreamer plugins using apt-get-install and uses the **install-script** for building Kinesis Video Streams producer, gstkvssink, libcurl, libssl and log4cplus.

 Please note that the install-script sets the PKG_CONFIG_PATH as follows
`export PKG_CONFIG_PATH=/usr/lib/x86_64-linux-gnu/pkgconfig/:/opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/pkgconfig/` for searching the system install locations for the GStreamer plugins and gstkvssink.

```
~ docker images
REPOSITORY            TAG                 IMAGE ID            CREATED             SIZE
rtmpsrcelementcheck   latest              4a74a60f5252        2 days ago          1.03GB
ubuntu                18.04               94e814e2efa8        2 weeks ago         88.9MB

➜  ~ docker run -it 4a74a60f5252
root@6fd39edec343:/opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build# source set_kvs_sdk_env.sh
root@6fd39edec343:/opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build# gst-inspect-1.0 rtmpsrc
Factory Details:
  Rank                     primary (256)
  Long-name                RTMP Source
  Klass                    Source/File
  Description              Read RTMP streams
  Author                   Bastien Nocera <hadess@hadess.net>, Sebastian Dröge <sebastian.droege@collabora.co.uk>

Plugin Details:
  Name                     rtmp
  Description              RTMP source and sink

root@6fd39edec343:/opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build#gst-inspect-1.0 |grep soup
soup:  souphttpsrc: HTTP client source
soup:  souphttpclientsink: HTTP client sink

root@6fd39edec343:/opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build#gst-inspect-1.0 |grep ximage
ximagesrc:  ximagesrc: Ximage video source
ximagesink:  ximagesink: Video sink
root@6fd39edec343:/opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build#

root@6fd39edec343:/opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-vid
eo-native-build#gst-inspect-1.0 |grep kvssink
kvssink:  kvssink: KVS Sink
```

The GStreamer samples provided with KVS ingest includes rtspsrc (IP camera), filesrc, webcam and USB camera. If you have a specific requirement other than these video sources, please send email to kinesis-video-support@amazon.com with your use case.

Thanks

chehefen, 2019-04-01:
Can you try renaming kinesis-video-native-build/downloads to something else like kinesis-video-native-build/downloadsCustom. Then run min-install-script again and check if `gst-launch-1.0 rtmpsrc` works? I think because gstreamer stores its cache in common location, the existing cache caused `gst-launch-1.0` to use the binary in kinesis-video-native-build/downloads.

unicornss, 2019-04-02:
Closing the issue as the question has been answered. 
Please re-open or create a new one if additional support is required. 

Thanks

closed 2019-04-02 by unicornss

cdebost, 2019-04-05:
@unicornss Ah, I should have looked for where that variable was declared. My issues likely stemmed from a gotcha with the `rtmp` plugin: the plugin will be skipped if you forget to install librtmp first, e.g. `apt-get install librtmp-dev`. It wasn't until I combed through the `install-script` output that I realized the rtmp plugin wasn't installed because librtmp couldn't be found.

@chehefen Good point, but I was running the installation in a Dockerfile, which wouldn't keep the old cache around. I think I was mixing up my attempts with `min-install-script` vs `install-script`, a `min-install-script` installation should not change what `gst-launch-1.0` points to, unless there is leftover cache like you mentioned.

For posterity, I'll add my Dockerfile that I ended up with so that both `rtmpsrc` and `kvssink` are available to `gst-launch-1.0`, in case someone else is looking to feed an rtmp source into Kinesis like I was. Thanks for your time!

```Dockerfile
FROM ubuntu:17.10

RUN apt-get update
RUN apt-get install -y git cmake g++ libssl-dev libcurl4-openssl-dev liblog4cplus-1.1-9 liblog4cplus-dev
RUN apt-get install -y libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-base-apps gstreamer1.0-x
RUN apt-get install -y gstreamer1.0-plugins-bad gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly gstreamer1.0-tools

WORKDIR /opt/
RUN git clone https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git
WORKDIR /opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/
RUN chmod a+x ./min-install-script
RUN ./min-install-script

RUN cp libgstkvssink.so /usr/lib/x86_64-linux-gnu/gstreamer-1.0/
```

Unfortunately, although I was able to install everything I needed, I don't have enough experience with gstreamer to know what pipeline I would need to convert an rtmpsrc into something kvssink can take, but if I figure out a working pipeline I would be happy to post an example to document how to do rtmp -> kinesis.

nmarus, 2020-02-11:
@cdebost Were you ever able to figure out a working pipeline for ramp -> kinesis ?

nmarus, 2020-02-12:
@cdebost IO was able to get this figured out and thought I'd share if you are still looking for a solution. The missing part for me was that gstreamer can only pull RTMP streams and my source could only push RTMP streams. I used SRS to bounce the RTMP stream off of. 

```
# launch srs container
docker run -d -p 1935:1935 -p 1985:1985 -p 8080:8080 ossrs/srs:3
```

Then following your ubuntu setup above, the following pulls the stream from SRS and pushes it to Kinesis Video:

```
gst-launch-1.0 -v \
  rtmpsrc name=rtmpsrc blocksize=1024 do-timestamp=true location="rtmp://${SRS_RTMP_HOST}/live/${AWS_KINESIS_STREAM_NAME}" \
  ! flvdemux name=demux demux.video \
  ! h264parse \
  ! video/x-h264, format=avc,alignment=au \
  ! kvssink stream-name=${AWS_KINESIS_STREAM_NAME} storage-size=512
```