awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #198: Failed to load plugin 'C:\msys64\mingw64\lib\gstreamer-1.0\libgstx265.dll'

marcisrael, 2019-03-27:
I setup the whole environment on a windows 10 machine inside MinGW (64 bits). It ended with success. However, running any gst commandt (gst-device-monitor-1.0, gst-launch-1.0 or gstkvsplugintest.exe)) leads to the error Failed to load plugin 'C:\msys64\mingw64\lib\gstreamer-1.0\libgstx265.dll'. The DLL is there in the right directory. Thank you for your help.

marc

unicornss, 2019-03-27:
Hi @marcisrael,

Could you please provide more details on the issue you are noticing? Are you planning to use the KVS for sending H265 streams (libgstx265.dll (x265 — x265-based H265 plugins))?

Thanks

marcisrael, 2019-03-28:
Hi. Nope, I'm not planning to send H265 streams, I was trying to stream the feed from a USB Cam hooked onto the PC (It's detected by gst-device-monitor-1.0.). However, when I launch gst-launch-1.0, it stops with the above error. I tried kinesis_video_gstreamer_sample_app.exe and got the same error. Same when I try to stream am H.264 MP4 

gst-launch-1.0 -v  filesrc location="myvideo.mp4" ! qtdemux name=demux ! queue ! h264parse !  video/x-h264,stream-format=avc,alignment=au ! kvssink name=sink stream-name="rekostream" streaming-type=offline demux. ! queue ! aacparse ! sink.

Thanks for your help.

Marc

unicornss, 2019-03-28:
Hi @marcisrael ,

Thanks for the logs and it shows that the fragments got persisted after 07:42:14.  This suggests that the gstreamer pipeine is able to ingest to KVS successfully. 

If you are seeing the "InvalidCodecPrivateDataException" in the console, could you please retry the streaming with a different stream name (e.g. _fluxvideoReko1_ ) to ensure that the stream _fluxvideoReko_ has the correct codec values.  

> 
> [DEBUG][2019-03-28 07:42:20] fragmentAckReceivedHandler invoked
> [DEBUG][2019-03-28 07:42:20] Curl post body write function for stream: fluxvideoReko and upload handle: 2 returned: {"EventType":"PERSISTED","FragmentTimecode":163730,"FragmentNumber":"91343852333181432531331347018955790166615479450"}

> getStreamData(): Indicating an EOS after last persisted ACK is received for stream upload handle 2
[INFO ][2019-03-28 07:42:20] Reported end-of-stream for fluxvideoReko. Upload handle: 2
[DEBUG][2019-03-28 07:42:20] Wrote 0 bytes to Kinesis Video. Upload stream handle: 2
[DEBUG][2019-03-28 07:42:20] terminate curl for handle 2
[DEBUG][2019-03-28 07:42:20] streamClosedHandler invoked for upload handle: 2
[DEBUG][2019-03-28 07:42:20] Close connection: fluxvideoReko, upload handle: 2, is EOS: 1, is in Shutdown: 0
[DEBUG][2019-03-28 07:42:20] Kinesis Video stream fluxvideoReko is Closed.
[INFO ][2019-03-28 07:42:21] **Sending eos**

In real-time mode  (e.g. webcam or usbcam) the video, GStreamer source element provides the  data at the video playback/read speed; but in the file upload (aka offline upload mode), GStreamer **filesrc** element sends the data to KVS as fast as it could read from the file and KVS SDK transmits it to the cloud at the network speed.   Hope this explains the reason why you are seeing the end of stream (EOS) within 20 seconds.

However, there were few dropped frame errors in the beginning of the streaming.  
> viewItemRemoved(): Reporting a dropped frame/fragment.
[DEBUG][2019-03-28 07:42:00] droppedFrameReportHandler invoked
[WARN ][2019-03-28 07:42:00] Reporting dropped frame. Frame timecode 1245050000
[INFO ][2019-03-28 07:42:00] 

This could happen if the the storage size is full ; the default  storage-size=128 but you can set it to higher size if your file size is large.  **GStreamer KVS Sink Element Parameters Reference ** https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin-parameters.html

Thanks

chehefen, 2019-04-01:
Hi, can you try
```
gst-launch-1.0 -v filesrc location="myvideo.mp4" ! qtdemux name=demux ! queue ! h264parse ! kvssink name=sink stream-name="fluxvideoReko" aws-region="eu-west-1" streaming-type=offline demux. ! queue ! aacparse ! sink.
```
If you are uploading a file, streaming-type=offline is required.

unicornss, 2019-04-02:
Closing the issue for now as the question has been answered. Please re-open or create a new one if additional support is required.
Thanks

closed 2019-04-02 by unicornss