awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #199: Is it possible to produce h265 video stream?

ib1368, 2019-03-27:
I tried to use the video sample app to stream h265 encoded data, but seems it crashes after not figuring out the codec private data. [In fact it is null](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/c279068b5d32735db4dc8a266751024489bf9b9b/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L343).

Is it possible to stream h265 encoded data? I am receiving encoded data using RTP protocol.

Tannz0rz, 2019-03-27:
I have no affiliation with the KVS team, so I cannot attest to whether or not it supports H265, but the issue at face value seems to be on GStreamer's end. The structure is inevitably different from H264, so I would initially check what values it contains thusly:

```
    for(guint index = 0; index < gst_structure_n_fields(gststructforcaps); index++) {
        g_print(gst_structure_nth_field_name(gststructforcaps, index));
    }
```

If that fails to yield useful information, you should look into directly parsing the codec data yourself:
https://github.com/agx/gst-plugins-bad/blob/master/gst/videoparsers/gsth265parse.c#L996

Good luck.

MushMal, 2019-03-27:
@ib1368 indeed, the main issue is how to get theCPD for h265 in it's entirety so it can be passed into KVS SDK core. The core itself can parse the h265 CPD bits which is SPS, PPS, VPS but they are packaged entirely differently.

ib1368, 2019-03-29:
Thanks.
- Is it possible to stream corrupted h265 data or if we persist the data there, it means it was successful?
- How can we verify the stream persisted well?

ib1368, 2019-03-29:
What should be CODEC ID for H265? Based on [this line](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/c279068b5d32735db4dc8a266751024489bf9b9b/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L51), it is "V_MPEG4/ISO/AVC" for H264.

closed 2019-03-29 by ib1368