awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #200: amazonlinux-docker build fails on macOS.

msjoshi, 2019-04-03:
I'm trying to build the docker image using the following command. 

```
docker build --no-cache -t rtspdockertest -f docker_native_scripts/amazonlinux-docker/Dockerfile .
```

It fails with the following error. 

```
checking for flex... flex
checking whether lex is flex... no
checking lex output file root... configure: error: cannot find output from flex; giving up
The command '/bin/sh -c ./install-script -j 4 -d' returned a non-zero code: 1
```

Anyone seen this issue?

unicornss, 2019-04-03:
The AMI Linux KVS docker build seems successful without any issues.
```
$ docker build -t amidocker .
Sending build context to Docker daemon  3.584kB
Step 1/14 : FROM amazonlinux:2
2: Pulling from library/amazonlinux
f64ae36417d7: Pull complete
Digest: sha256:d4a4328d679534af47c7a765d62a9195eb27f9a95c03213fca0a18f95aa112cd
Status: Downloaded newer image for amazonlinux:2
 ---> 01da4f8f9748
Step 2/14 : RUN yum update -y && 	yum upgrade -y && 	yum install -y autoconf && 	yum install -y automake && 	yum install -y bison && 	yum install -y bzip2 && 	yum install -y cmake && 	yum install -y curl && 	yum install -y diffutils && 	yum install -y flex  && 	yum install -y gcc  && 	yum install -y gcc-c++ &&   	yum install -y git && 	yum install -y gmp-devel && 	yum install -y libffi && 	yum install -y libffi-devel && 	yum install -y libmpc-devel && 	yum install -y libtool && 	yum install -y m4 && 	yum install -y mpfr-devel && 	yum install -y pkgconfig && 	yum install -y vim && 	yum install -y wget && 	yum install -y make && 	yum install -y xz
 ---> Running in b71def34a9e9
Loaded plugins: ovl, priorities
Resolving Dependencies
--> Running transaction check

```
\:
\:
```
Removing intermediate container 46768f0a45cd
 ---> 78343be34365
Step 14/14 : WORKDIR /opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/
 ---> Running in b01fd2ab95af
Removing intermediate container b01fd2ab95af
 ---> b5c4425fbe18
Successfully built b5c4425fbe18
Successfully tagged amidocker:latest

```

```
➜  tmpdocker docker images
REPOSITORY            TAG                 IMAGE ID            CREATED             SIZE
amidocker             latest              b5c4425fbe18        5 seconds ago       2.68GB
amazonlinux           2                   01da4f8f9748        4 weeks ago         162MB
```

msjoshi, 2019-04-04:
@unicornss 

Issue was on our end. We modified the Dockerfile for our CI pipeline and its fixed.

closed 2019-04-04 by msjoshi