awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #202: Sending a bunch of JPEGs each second to kinesis stream

arnamoy10, 2019-04-17:
Hello;

I have an application that produces a bunch of JPEG images every second.  I want to use the C++ producer library to send these bunch of JPEG images every second to kinesis stream.  Can anyone please give me a pointer to how this can be achieved?  I have seen the demo application with video file but that is not helpful for dealing with JPEGs.

Thanks a lot for your help.

MushMal, 2019-04-19:
@arnamoy10 you might try to use MPEGJ format. What this really is a stream with all I-frames and all are JPEGs. This is popular for small density low frame rate streams.

In order to process it, you might want to use the following StreamDefinition sample

```
auto stream_definition = make_unique<StreamDefinition>(stream_name,
                                                               hours(RETENTION_HOURS),
                                                               nullptr,
                                                               "",
                                                               STREAMING_TYPE_REALTIME,
                                                               "video/x-matroska",
                                                               milliseconds(max_stream_latency_ms),
                                                               seconds(2),
                                                               milliseconds(1),
                                                               true,
                                                               true,
                                                               true,
                                                               true,
                                                               true,
                                                               true,
                                                               NAL_ADAPTATION_FLAG_NONE,
                                                               25,
                                                               4 * 1024 * 1024,
                                                               seconds(buffer_duration_seconds),
                                                               seconds(20),
                                                               seconds(50),
                                                               "V_MS/VFW/FOURCC",
                                                               nullptr,
                                                               0,
                                                               MKV_TRACK_INFO_TYPE_VIDEO);
```

MushMal, 2019-07-24:
Resolving

closed 2019-07-24 by MushMal