awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #203: KinesisVideoStream->putFrame() doesn't return

ameyat2812, 2019-05-01:
Hi @MushMal @zhiyua-git 

I am using Gstreamer streaming thread to get video frames in appsink and use the same thread to push these video frames to Kinesis by calling putFrame. Something similar to what you guys suggest in gstreamer sample app. I am observing that occasionally, the putFrame API doesn't return making my gstreamer pipeline to go unresponsive.

I haven't seen any ERRORS/WARN on my stream when it got locked however in the past, I have seen STATUS_INVALID_PARSED_ACK_TYPE being thrown in the past.  [Check this issue] (https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/135)

I am currently using 1.7.7 version of SDK.

Regards,
Ameya

MushMal, 2019-05-01:
@ameyat2812 KVS SDK has two streaming modes. In the offline mode the SDK will block for the availability before proceeding with a putFrame call. The availability comes from storage or temporal availability. Could you please clarify a little more your scenario? If you intend to stream real-time then your streaming type (STREAMING_TYPE enum) in StreamCaps should be STREAMING_TYPE_REALTIME.

ameyat2812, 2019-05-02:
Here are the details of how I am using it

1. It is a simple NVR kinda use case where the NVR talks to multiple cameras. These cameras are context separated.
2. I am creating an instance of KinesisVideoProducer with all the necessary callback providers and keeping it handy in my wrapper class per camera. And I also have multiple streams to upload to per camera. These streams will be created, started, pushed frames to and stopped as and when needed. (mind that I am not using kvssink element and tapping into SDK myself)
3. There is a stream (let us call it an Activity stream) which uploads events of interest to kinesis. The Activity stream upload pipeline looks something like this

rtspsrc ! rtph264depay ! video/x-h264,stream-format=avc,alignment=au  ! appsink ! <decision making > ! create and start if not created and upload to activity stream

4. All the streams per camera share the same producer instance and yes they are configured to be STREAMING_TYPE_REALTIME.

5. I do have mutex protection around the stream handles so that no other thread can call any of KinesisVideoStream member while another call is in progress.

The above-mentioned setup works 99% of the time and I also have ERROR handling to recreate streams for fatal errors and often call KinesisVideoStream->resetConnection() on stale connection and similar warning messages. However, the putFrame() blocking issue happens every now and then and that makes my whole pipeline go unresponsive.

MushMal, 2019-05-02:
@ameyat2812 thanks for the info. In case of the realtime streaming the PIC itself doesn't block (modulo the low-level interlocking for multi-stream scenario but in this case the operations are prompt so it shouldn't be an issue). This said, the PIC uses callbacks for various operations. All of these callbacks are intended to be "prompt" - aka - they should ideally complete their execution within the thread quanta or spin up a separate execution unit (another thread) to run the processing. For example, our networking API calls are inherently asynchronous. The result of the call is fed in after the call completes but the callback is not blocked.

Can you validate whether any of your callbacks can block? Including the authentication callbacks.

ameyat2812, 2019-05-06:
@MushMal Thanks for that insight.

I added some extra logs in the callback handling functions and can confirm that all the callbacks returned back to SDK.

I have a logic to map callback events to messages which are added to a queue in my wrapper class. I process and take appropriate actions mainly before I push any new frames to the stream. So here is an example scenario,

1. Create Stream
2. Start Stream - All the kinesis video callbacks are converted to respective messages and added to the queue at this point onwards
3. Push Video (do it in the loop)
- Check for any error/connection stale messages in the queue
- Recreate the stream on Error and reset the connection on stale connections
- if stream ready then putFrame()

The locking issue is happening very frequently now, especially after I have added multiple cameras to the NVR. During busy hours a lot of the cameras detect activity and start pushing to Kinesis. I have also observed that sometimes I do get "STATUS_ACK_ERR_INVALID_MKV_DATA" error (I do return the error reporting callback) and then I don't seem to receive any Frag Acknowledgment back from SDK for a while. As I am still pushing to stream, the putFrame() gets locked and never returns.

BTW. I am not taking any actions for ERRORs STATUS_ACK_ERR_INVALID_MKV_DATA to STATUS_ACK_ERR_UNKNOWN_ACK_ERROR (0x5200005e to 0x5000006F) which also includes STATUS_ACK_ERR_INVALID_MKV_DATA error.

Looking forward to your response.

MushMal, 2019-05-06:
@ameyat2812 this sounds like a reasonable approach however it's extremely difficult to pinpoint the issue without any logs. 
Could you please first of all eliminate the CPU load as a factor. Simply check to see if your threads are starving. I highly doubt that this would be a root cause as the SDK is very lightweight and it sounds like you are blocked for a long time but nevertheless.

In real-time mode the SDK doesn't block at all and this is why I am puzzled. Unless it blocks on the callbacks there is very little processing that it does. If the networking is an issue it will still package and put the frame even if it means evicting the oldest frame in the stream (causing frame dropped notification callback). If it still can't put the frame then the putFrame call will fail with an appropriate error.

I would like to get more information on the blocked thread. Does it ever return?

ameyat2812, 2019-05-07:
Hi @MushMal,

I was able to actually something similar locally and I have sent logs to kinesis-video-support@amazon.com. Please have a look at it and let me know what you think.

MushMal, 2019-05-07:
@ameyat2812 would you mind forwarding again? We don't seem to have received it.

ameyat2812, 2019-05-16:
@MushMal the issue turned out to be because of the way I was using putFrame(). I had a lock around putFrame() call for the streamReady mutex. This was to check if the stream is ready before putting any frames into it. But that seemed to cause a deadlock if the stream would run into errors triggering multiple callbacks, sometimes on the same thread.

I have removed the lock and things are flowing again. Thanks a ton for your help.

closed 2019-05-16 by ameyat2812