awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #204: Unable to stream of 1080p 30FPS

pavankuppa, 2019-05-04:
AWS Producer Library C++ using gstreamer

gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=1920,height=1080,framerate=30/1 ! x264enc  bframes=0 key-int-max=45 bitrate=2048 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au,profile=baseline! kvssink stream-name="xxxx" access-key="xxxx" secret-key="xxxxxxx" aws-region="xxxx"

pavankuppa, 2019-05-04:
If i change the FPS to 4 then only it was working, but my camera support 1080p with 30FPS

MushMal, 2019-05-05:
Hi @pavankuppa 

Could you please share a little more information about the device and the camera? General info on your scenario would be helpful. 

Moreover, could you please include the logs so we can help investigating.

pavankuppa, 2019-05-08:
Hi @MushMal 
PFA, which have default(gst-launch-1.0) and aws(gst-launch-1.0)
#### Using AWS
[aws.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3157819/aws.txt)
[aws_kinesis_1080p_4fps.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3157830/aws_kinesis_1080p_4fps.log)
[aws_kinesis_1080_15_fps.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3157831/aws_kinesis_1080_15_fps.log)

### Default
[default.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3157820/default.txt)

### Chrome Media
[chrome-media-internals.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3157826/chrome-media-internals.txt)

chehefen, 2019-05-08:
@pavankuppa Can you remove the kinesis-video-native-build/downloads directory and use min-install-script to build the plugin? min-install-script will build it against you default gstreamer. make sure the following dependencies are installed before running min-install-script
```
libssl-dev libcurl4-openssl-dev liblog4cplus-1.1-9 liblog4cplus-dev
libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-base-apps
gstreamer1.0-plugins-bad gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly gstreamer1.0-tools
```

pavankuppa, 2019-05-09:
@chehefen Thanks for your info. It is working

MushMal, 2019-05-09:
Resolving. Please re-open if the issue persists.

closed 2019-05-09 by MushMal