awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #205: ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.

Raghava248, 2019-05-07:
Iam trying to stream video through amazon kinesis using gsteamer

 gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name="" storage-size=512 access-key="" secret-key="" aws-region=""

using this command
[image]

[image]e

please help me resloving this issue

unicornss, 2019-05-07:
Hi @Raghava248  

Could you tell us more about the setup ?   Is the log from a laptop with a web camera running Ubuntu?

Thanks

Raghava248, 2019-05-08:
no its not a webcam iam trying to use pc camera @unicornss

MushMal, 2019-05-08:
@Raghava248 could you please clarify. Are you running on a Windows platform? If so, what's the Windows version? You can check out some of the samples for Windows here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-windows-msvc.md

Raghava248, 2019-05-08:
No iam using ubuntu 16.04 @MushMal

unicornss, 2019-05-08:
@Raghava248 

The command you are using is referring to video source from a media device index 0 **device=/dev/video0** .  

```
gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name="" storage-size=512 access-key="" secret-key="" aws-region=""
```

Could you try with the `device=/dev/video1` 

For more details how to determine the media device available in your system please check:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#discovering-available-devices

Thanks

chehefen, 2019-05-08:
@Raghava248 can you give us the output from running `gst-device-monitor-1.0`?

Raghava248, 2019-05-09:
[image]
[image]
Thank you
@chehefen

Raghava248, 2019-05-09:
@unicornss 
Yeah i tried with device=/dev/video1
But it is giving some warning
(gst-plugin-scanner:12664): GStreamer-WARNING **: Failed to load plugin '/home/rohan/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libfl.so': /home/rohan/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libfl.so: undefined symbol: yylex
[image]
[image]

Thank you

chehefen, 2019-05-09:
@Raghava248 your camera doesnt support outputting h264 encoded video directly. You need to encode the stream with a h264 encoder. Can you try this pipeline?
```
gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=512 ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name="" storage-size=512 access-key="" secret-key="" aws-region=""
```

Raghava248, 2019-05-10:
I tried the you said was able to write some bytes of data into kinesis video steam but this error popped
and Iam unable to see the video in aws kinesis video console
Iam getting this error in between 
[image]
@chehefen 
please help
thank u

MushMal, 2019-05-10:
@Raghava248 the error you are getting indicates that your fragment duration is over the max which is 10 seconds. Could you please configure your encoder to produce less than 10 second GoPs (Group of Pictures). For example. Many encoders have a setting to force output a I-frame at a certain intervals. If you have 30 FPS encoding then you can configure the encoder to produce an I-frame every 60 frames which would result in 2 second fragmentation.

chehefen, 2019-05-10:
@Raghava248 you can reduce the gop size by reducing key-int-max like the following pipeline
```
gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! x264enc bframes=0 key-int-max=30 bitrate=512 ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! kvssink stream-name="" storage-size=512 access-key="" secret-key="" aws-region=""
```

Raghava248, 2019-05-11:
Yeah I am able to resolve the issue by following your instructions @chehefen  @MushMal  thank u
but at the begining of the execution Iam getting this warning please help me with this
(gst-plugin-scanner:6451): GStreamer-WARNING **: Failed to load plugin '/home/rohan/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libfl.so': /home/rohan/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libfl.so: undefined symbol: yylex
I had installed

chehefen, 2019-05-13:
are you viewing in the correct region? By default the plugin streams to us-west-2.

Raghava248, 2019-05-14:
Yeah thank you I solved the issue but Iam unable to stream any video in kinesis video stream
@chehefen 
Thank you

chehefen, 2019-05-14:
@Raghava248  what do you mean that you are unable to stream any video in kinesis video stream? Can you be more precise about the issue? Are you seeing any errors from the producer? or you are not able to view the stream in the console? or something else.

Raghava248, 2019-05-14:
Yeah I solved the issue and I am able to stream video in kinesis(console). Is there any process to dowload the streamed video from amazon kinesis

closed 2019-05-16 by Raghava248

Raghava248, 2019-05-18:
Setting up for KVS Producer SDK installation
Include building GStreamer extra plugins= Y
Checking openssl at /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libssl.a
Checking curl at /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libcurl.a
Checking log4cplus at /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/liblog4cplus.dylib/.so
Checking gettext at /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libgettextlib.dylib/.so
Checking libffi at /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libffi.a
Checking pcre at /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libpcre.dylib/.so
Checking zlib at /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libz.dylib/.so
Checking iconv at /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libiconv.dylib/.so
Checking util-linux at /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libmount.dylib/.so
Checking glib at /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libglib-2.0.dylib/.so
Checking googletest at /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads
automake not found. Installing
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed
0 1474k 0 4863 0 0 6540 0 0:03:50 --:--:-- 0:03:50 6545
curl: (23) Failed writing body (1949 != 2716)
Iam getting this error while trying to run ./install-script on raspberry pi

str3tch, 2019-09-18:
I also get this warning at the beginning, as the OP reported as well (however there is no mention of a resolution in this ticket)

```
(gst-plugin-scanner:12664): GStreamer-WARNING **: Failed to load plugin '/vagrant/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libfl.so': /home/rohan/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libfl.so: undefined symbol: yylex
```

Does this matter? is it benign?

MushMal, 2019-09-18:
@str3tch the question about the GST yylex warnings is perhaps better addressable in GStreamer forums. Could you please open a separate ticket if you are having any issues streaming to KVS. Please include the details. If you have logs, please send them to kinesis-video-support@amazon.com

imharish63, 2021-08-18:
> @Raghava248 your camera doesnt support outputting h264 encoded video directly. You need to encode the stream with a h264 encoder. Can you try this pipeline?
>
> ```
> gst-launch-1.0 v4l2src [...]

if i use this code for udp with a host address as sink. what shall i use at receiver

MohammedJassem, 2021-09-21:
@Raghava248 I'm actually having the same problem as you and while doing the same steps that you did I can't find a solution so I'm asking all of you for help, thanks

Avinash-Mahto, 2023-08-16:
Hi @chehefen @MushMal could you please help me in this , I am getting the same error which was being received by @Raghava248. I tried all the steps which was discussed above but still no solution.

I am using my Dell laptop based webcam. My host operating system is Windows based and on top of that I installed Oracle Virtual Machine and installed Ubuntu 22 and integrated webcam with Ubuntu machine.

I am getting following error. Error highlighted in yellow lines. Please help.

[image]