awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #206: ERROR occured during ./install-script in raspberry pi for streaming video in amazon kinesis in gstreamer

Raghava248, 2019-05-16:


chehefen, 2019-05-20:
can you get the following packages:
```
$ sudo apt-get update
$ sudo apt-get install libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-base-apps
$ sudo apt-get install gstreamer1.0-plugins-bad gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly gstreamer1.0-tools gstreamer1.0-omx
```
Then run min-install-script? Thanks

chehefen, 2019-05-20:
Can you tell us about your use case of kvs? Thanks

Raghava248, 2019-05-21:
Yeah Thank you I solved the problem but the thing is while doing live stream 
Iam facing delay currently Iam using this command
gst-launch-1.0 -v v4l2src device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc  bframes=0 key-int-max=45 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName storage-size=128 access-key="YourAccessKey" secret-key="YourSecretKey"
In raspberry pi so please help me how to reduce latency

chehefen, 2019-05-23:
what kind of latency are you expecting the achieve? are you using the stream for playback or realtime analysis?

Raghava248, 2019-05-24:
we are using stream for real-time analysis and the delay is around 15 seconds

chehefen, 2019-05-24:
by default the plugin streams to us-west-2. You can add region="blah" the kvssink plugin to stream to other regions.

Raghava248, 2019-05-24:
Iam trying to achieve delay less than 5 seconds during streaming video in kinesis video streamig console
so help me if there are any changes in parameters in this command to reduce the delay
gst-launch-1.0 -v v4l2src device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc bframes=0 key-int-max=45 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName storage-size=128 access-key="YourAccessKey" secret-key="YourSecretKey"

chehefen, 2019-05-24:
can you tell us where are you located? If you are streaming from asia, you should use ap-northeast-1 like such
```
gst-launch-1.0 -v v4l2src device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc bframes=0 key-int-max=45 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName storage-size=128 access-key="YourAccessKey" secret-key="YourSecretKey" aws-region="ap-northeast-1"
```

Raghava248, 2019-05-24:
my location is us-esat-1(virginia)
thank you

chehefen, 2019-05-24:
can you try this pipeline
```
gst-launch-1.0 -v v4l2src device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc bframes=0 key-int-max=30 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name=YourStreamName storage-size=128 access-key="YourAccessKey" secret-key="YourSecretKey" aws-region="us-east-1"
```

chehefen, 2019-05-24:
Can you tell us how are you consuming the video for real-time analytics? Thanks.

Raghava248, 2019-05-25:
so Iam planning that video stream to store in aws s3 and the use it for further processing

MushMal, 2019-05-28:
@Raghava248 thanks for the question. I assume you are observing the "latency" by viewing the console playback? Just a note that the console playback is not a "real-time" playback. It's a fragment-based playback which has to get the time index, retrieve the fragment from the store, generate mpeg so the browsers MSE can render the fragments on your browser. Similar to HLS or MPEG-DASH. The variable factor in this mode is not only the fragment duration as the fragment needs to be fully assembled before the playback is attempted but also the storage and indexing latencies.
In general, you can attempt to lower your fragmentation rate by controlling key-int-max and frame rate but the recommendation is to not produce lesser than a second fragments.
If, however, you would like to have a real-time rendering, you might want to use the GetMedia API. The parser library has a number of examples for the decoding/playback as well as downloading the fragments: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/parser-library.html

closed 2019-05-30 by Raghava248