awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #208: Iam getting an error while running ./install-script for Kinesis video stream in Raspberry pi

Raghava248, 2019-05-18:


zhiyua-git, 2019-05-18:
Hi @Raghava248,

From the log seems either your network was having issue to the link or your disk run out of space causing incomplete download of automake.

As install script will pull down all dependencies and build from scratch, it will take very long time to build on RaspberryPi (~12 hours). If possible, can you remove downloads directory under kinesis-video-native-build directory and follow the steps in [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#install-steps-for-ubuntu-17x-and-raspbian-stretch-using-apt-get) to use min-install-script, this can be built in ~30 minutes.

Thanks,
ZH

Raghava248, 2019-05-18:
Yeah I solved that thank you
The new error Iam getting is
gst-launch-1.0 -v v4l2src device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc  bframes=0 key-int-max=45 bitrate=512 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name= storage-size=512 access-key="" secret-key="" aws-region="" 
Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ./kvs_log_configuration
INFO - Initializing curl.
2019-05-18 17:32:52 [1995522048] INFO - 
createKinesisVideoClient(): Creating Kinesis Video Client
2019-05-18 17:32:52 [1995522048] INFO - 
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
2019-05-18 17:32:52 [1995522048] INFO - 
heapInitialize(): Creating AIV heap.
2019-05-18 17:32:52 [1995522048] INFO - 
heapInitialize(): Heap is initialized OK
2019-05-18 17:32:52 [1995522048] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1558197172639066798 Expiration: 0
2019-05-18 17:32:52 [1995522048] INFO - New credentials expiration is 1558199572
2019-05-18 17:32:52 [1995522048] DEBUG - createDeviceHandler invoked
2019-05-18 17:32:52 [1995522048] INFO - 
createDeviceResultEvent(): Create device result event.
2019-05-18 17:32:52 [1995522048] DEBUG - clientReadyHandler invoked
2019-05-18 17:32:52 [1995522048] DEBUG - Awaiting for the producer to become ready...
2019-05-18 17:32:52 [1995522048] DEBUG - Kinesis Video producer is Ready.
2019-05-18 17:32:52 [1995522048] INFO - try creating stream
2019-05-18 17:32:52 [1995522048] INFO - Creating Kinesis Video Stream Aerologiks2
2019-05-18 17:32:52 [1995522048] INFO - 
createKinesisVideoStream(): Creating Kinesis Video Stream.
2019-05-18 17:32:52 [1995522048] DEBUG - describeStreamHandler invoked
2019-05-18 17:32:52 [1995522048] DEBUG - Awaiting for the stream to become ready...
2019-05-18 17:32:52 [1995522048] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1558197172640586071 Expiration: 0
2019-05-18 17:32:52 [1437906032] WARN - HTTP Error 403: Response: {"Message":"User: arn:aws:iam::393234083493:user/kinesis-video-raspberry-pi-producer is not authorized to perform: kinesisvideo:DescribeStream on resource: Aerologiks2"}
Request URL: https://kinesisvideo.us-east-1.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAVXDUT52SRK3Z6MRW/20190518/us-east-1/kinesisvideo/aws4_request, SignedHeaders=content-type;host;user-agent;x-amz-date, Signature=15292b36146a8367a0befda205a0a467410942bb92536df37dfcca58bf571264
    content-type: application/json
    host: kinesisvideo.us-east-1.amazonaws.com
    user-agent: AWS-SDK-KVS-CLIENT/1.7.9 GCC/6.3.0 Linux/4.9.26-v7+ armv7l
    X-Amz-Date: 20190518T163252Z
2019-05-18 17:32:54 [1437906032] DEBUG - describeStream response: {"Message":"User: arn:aws:iam::393234083493:user/kinesis-video-raspberry-pi-producer is not authorized to perform: kinesisvideo:DescribeStream on resource: Aerologiks2"}
2019-05-18 17:32:54 [1437906032] INFO - Describe stream did not find the stream Aerologiks2 in Kinesis Video (stream will be created)
2019-05-18 17:32:54 [1437906032] INFO - 
describeStreamResultEvent(): Describe stream result event.
2019-05-18 17:32:54 [1437906032] ERROR - Submitting event result for stream: 2191209 failed with: 1375731728
2019-05-18 17:32:54 [1437906032] ERROR - Reporting stream error. Errored timecode: 0 Status: 0x52000010
2019-05-18 17:32:54 [1437906032] INFO - Freeing Kinesis Video Stream Aerologiks2
2019-05-18 17:33:22 [1995522048] INFO - 
freeKinesisVideoStream(): Freeing Kinesis Video stream.
2019-05-18 17:33:22 [1995522048] ERROR - Failed to create Kinesis Video Stream - timed out.
2019-05-18 17:33:22 [1995522048] DEBUG - streamClosedHandler invoked for upload handle: 18446744073709551615
2019-05-18 17:33:22 [1437906032] INFO - Freeing Kinesis Video Stream Aerologiks2
2019-05-18 17:33:22 [1437906032] INFO - try creating stream
2019-05-18 17:33:23 [1995522048] INFO - Creating Kinesis Video Stream Aerologiks2
2019-05-18 17:33:23 [1995522048] INFO - 
createKinesisVideoStream(): Creating Kinesis Video Stream.
2019-05-18 17:33:23 [1995522048] DEBUG - describeStreamHandler invoked
2019-05-18 17:33:23 [1995522048] DEBUG - Awaiting for the stream to become ready...
2019-05-18 17:33:23 [1995522048] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1558197203042688473 Expiration: 0
2019-05-18 17:33:23 [1437906032] WARN - HTTP Error 403: Response: {"Message":"User: arn:aws:iam::393234083493:user/kinesis-video-raspberry-pi-producer is not authorized to perform: kinesisvideo:DescribeStream on resource: Aerologiks2"}
Request URL: https://kinesisvideo.us-east-1.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAVXDUT52SRK3Z6MRW/20190518/us-east-1/kinesisvideo/aws4_request, SignedHeaders=content-type;host;user-agent;x-amz-date, Signature=073d17db83dc8b4aa66e3890c61bd8bd24297f988a8b034a569e6ed9e1bf9ace
    content-type: application/json
    host: kinesisvideo.us-east-1.amazonaws.com
    user-agent: AWS-SDK-KVS-CLIENT/1.7.9 GCC/6.3.0 Linux/4.9.26-v7+ armv7l
    X-Amz-Date: 20190518T163323Z
2019-05-18 17:33:24 [1437906032] DEBUG - describeStream response: {"Message":"User: arn:aws:iam::393234083493:user/kinesis-video-raspberry-pi-producer is not authorized to perform: kinesisvideo:DescribeStream on resource: Aerologiks2"}
[... cut ...]
  File "/usr/lib/python2.7/dist-packages/cryptography/x509/__init__.py", line 7, in <module>
    from cryptography.x509.base import (
  File "/usr/lib/python2.7/dist-packages/cryptography/x509/base.py", line 16, in <module>
    from cryptography.x509.extensions import Extension, ExtensionType
  File "/usr/lib/python2.7/dist-packages/cryptography/x509/extensions.py", line 30, in <module>
    class _SubjectPublicKeyInfo(univ.Sequence):
AttributeError: 'module' object has no attribute 'Sequence'
@zhiyua-git 
Thank u

zhiyua-git, 2019-05-18:
Hi 
You need to set [IAM policy](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-iam.html#how-iam-policies) to avoid “"Message":"User: arn:aws:iam::393234083493:user/kinesis-video-raspberry-pi-producer is not authorized to perform: kinesisvideo:DescribeStream on resource: Aerologiks2"}”

To find the ./kvs_log_configuration, you need to run the command under kinesis-video-native-build directory or copy it to your current directory from kinesis-video-native-build.

Thanks,
ZH

MushMal, 2019-05-18:
Resolving as answered. Please create another issue if you have other questions.

closed 2019-05-18 by MushMal

Raghava248, 2019-05-21:
Thank you @zhiyua-git  I solved it