awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #210: com.amazonaws.kinesisvideo.producer.ProducerException: Failed to put a frame into the stream. StatusCode: 0x3200000c

engr77, 2019-05-28:
I have a RTSP stream that I am receiving locally as a bytebuffer on android platform. I am passing that bytebuffer to Kinesis as:

   public void onFrameDataAvailable(final ByteBuffer data) {
        final long currentTimeMs = System.currentTimeMillis();

        final int flags = FRAME_FLAG_NONE;

        final KinesisVideoFrame frame = new KinesisVideoFrame(
                frameIndex++,
                flags,
                currentTimeMs * HUNDREDS_OF_NANOS_IN_MS,
                currentTimeMs * HUNDREDS_OF_NANOS_IN_MS,
                FRAME_DURATION_1_MS * HUNDREDS_OF_NANOS_IN_MS,
                data);

After about 500+ frames, I always run into error below:

05-27 01:27:21.479  3374  3577 D KinesisVideoAndroidClient: PutFrame index: 503, pts: 15589456414780000, dts: 15589456414780000, duration: 10000, keyFrame: false, flags: 0
05-27 01:27:21.487  3374  3577 D KinesisVideoAndroidClient: Data availability notification. Upload handle: 0, Size: 153106559, Duration 287660000 
05-27 01:27:21.495  1529  1529 D wpa_supplicant: wlan0: Control interface command 'SIGNAL_POLL'
05-27 01:27:21.501  3374  3577 D KinesisVideoAndroidClient: PutFrame index: 504, pts: 15589456415010000, dts: 15589456415010000, duration: 10000, keyFrame: false, flags: 0
05-27 01:27:21.503  1529  1529 D wpa_supplicant: CTRL-DEBUG: global_ctrl_sock-sendto: sock=11 sndbuf=229376 outq=0 send_len=48
05-27 01:27:21.504  3374  3577 D KinesisVideoAndroidClient: Data availability notification. Upload handle: 0, Size: 153452172, Duration 287890000 
05-27 01:27:21.512   633  3582 E C2DColorConvert: unknown format passed for luma alignment number
05-27 01:27:21.516   633  3582 E C2DColorConvert: unknown format passed for luma alignment number
05-27 01:27:21.559  3374  3577 D KinesisVideoAndroidClient: PutFrame index: 505, pts: 15589456415580000, dts: 15589456415580000, duration: 10000, keyFrame: false, flags: 0
05-27 01:27:21.563  3374  3577 D KinesisVideoAndroidClient: Data availability notification. Upload handle: 0, Size: 153797785, Duration 288460000 
05-27 01:27:21.583   633  3582 E C2DColorConvert: unknown format passed for luma alignment number
05-27 01:27:21.594   633  3582 E C2DColorConvert: unknown format passed for luma alignment number
05-27 01:27:21.596   633  3582 E C2DColorConvert: unknown format passed for luma alignment number
05-27 01:27:21.638  3374  3577 D KinesisVideoAndroidClient: PutFrame index: 506, pts: 15589456416380000, dts: 15589456416380000, duration: 10000, keyFrame: false, flags: 0
05-27 01:27:21.643  3374  3577 D KinesisVideoAndroidClient: Data availability notification. Upload handle: 0, Size: 154143398, Duration 289260000 
05-27 01:27:21.719  3374  3577 D KinesisVideoAndroidClient: PutFrame index: 507, pts: 15589456417180000, dts: 15589456417180000, duration: 10000, keyFrame: false, flags: 0
05-27 01:27:21.724  3374  3577 D KinesisVideoAndroidClient: Data availability notification. Upload handle: 0, Size: 154489011, Duration 290060000 
05-27 01:27:21.753  3374  3577 D KinesisVideoAndroidClient: PutFrame index: 508, pts: 15589456417520000, dts: 15589456417520000, duration: 10000, keyFrame: false, flags: 0
05-27 01:27:21.754  3374  3577 W platform-utils: 
05-27 01:27:21.754  3374  3577 W platform-utils: putKinesisVideoFrame(): Failed to put a frame with status code 0x3200000c
05-27 01:27:21.754  3374  3577 D platform-utils: 
05-27 01:27:21.754  3374  3577 D platform-utils: throwNativeException(): Throwing com/amazonaws/kinesisvideo/producer/ProducerException with message: Failed to put a frame into the stream.
05-27 01:27:21.757  3374  3577 E AndroidRuntime: FATAL EXCEPTION: player_main
05-27 01:27:21.757  3374  3577 E AndroidRuntime: Process: co.xx, PID: 3374
05-27 01:27:21.757  3374  3577 E AndroidRuntime: java.lang.RuntimeException: com.amazonaws.kinesisvideo.producer.ProducerException: Failed to put a frame into the stream. StatusCode: 0x3200000c
05-27 01:27:21.757  3374  3577 E AndroidRuntime: 	at co.xx.media.KinesisMediaSource.putFrame(KinesisMediaSource.java:101)
05-27 01:27:21.757  3374  3577 E AndroidRuntime: 	at co.xx.media.KinesisMediaSource.onFrameDataAvailable(KinesisMediaSource.java:94)
05-27 01:27:21.757  3374  3577 E AndroidRuntime: 	at co.xx.media.MediaPlayerExtended.OnVideoRendererFrameAvailable(MediaPlayerExtended.java:274)
05-27 01:27:21.757  3374  3577 E AndroidRuntime: Caused by: com.amazonaws.kinesisvideo.producer.ProducerException: Failed to put a frame into the stream. StatusCode: 0x3200000c
05-27 01:27:21.757  3374  3577 E AndroidRuntime: 	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.putKinesisVideoFrame(Native Method)
05-27 01:27:21.757  3374  3577 E AndroidRuntime: 	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.putFrame(NativeKinesisVideoProducerJni.java:495)
05-27 01:27:21.757  3374  3577 E AndroidRuntime: 	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerStream.putFrame(NativeKinesisVideoProducerStream.java:292)
05-27 01:27:21.757  3374  3577 E AndroidRuntime: 	at com.amazonaws.kinesisvideo.internal.mediasource.ProducerStreamSink.onFrame(ProducerStreamSink.java:52)
05-27 01:27:21.757  3374  3577 E AndroidRuntime: 	at co.xx.media.KinesisMediaSource.putFrame(KinesisMediaSource.java:99)

MushMal, 2019-05-30:
You are getting StatusCode: 0x3200000c

https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html

0x3200000c | STATUS_MKV_LARGE_FRAME_TIMECODE | The max frame timecode was reached. The MKV format uses signed 16 bits to represent the relative timecode of the frame to the beginning of the cluster. The error is generated if the frame timecode cannot be represented. This error indicates either a bad timecode scale selection or the cluster duration is too long, so representing the frame timecode overflows the signed 16-bit space.
-- | -- | --

This is due to the way you are submitting the flags for the frames. In your case, you are not setting the Key-frame flag indicator. You need to somehow determine whether the submitted frame is a key-frame (Idr-frame for H264) and mark the flag by bitwise OR-ing with FRAME_FLAG_KEY_FRAME

MushMal, 2019-07-24:
Resolving for now. Please reopen if still is an issue

closed 2019-07-24 by MushMal