awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #211: HTTP Error 403: Response: {"message": Signature expired}

h5li, 2019-05-29:
Hello! I have been setting up kinesis video stream these days but I got these bugs. I searched for the GitHub. It seems like nobody else has such an issue. May you help me? Thanks a lot!

I am currently using raspberry pi  and have all below packages installed:

git, cmake, libtool, libtool-bin, automake, bison, g++, pkg-config, flex, openjdk-8-jdk
libssl-dev libcurl4-openssl-dev liblog4cplus-1.1-9 liblog4cplus-dev
libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-base-apps
gstreamer1.0-plugins-bad gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly gstreamer1.0-tools
gstreamer1.0-omx

then I run ./min-install-script to build the kinesis. Then I ran a simple example
./kinesis_video_gstreamer_sample_app <StreamName>

I got the following logs and errors:

[INFO ][2019-02-27 02:50:43] Using region: us-west-2
[INFO ][2019-02-27 02:50:43] Using aws credentials for Kinesis Video Streams
[INFO ][2019-02-27 02:50:43] No session token was detected.
[INFO ][2019-02-27 02:50:43] Initializing curl.
[INFO ][2019-02-27 02:50:43] 
createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2019-02-27 02:50:43] 
heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ][2019-02-27 02:50:43] 
heapInitialize(): Creating AIV heap.
[INFO ][2019-02-27 02:50:43] 
heapInitialize(): Heap is initialized OK
[DEBUG][2019-02-27 02:50:43] Refreshing credentials. Force refreshing: 0 Now time is: 1551235843309381916 Expiration: 0
[INFO ][2019-02-27 02:50:43] New credentials expiration is 1551238243
[DEBUG][2019-02-27 02:50:43] createDeviceHandler invoked
[INFO ][2019-02-27 02:50:43] 
createDeviceResultEvent(): Create device result event.
[DEBUG][2019-02-27 02:50:43] clientReadyHandler invoked
[DEBUG][2019-02-27 02:50:43] Awaiting for the producer to become ready...
[DEBUG][2019-02-27 02:50:43] Kinesis Video producer is Ready.
[DEBUG][2019-02-27 02:50:43] Client is ready
[INFO ][2019-02-27 02:50:43] Creating Kinesis Video Stream KinesisStream
[INFO ][2019-02-27 02:50:43] 
createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2019-02-27 02:50:43] describeStreamHandler invoked
[DEBUG][2019-02-27 02:50:43] Awaiting for the stream to become ready...
[DEBUG][2019-02-27 02:50:43] Refreshing credentials. Force refreshing: 0 Now time is: 1551235843310716291 Expiration: 0
[WARN ][2019-02-27 02:50:43] HTTP Error 403: Response: {"message":"Signature expired: 20190227T025043Z is now earlier than 20190529T174926Z (20190529T175426Z - 5 min.)"}
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAZRAZ6SSTLGQDDTZT/20190227/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;user-agent;x-amz-date, Signature=7a483eb3315d1e705475b9466cdf0f8f09705cef8add927209cc1ed305b77c0c
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS/1.7.9 GCC/6.3.0 Linux/4.14.98-v7+ armv7l
    X-Amz-Date: 20190227T025043Z
[DEBUG][2019-02-27 02:50:43] describeStream response: {"message":"Signature expired: 20190227T025043Z is now earlier than 20190529T174926Z (20190529T175426Z - 5 min.)"}
[INFO ][2019-02-27 02:50:43] Describe stream did not find the stream KinesisStream in Kinesis Video (stream will be created)
[INFO ][2019-02-27 02:50:43] 
describeStreamResultEvent(): Describe stream result event.
[ERROR][2019-02-27 02:50:43] Submitting event result for stream: 25775809 failed with: 1375731728
[ERROR][2019-02-27 02:50:43] Reporting stream error. Errored timecode: 0 Status: 1375731728
[INFO ][2019-02-27 02:51:13] Freeing Kinesis Video Stream KinesisStream
[INFO ][2019-02-27 02:51:13] 
freeKinesisVideoStream(): Freeing Kinesis Video stream.
[DEBUG][2019-02-27 02:51:13] streamClosedHandler invoked for upload handle: 18446744073709551615
[ERROR][2019-02-27 02:51:13] Failed to create Kinesis Video Stream - timed out.
[ERROR][2019-02-27 02:51:13] Failed to initialize kinesis video.
[INFO ][2019-02-27 02:51:13] 
freeKinesisVideoClient(): Freeing Kinesis Video Client
[INFO ][2019-02-27 02:51:13] 
commonHeapDebugCheckAllocator(): Heap is initialized
[INFO ][2019-02-27 02:51:13] 
commonHeapDebugCheckAllocator(): Heap limit: 				134217728
[INFO ][2019-02-27 02:51:13] 
commonHeapDebugCheckAllocator(): Heap size: 				0
[INFO ][2019-02-27 02:51:13] 
commonHeapDebugCheckAllocator(): Number of allocations: 			0
[INFO ][2019-02-27 02:51:13] 
aivHeapDebugCheckAllocator(): Allocated blocks pointer: 			(nil)
[INFO ][2019-02-27 02:51:13] 
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2019-02-27 02:51:13] 
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2019-02-27 02:51:13] 
aivHeapDebugCheckAllocator(): Free blocks pointer: 				0x6d9e5008
[INFO ][2019-02-27 02:51:13] 
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2019-02-27 02:51:13] 
aivHeapDebugCheckAllocator(): Block:	0x6d9e5008		size:	134217687
[INFO ][2019-02-27 02:51:13] 
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2019-02-27 02:51:13] 
heapRelease(): Freeing native heap.

Thank your very much!

chehefen, 2019-05-29:
looking at 20190227T025043Z, it seems that your clock is not in sync. can you run `timedatectl` to sync time and try again?

h5li, 2019-05-30:
Thank you very much for your help. I think I solved this problem but here it rised another issue which it cannot find /dev/device0. I am looking at other issues about this. Thank you very much. #87

chehefen, 2019-05-30:
you can run `gst-device-monitor-1.0` to see what devices are available.

h5li, 2019-05-30:
It returns failed to start device monitor

pi@raspberrypi:/dev $ gst-device-monitor-1.0
0:00:00.000408177 22169   0xae4c00 INFO                GST_INIT gst.c:510:init_pre: Initializing GStreamer Core Library version 1.10.4
0:00:00.000742395 22169   0xae4c00 INFO                GST_INIT gst.c:511:init_pre: Using library installed in /usr/lib/arm-linux-gnueabihf
0:00:00.000828540 22169   0xae4c00 INFO                GST_INIT gst.c:522:init_pre: Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l
0:00:00.002680621 22169   0xae4c00 INFO                GST_INIT gstmessage.c:126:_priv_gst_message_initialize: init messages
0:00:00.006166917 22169   0xae4c00 INFO                GST_INIT gstcontext.c:83:_priv_gst_context_initialize: init contexts
0:00:00.009743838 22169   0xae4c00 INFO      GST_PLUGIN_LOADING gstplugin.c:316:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.010793680 22169   0xae4c00 INFO      GST_PLUGIN_LOADING gstplugin.c:224:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.010935346 22169   0xae4c00 INFO      GST_PLUGIN_LOADING gstplugin.c:226:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.011143471 22169   0xae4c00 INFO            GST_REGISTRY gstregistry.c:1738:ensure_current_registry: reading registry cache: /home/pi/.cache/gstreamer-1.0/registry.arm.bin
0:00:00.079848145 22169   0xae4c00 INFO            GST_REGISTRY gstregistrybinary.c:619:priv_gst_registry_binary_read_cache: loaded /home/pi/.cache/gstreamer-1.0/registry.arm.bin in 0.068556 seconds
0:00:00.080275175 22169   0xae4c00 INFO            GST_REGISTRY gstregistry.c:1594:scan_and_update_registry: Validating plugins from registry cache: /home/pi/.cache/gstreamer-1.0/registry.arm.bin
0:00:00.097371760 22169   0xae4c00 INFO            GST_REGISTRY gstregistry.c:1696:scan_and_update_registry: Registry cache has not changed
0:00:00.097507854 22169   0xae4c00 INFO            GST_REGISTRY gstregistry.c:1773:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.097581500 22169   0xae4c00 INFO                GST_INIT gst.c:720:init_post: GLib runtime version: 2.50.3
0:00:00.097624000 22169   0xae4c00 INFO                GST_INIT gst.c:722:init_post: GLib headers version: 2.50.3
0:00:00.097658947 22169   0xae4c00 INFO                GST_INIT gst.c:723:init_post: initialized GStreamer successfully
Probing devices...

0:00:00.098350248 22169   0xae4c00 WARN                 default gstdevicemonitor.c:460:gst_device_monitor_start:<devicemonitor0> No filters have been set, will expose all devices found
0:00:00.157877386 22169   0xae4c00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstpulse.so" loaded
0:00:00.157985771 22169   0xae4c00 INFO    GST_DEVICE_PROVIDER_FACTORY gstdeviceproviderfactory.c:272:gst_device_provider_factory_get: getting device provider "pulsedeviceprovider"
0:00:00.168585336 22169   0xae4c00 INFO      GST_PLUGIN_LOADING gstplugin.c:842:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstvideo4linux2.so" loaded
0:00:00.168689190 22169   0xae4c00 INFO    GST_DEVICE_PROVIDER_FACTORY gstdeviceproviderfactory.c:272:gst_device_provider_factory_get: getting device provider "v4l2deviceprovider"
0:00:00.177600738 22169   0xae4c00 ERROR                  pulse pulsedeviceprovider.c:439:gst_pulse_device_provider_start:<pulsedeviceprovider0> Failed to connect: Connection refused
Failed to start device monitor!

chehefen, 2019-05-30:
can you follow the trouble shooting steps here and see if it helps? https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#raspberry-pi-failure-to-load-the-camera-device. Thanks.

h5li, 2019-05-30:
Thank you very much! Now my Kinesis Stream is coming alive! Thanks for your awesome help!

closed 2019-05-30 by h5li