awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #215: error:   FRAMES_MISSING_FOR_TRACK

cigiled, 2019-06-17:
Hi, 
       I  uploaded the video an audio to KVS and had a error.
               error:    "ErrorCode":"FRAMES_MISSING_FOR_TRACK","ErrorId":4011
      what's the reason make it ?

unicornss, 2019-06-17:
Hi @cigiled ,

Can you confirm if the media has both audio and video stream?

Also, could you provide the following details:

- Which sample application or gst-launch-1.0 command you are using to send the media stream?
- what is the type of media you are trying to send to KVS? Is it RTSP, webcam or file?

Thanks

cigiled, 2019-06-18:
Hi,  Thanks.
 I create 2 pthreads for video and audio and uploading them to KVS, I do not call the gstreamer interface.
the video format is h264, the  audio format is aac.  i get them form a MP4 file and by runing ffmpeg order to  extract .
what's the reason make it ?

unicornss, 2019-06-18:
Hi @cigiled 

Were you able to refer the audio video sample kinesis_video_gstreamer_audio_video_sample_app.cpp 
that demonstrates how to send frames to KVS SDK using the [putFrame helper](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_audio_video_sample_app.cpp#L542 )

> data->putFrameHelper->putFrameMultiTrack(frame, track_id == DEFAULT_VIDEO_TRACKID);

Can you send the logs also?  Thanks

cigiled, 2019-06-18:
Hi,
     Thank you very much.
     The log at the attachment.
     I refered the kinesis_video_gstreamer_audio_video_sample_app.cpp and changed it, I don't call the gstremer interface.
[log.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3301368/log.txt)

cigiled, 2019-06-18:
Hi @unicornss 
    Thank you very much you can help me.
    My code file in the attachment.
    
[kinesis_audio_video.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3301395/kinesis_audio_video.txt)

Thanks.

unicornss, 2019-06-18:
Thanks for the logs. In the meantime, could you confirm if you are able to test video only streaming (h264 frames) as in [kinesis_video_gstreamer_sample_app.cpp](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp) using the [putFrame](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L318) .

In addition to the sample, you can also refer to the [test](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/f332a735a82b80b76446d3ce2aa1172af046ef1a/kinesis-video-producer/tst/ProducerFunctionalityTest.cpp#L29) which sends dummy frame data.

  
Note: Please ensure no keys are included in the logs and snippets.

MushMal, 2019-07-24:
Any updates on this issue you guys?

unicornss, 2019-11-11:
Closing the issue for now. Please reopen or create a new issue if needed.

closed 2019-11-11 by unicornss