awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #216: Problem with RTSP - data isn't sent to kinesis

ArthurAzrieli, 2019-06-17:
Hi, I have a working stream that I checked in VLC (using [](https://github.com/Ullaakut/RTSPAllTheThings)) and when I run the container it creates a Kinesis stream in AWS but has errors related to connecting with the RTSP stream and doesn't send data to the Kinesis. Help will be appreciated.

`docker run -it <image_id> <access_key> <secret_key> rtsp://0.0.0.0:8554/live.sdp my_stream`

[INFO ][2019-06-17 11:50:44] Using region: us-west-2
[INFO ][2019-06-17 11:50:44] Using aws credentials for Kinesis Video Streams
[INFO ][2019-06-17 11:50:44] No session token was detected.
[INFO ][2019-06-17 11:50:44] Initializing curl.
[INFO ][2019-06-17 11:50:44] 
createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2019-06-17 11:50:44] 
heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ][2019-06-17 11:50:44] 
heapInitialize(): Creating AIV heap.
[INFO ][2019-06-17 11:50:44] 
heapInitialize(): Heap is initialized OK
[DEBUG][2019-06-17 11:50:44] Refreshing credentials. Force refreshing: 0 Now time is: 1560772244495419546 Expiration: 0
[INFO ][2019-06-17 11:50:44] New credentials expiration is 1560774644
[DEBUG][2019-06-17 11:50:44] createDeviceHandler invoked
[INFO ][2019-06-17 11:50:44] 
createDeviceResultEvent(): Create device result event.
[DEBUG][2019-06-17 11:50:44] clientReadyHandler invoked
[DEBUG][2019-06-17 11:50:44] Awaiting for the producer to become ready...
[DEBUG][2019-06-17 11:50:44] Kinesis Video producer is Ready.
[DEBUG][2019-06-17 11:50:44] Client is ready
[INFO ][2019-06-17 11:50:44] Creating Kinesis Video Stream my_stream
[INFO ][2019-06-17 11:50:44] 
createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2019-06-17 11:50:44] describeStreamHandler invoked
[DEBUG][2019-06-17 11:50:44] Awaiting for the stream to become ready...
[DEBUG][2019-06-17 11:50:44] Refreshing credentials. Force refreshing: 0 Now time is: 1560772244496444115 Expiration: 0
[WARN ][2019-06-17 11:50:45] HTTP Error 404: Response: {"Message":"The requested stream is not found or not active."}
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=<access_key>/20190617/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;user-agent;x-amz-date, Signature=<signature>
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS/1.7.9 GCC/7.4.0 Linux/3.10.0-514.26.2.el7.x86_64 x86_64
    X-Amz-Date: 20190617T115044Z
[DEBUG][2019-06-17 11:50:45] describeStream response: {"Message":"The requested stream is not found or not active."}
[INFO ][2019-06-17 11:50:45] Describe stream did not find the stream my_stream in Kinesis Video (stream will be created)
[INFO ][2019-06-17 11:50:45] 
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2019-06-17 11:50:45] createStreamHandler invoked
[DEBUG][2019-06-17 11:50:45] createStreamHandler post body: {"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","MediaType":"video/h264","StreamName":"my_stream"}

[DEBUG][2019-06-17 11:50:45] Refreshing credentials. Force refreshing: 0 Now time is: 1560772245587613872 Expiration: 0
[DEBUG][2019-06-17 11:50:46] createStream response: {"StreamARN":"arn:aws:kinesisvideo:us-west-2:<aws_account_id>:stream/my_stream/1560772246547"}
[INFO ][2019-06-17 11:50:46] Created new Kinesis Video stream: arn:aws:kinesisvideo:us-west-2:<aws_account_id>:stream/my_stream/1560772246547
[INFO ][2019-06-17 11:50:46] 
createStreamResultEvent(): Create stream result event.
[DEBUG][2019-06-17 11:50:46] streamingEndpointHandler invoked
[DEBUG][2019-06-17 11:50:46] Refreshing credentials. Force refreshing: 0 Now time is: 1560772246686713164 Expiration: 0
[DEBUG][2019-06-17 11:50:47] getStreamingEndpoint response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ][2019-06-17 11:50:47] streaming to endpoint: https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com
[INFO ][2019-06-17 11:50:47] 
getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2019-06-17 11:50:47] streamingTokenHandler invoked
[DEBUG][2019-06-17 11:50:47] Refreshing credentials. Force refreshing: 1 Now time is: 1560772247735296464 Expiration: 1560774644
[INFO ][2019-06-17 11:50:47] New credentials expiration is 1560774647
[INFO ][2019-06-17 11:50:47] 
getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2019-06-17 11:50:47] streamReadyHandler invoked
[DEBUG][2019-06-17 11:50:47] Kinesis Video stream my_stream is Ready.
[DEBUG][2019-06-17 11:50:47] Stream is ready

(gst-plugin-scanner:13): GStreamer-WARNING **: Failed to load plugin '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libfl.so': /opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libfl.so: undefined symbol: yylex
[INFO ][2019-06-17 11:50:49] Streaming from rtsp source
Error received from element source: Could not open resource for reading and writing.
Debugging information: gstrtspsrc.c(6775): gst_rtspsrc_retrieve_sdp (): /GstPipeline:rtsp-kinesis-pipeline/GstRTSPSrc:source:
[... 30 lines cut ...]
[INFO ][2019-06-17 11:50:52] 
getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2019-06-17 11:50:52] streamReadyHandler invoked
[DEBUG][2019-06-17 11:50:52] Kinesis Video stream my_stream is Ready.
[INFO ][2019-06-17 11:50:52] Freeing Kinesis Video Stream my_stream
[DEBUG][2019-06-17 11:50:52] Stream is ready
[INFO ][2019-06-17 11:50:52] Streaming from rtsp source
Error received from element source: Could not open resource for reading and writing.
Debugging information: gstrtspsrc.c(6775): gst_rtspsrc_retrieve_sdp (): /GstPipeline:rtsp-kinesis-pipeline/GstRTSPSrc:source:
Failed to connect. (Generic error)
[INFO ][2019-06-17 11:50:52] 
stopKinesisVideoStream(): Stopping Kinesis Video Stream 00007f0a35e2cf31.
[DEBUG][2019-06-17 11:50:52] Awaiting for the stream my_stream to stop...
[DEBUG][2019-06-17 11:50:52] streamClosedHandler invoked for upload handle: 18446744073709551615
[DEBUG][2019-06-17 11:50:52] Kinesis Video stream my_stream is Closed.
[INFO ][2019-06-17 11:50:52] Freeing Kinesis Video Stream my_stream
[INFO ][2019-06-17 11:50:52] 
freeKinesisVideoStream(): Freeing Kinesis Video stream.
[INFO ][2019-06-17 11:50:52] Attempt to recreate kinesis video stream
[INFO ][2019-06-17 11:50:52] Creating Kinesis Video Stream my_stream
[INFO ][2019-06-17 11:50:52] 
createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2019-06-17 11:50:52] describeStreamHandler invoked
[DEBUG][2019-06-17 11:50:52] Awaiting for the stream to become ready...
[DEBUG][2019-06-17 11:50:52] Refreshing credentials. Force refreshing: 0 Now time is: 1560772252323535076 Expiration: 0

unicornss, 2019-06-17:
Please check if the RTSP URL `rtsp://0.0.0.0:8554/live.sdp`  is valid and accessible from the Docker container.

ArthurAzrieli, 2019-06-17:
@unicornss thank you so much for your help, that was exactly the issue, started docker with host mode and now it works

closed 2019-06-17 by ArthurAzrieli

yuefeic, 2020-03-18:
Hi, I am new to docker as well as new to kinesis and gstreamer. Can you elaborate on the command you used?