awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #217: readme mistake: omxh264enc periodicity-idr -> periodicty-idr

alexeypolitov, 2019-06-19:
In omxh264enc parameters you use `periodicity-idr`, but correct parameter name will be `periodicty-idr`
```
gst-launch-1.0 -v v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! omxh264enc periodicity-idr=45 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=YourStreamName access-key="YourAccessKey" secret-key="YourSecretKey"
```

unicornss, 2019-06-20:
Thanks @alexeypolitov for noticing the parameter name incorrectly spelled in this particular gst-launch-1.0 command.  We will update in the upcoming release. 

Thanks again.

chehefen, 2019-07-25:
README has been corrected in the release 2.0.0.

closed 2019-07-25 by chehefen