awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #218: Error regarding FRAGMENT TIMECODE using RTSP

dolivera130, 2019-06-20:
Hello 

When I run the gstreamer app I received these kind of errors, repeatedly (with my Mac camera it doesn't happen), but when I run this with an external camera connected thru Ethernet using RTSP protocol is when I get this error… this seems to be a frame timeset error, not sure if this has to be with the bandwith or this is some problem with the gstreamer application itself

I’ve done some research , and it’s mentioned that this fixes the problem (in green):
 
“
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/185
 
I was able to resolve my issue by passing the width, height, and framerate:
./kinesis_video_gstreamer_sample_app test -w 640 -h 480 -f 15
“

I also tried the -w -h -f from below, but didn’t solve the problem (it’s also on the log file I’m attaching)…
 
8c8590c030e3:kinesis-video-native-build diegoli$ ./kinesis_video_gstreamer_sample_app LiveRekognitionVideoAnalysisBlog "rtsp://192.168.3.7:554/profile4/media.smp" -w 640 -h 480 -f 15

Here is the special part from the log from my machine
[logs_13Jun_DiegoMachine.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3308403/logs_13Jun_DiegoMachine.log)
 with the error:
 
 
./kinesis_video_gstreamer_sample_app LiveRekognitionVideoAnalysisBlog "rtsp://192.168.3.7:554/profile4/media.smp"
 
 
[DEBUG][2019-06-13 15:13:19] Curl post body write function for stream: LiveRekognitionVideoAnalysisBlog and upload handle: 0 returned: {"EventType":"RECEIVED","FragmentTimecode":0,"FragmentNumber":"91343852333182185060225948131993454139249370214"}
 
[DEBUG][2019-06-13 15:13:19] fragmentAckReceivedHandler invoked
[WARN ][2019-06-13 15:13:19] Reporting fragment ack received. Ack timecode 0
[DEBUG][2019-06-13 15:13:19] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-13 15:13:19] Curl post body write function for stream: **LiveRekognitionVideoAnalysisBlog and upload handle: 0 returned:{"EventType":"ERROR","FragmentTimecode":387,"FragmentNumber":"91343852333182185065177708289134975337538652475","ErrorCode":"FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS","ErrorId":4004}**

Enclosed the log file.
Could you please help me?

Thanks
Diego

chehefen, 2019-06-20:
Can you apply this patch while in native-build directory and see if it helps? thanks
[patch.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3308442/patch.txt)

dolivera130, 2019-06-20:
Hi, What is the procedure to apply this patch? Could you please send me the steps? And how to verify it was successfully applied?
Thanks!

chehefen, 2019-06-21:
you can navigate to the kinesis-video-native-build directory, run `git apply /path/to/patch.txt`, then run `make`. Then you can use the same command to start streaming and verify if the same error occurs or not.

MushMal, 2019-12-04:
@dolivera130 have you had a chance to try Felix's suggestion?

disa6302, 2020-02-25:
Closing this issue for now. Feel free to reopen/create new issue in the future.

closed 2020-02-25 by disa6302