awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #219: Issue with RTSP, no data is shown in Kinesis

ievbu, 2019-06-20:
I have followed steps (https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/tree/master/docker_native_scripts/amazonlinux-docker) to create producer for RTSP streaming to Kinesis. However, I could not run it successfully (I have tried lots of variations of `gst-launch-1.0` command that were suggested in issues). Here are some logs (some private information is deleted):
```
Setting pipeline to PAUSED ...
[INFO ][2019-06-20 11:23:35] Initializing curl.
[INFO ][2019-06-20 11:23:35] 
createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2019-06-20 11:23:35] 
heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
[INFO ][2019-06-20 11:23:35] 
heapInitialize(): Creating AIV heap.
[INFO ][2019-06-20 11:23:35] 
heapInitialize(): Heap is initialized OK
[DEBUG][2019-06-20 11:23:35] Refreshing credentials. Force refreshing: 0 Now time is: 1561029815113223864 Expiration: 0
[INFO ][2019-06-20 11:23:35] New credentials expiration is 1561032215
[DEBUG][2019-06-20 11:23:35] createDeviceHandler invoked
[INFO ][2019-06-20 11:23:35] 
createDeviceResultEvent(): Create device result event.
[DEBUG][2019-06-20 11:23:35] clientReadyHandler invoked
[DEBUG][2019-06-20 11:23:35] Awaiting for the producer to become ready...
[DEBUG][2019-06-20 11:23:35] Kinesis Video producer is Ready.
[INFO ][2019-06-20 11:23:35] try creating stream
[INFO ][2019-06-20 11:23:35] Creating Kinesis Video Stream kinesis_test_stream
[INFO ][2019-06-20 11:23:35] 
createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2019-06-20 11:23:35] describeStreamHandler invoked
[DEBUG][2019-06-20 11:23:35] Awaiting for the stream to become ready...
[DEBUG][2019-06-20 11:23:35] Refreshing credentials. Force refreshing: 0 Now time is: 1561029815113389221 Expiration: 0
[DEBUG][2019-06-20 11:23:35] describeStream response: {"StreamInfo":{"CreationTime":1.560939730564E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"...","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"...","StreamName":"kinesis_test_stream","Version":"..."}}
[DEBUG][2019-06-20 11:23:35] Discovered existing Kinesis Video stream: ...
[DEBUG][2019-06-20 11:23:35] stream arn in stream_info struct: ...
[INFO ][2019-06-20 11:23:35] 
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2019-06-20 11:23:35] streamingEndpointHandler invoked
[DEBUG][2019-06-20 11:23:35] Refreshing credentials. Force refreshing: 0 Now time is: 1561029815408638377 Expiration: 0
[DEBUG][2019-06-20 11:23:35] getStreamingEndpoint response: {"DataEndpoint":"https://<...>.kinesisvideo.eu-west-1.amazonaws.com"}
[INFO ][2019-06-20 11:23:35] streaming to endpoint: https://<...>.kinesisvideo.eu-west-1.amazonaws.com
[INFO ][2019-06-20 11:23:35] 
getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2019-06-20 11:23:35] streamingTokenHandler invoked
[DEBUG][2019-06-20 11:23:35] Refreshing credentials. Force refreshing: 1 Now time is: 1561029815669021719 Expiration: 1561032215
[INFO ][2019-06-20 11:23:35] New credentials expiration is 1561032215
[INFO ][2019-06-20 11:23:35] 
getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2019-06-20 11:23:35] streamReadyHandler invoked
[DEBUG][2019-06-20 11:23:35] Kinesis Video stream kinesis_test_stream is Ready.
Stream is ready
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
Progress: (connect) Connecting to rtsp://<...>
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
Progress: (request) SETUP stream 0
Progress: (open) Opened Stream
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
Progress: (request) Sent PLAY request
[INFO ][2019-06-20 11:23:37] try starting stream.
[INFO ][2019-06-20 11:23:37] 
kinesisVideoStreamFormatChanged(): Stream format changed.
[DEBUG][2019-06-20 11:23:37] putStreamHandler invoked
[DEBUG][2019-06-20 11:23:37] Created a new PutMedia request with stream upload handle: 0
[INFO ][2019-06-20 11:23:37] 
putStreamResultEvent(): Put stream result event.
[INFO ][2019-06-20 11:23:37] Creating new connection for Kinesis Video stream: kinesis_test_stream
[DEBUG][2019-06-20 11:23:37] Refreshing credentials. Force refreshing: 0 Now time is: 1561029817958604718 Expiration: 0
[INFO ][2019-06-20 11:23:37] 
getKinesisVideoStreamMetrics(): Get stream metrics for Stream 00000000012be411.
[INFO ][2019-06-20 11:23:37] 
getKinesisVideoMetrics(): Get the memory metrics size.
[DEBUG][2019-06-20 11:23:37] Kinesis Video client and stream metrics
	>> Overall storage byte size: 536870912
	>> Available storage byte size: 536450206
	>> Allocated storage byte size: 420706
	>> Total view allocation byte size: 144072
	>> Total streams frame rate (fps): 25
	>> Total streams transfer rate (bps): 33554432 (32768 Kbps)
	>> Current view duration (ms): 0
	>> Overall view duration (ms): 0
	>> Current view byte size: 420657
	>> Overall view byte size: 420657
	>> Current frame rate (fps): 25
	>> Current transfer rate (bps): 33554432 (32768 Kbps)
[INFO ][2019-06-20 11:23:38] 
getKinesisVideoStreamMetrics(): Get stream metrics for Stream 00000000012be411.
[INFO ][2019-06-20 11:23:38] 
getKinesisVideoMetrics(): Get the memory metrics size.
[DEBUG][2019-06-20 11:23:38] Kinesis Video client and stream metrics
	>> Overall storage byte size: 536870912
	>> Available storage byte size: 535376229
	>> Allocated storage byte size: 1494683
	>> Total view allocation byte size: 144072
	>> Total streams frame rate (fps): 4783
	>> Total streams transfer rate (bps): 33554432 (32768 Kbps)
	>> Current view duration (ms): 1200
	>> Overall view duration (ms): 1200
	>> Current view byte size: 1493164
	>> Overall view byte size: 1493164
	>> Current frame rate (fps): 4783.15
	>> Current transfer rate (bps): 33554432 (32768 Kbps)
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
```
I see that there is unstable fps, maybe this is a problem, however, I cannot make any changes to camera settings.
In video streams dashboard I see warning "No fragments found
No video fragments were found. Verify that video is currently being streamed and that the fragment timecodes are correct."
However, in CloudWatch I can see some data:
[image]

Any help and tips would be really appreciated.

unicornss, 2019-06-20:
Hi @ievbu ,

- Could you please clarify if you are using KVS console playback for viewing the ingested video ?
- Is the region in the KVS console same as the one you are ingesting into (`eu-west-1`)?
- Are you able to see the ingested fragments (metric name is  PutMedia.IncomingFragments )  in the cloudwatch?

This metrics are  helpful to check the storage and view buffer size.

	>> Overall storage byte size: 536870912
	>> Available storage byte size: 535376229
	>> Allocated storage byte size: 1494683
	>> Total view allocation byte size: 144072

The fps metrics within the SDK is on that instantaneous  so please ignore that attribute for now.

ievbu, 2019-06-20:
Thank you for quick response.
- I am using media preview in dashboard (example url: https://eu-west-1.console.aws.amazon.com/kinesisvideo/streams/streamName/kinesis_test_stream?region=eu-west-1);
- yes, region is the same;
- no, I can see only ListFragments.Latency.

zhiyua-git, 2019-06-20:
Hi @ievbu ,

You data are ingested successfully and your AWS console link looks correct, but I don't see any playback related request from your account in the time period. 

As your PutMedia.IncomingFragments was not shown in console, can you double-check if you are logging into AWS console with a different account?

ievbu, 2019-06-21:
Hi @zhiyua-git, I am logging in with the same account. I am using this command for running: 
```
gst-launch-1.0 rtspsrc location="rtsp://<...>" short-header=TRUE ! rtph264depay ! h264parse ! video/x-h264,format=avc,alignment=au ! kvssink stream-name="kinesis_test_stream" storage-size=512 access-key="" secret-key="" aws-region="eu-west-1"
```
I have access to 2 different cameras. Running one of them causes situation as described in this issue. However, I tried to connect other camera and experienced different situation. Running 2nd rtsp is also unsuccessful, but logs differ, here is sample of 2nd camera log using same running command:
```
[DEBUG][2019-06-21 05:35:03] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-21 05:35:03] Wrote 8151 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-21 05:35:03] Pausing CURL read for upload handle: 0
[DEBUG][2019-06-21 05:35:03] Curl post body write function for stream: kinesis_test_stream and upload handle: 0 returned: {"EventType":"ERROR","FragmentTimecode":0,"FragmentNumber":"91343852333183393289704290663317977235036744295","ErrorCode":"MAX_FRAGMENT_DURATION_REACHED","ErrorId":4002}

[DEBUG][2019-06-21 05:35:03] describeStreamHandler invoked
[DEBUG][2019-06-21 05:35:03] streamErrorHandler invoked
[ERROR][2019-06-21 05:35:03] Reporting stream error. Errored timecode: 2300000 Status: 0x5200005e
[DEBUG][2019-06-21 05:35:03] Refreshing credentials. Force refreshing: 0 Now time is: 1561095303239768037 Expiration: 0
[DEBUG][2019-06-21 05:35:03] fragmentAckReceivedHandler invoked
[DEBUG][2019-06-21 05:35:03] describeStream response: {"StreamInfo":{"CreationTime":1.560939730564E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"<...>","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"<...>","StreamName":"kinesis_test_stream","Version":""}}
[DEBUG][2019-06-21 05:35:03] Discovered existing Kinesis Video stream: <...>
[DEBUG][2019-06-21 05:35:03] stream arn in stream_info struct: <...>
[INFO ][2019-06-21 05:35:03] 
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2019-06-21 05:35:03] streamingEndpointHandler invoked
[DEBUG][2019-06-21 05:35:03] Refreshing credentials. Force refreshing: 0 Now time is: 1561095303472430275 Expiration: 0
[DEBUG][2019-06-21 05:35:03] getStreamingEndpoint response: {"DataEndpoint":"<...>"}
[INFO ][2019-06-21 05:35:03] streaming to endpoint: <...>
[INFO ][2019-06-21 05:35:03] 
getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2019-06-21 05:35:03] streamingTokenHandler invoked
[DEBUG][2019-06-21 05:35:03] Refreshing credentials. Force refreshing: 1 Now time is: 1561095303688702251 Expiration: 1561097691
[INFO ][2019-06-21 05:35:03] New credentials expiration is 1561097703
[INFO ][2019-06-21 05:35:03] 
getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2019-06-21 05:35:03] streamReadyHandler invoked
[DEBUG][2019-06-21 05:35:03] putStreamHandler invoked
[DEBUG][2019-06-21 05:35:03] Created a new PutMedia request with stream upload handle: 1
[INFO ][2019-06-21 05:35:03] 
putStreamResultEvent(): Put stream result event.
[INFO ][2019-06-21 05:35:03] Creating new connection for Kinesis Video stream: kinesis_test_stream
[DEBUG][2019-06-21 05:35:03] Refreshing credentials. Force refreshing: 0 Now time is: 1561095303688796002 Expiration: 0
[DEBUG][2019-06-21 05:35:03] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 1
```
I am running this Docker image: 546150905175.dkr.ecr.us-west-2.amazonaws.com/kinesis-video-producer-sdk-cpp-amazon-linux.

unicornss, 2019-06-21:
Hi @ievbu 

For the camera 1, where you are seeing metrics in the logs, can you share us additional logs? would like to see whether you are receiving  ACKs successfully.  

Thanks

ievbu, 2019-06-25:
Here is how logs continue in 1st camera:
```
[INFO ][2019-06-20 11:23:38] Curl post header write function returned:x-amzn-RequestId: cc65c97c-6968-e85a-99e0-a786b3a2d452

[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Curl post body write function for stream: kinesis_test_stream and upload handle: 0 returned: {"EventType":"BUFFERING","FragmentTimecode":0,"FragmentNumber":"91343852333181897858136833923928323010938059545"}

[DEBUG][2019-06-20 11:23:38] fragmentAckReceivedHandler invoked
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:38] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:39] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[... cut ...]
[DEBUG][2019-06-20 11:23:42] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:42] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:42] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-06-20 11:23:42] Curl post body write function for stream: kinesis_test_stream and upload handle: 0 returned: {"EventType":"PERSISTED","FragmentTimecode":4013,"FragmentNumber":"91343852333181897868040354238211365942004345780"}
```
Later logs structure seems to be the same.

zhiyua-git, 2019-06-25:
Hi @ievbu ,

You log shows the video is ingested into KVS properly. Can you use the same accessKey/secretKey for your stream in the HLS viewer to see if you can see the playback here? https://aws-samples.github.io/amazon-kinesis-video-streams-hls-viewer/

Thanks,
-ZH

ievbu, 2019-06-25:
Thank you, yes, I can see playback for 1st camera using this viewer. 
And for the 2nd I receive error: 
```
[ERROR] {
  "message": "No fragments found in the stream for the streaming request.",
  "code": "ResourceNotFoundException",
  "time": "2019-06-25T05:45:27.342Z",
  "requestId": "03134170-577d-40eb-b66c-f27caf3f4e06",
  "statusCode": 404,
  "retryable": false,
  "retryDelay": 10.71161858054409
}
```

zhiyua-git, 2019-06-25:
Hi @ievbu ,

If you can see camera 1 in HLS viewer, then you should be able to view it in console too, at least it would fail with some other error other than "no fragment found". Can you double check your account id in your aws console? You can compare that with the account id inside your StreamArn in the log.

For your second camera, I can see this error in the log

[DEBUG][2019-06-21 05:35:03] Curl post body write function for stream: kinesis_test_stream and upload handle: 0 returned: {"EventType":"ERROR","FragmentTimecode":0,"FragmentNumber":"91343852333183393289704290663317977235036744295","ErrorCode":"MAX_FRAGMENT_DURATION_REACHED","ErrorId":4002}

This means your camera doesn't return any I frame in 10 seconds, can you check if you can tune the setting of your camera for a shorter I frame interval?

Thanks,
-ZH

ievbu, 2019-06-25:
Thank you @zhiyua-git, I can see 1st camera stream.
For the 2nd camera, I cannot change any settings, so I'm closing this issue. Thanks once again for your help.

closed 2019-06-25 by ievbu