awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #220: Send videos input, ie sequence of frames, to cloudformation stack

h5li, 2019-06-21:
Hello! I have recently been working on sending kinesis video inputs to cloud formation stack and cloud formation stack converts inputs as numpy array to my sagemaker endpoints. I checked some blogs about this idea. But it looks like kinesis video stream can only send frame by frame? Is there any way that we can send video by video?

Thank you very much! Appreciate your time and effort.

chehefen, 2019-06-28:
When you say video by video, do you mean that you already have the video as clips and want to send each clip to kvs? Thanks

unicornss, 2019-11-05:
As @chehefen has mentioned, please refer to Readme on how to send  files if you plan to video files... 
Closing the issue for now. Please reopen or create a new one if needed.

closed 2019-11-05 by unicornss