awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #221: Segmentation fault error when rerunning gst-launch-1.0

claw68, 2019-06-24:
Hi, is there a way to rerun gst-launch-1.0 without segmentation error?
I always have to run
> min-install-script
To run gst-launch-1.0 without segmentation error.

My platform is Windows 10 msys64

chehefen, 2019-06-28:
I cant reproduce your issue. What's your gstreamer pipeline like? How did you install the sdk? Did you try both the msvc and mingw installation? if so, you need to uninstall the gstreamer installed in the msvc installation first.

claw68, 2019-07-01:
Hi @chehefen 

I did not install the msvc installation only the mingw one. I followed this instructions:

https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-win.html#producer-sdk-win-mingw

then i run

> gst-launch-1.0 rtspsrc location="rtsp://url" ! rtph264depay ! h264parse ! kvssink stream-name="stream-name" storage-size=512 access-key="ACCESS" secret-key="secret" aws-region="region"

At first run, everything is fine. But if i run it again, I will get a segmentation error. I will have to run min-install-script to get rid of segmentation error. Is there a way to run gst-launch without rerunning min-install-script?

unicornss, 2019-07-03:
In order to check if any environment variables (library path) is causing this behavior, can you please run env before and after min-install-script ?

claw68, 2019-07-08:
> Segmentation fault (before ./min-install-script)

USERDOMAIN=GERMANIUM
OS=Windows_NT
COMMONPROGRAMFILES=C:\Program Files\Common Files
PROCESSOR_LEVEL=6
PSModulePath=C:\Program Files\WindowsPowerShell\Modules;C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules
CommonProgramW6432=C:\Program Files\Common Files
ANDROID_HOME=C:\Users\SVEN\AppData\Local\Android\Sdk\platform-tools\platform-tools
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
MSYSCON=mintty.exe
LANG=en_US.UTF-8
TZ=Asia/Shanghai
MSYSTEM_CARCH=x86_64
HOSTNAME=GERMANIUM
PUBLIC=C:\Users\Public
OLDPWD=/home/SVEN
CONFIG_SITE=/mingw64/etc/config.site
AWS_SECRET_ACCESS_KEY=SECRET_KEY
GST_PLUGIN_PATH=/home/SVEN
MSYSTEM_CHOST=x86_64-w64-mingw32
USERNAME=SVEN
JAVA_HOME=C:\Program Files\Java\jdk1.8.0_151
LOGONSERVER=\\GERMANIUM
PROCESSOR_ARCHITECTURE=AMD64
tmp=C:\Users\SVEN\AppData\Local\Temp
AWS_DEFAULT_REGION=ap-southeast-2
LOCALAPPDATA=C:\Users\SVEN\AppData\Local
COMPUTERNAME=GERMANIUM
FPS_BROWSER_APP_PROFILE_STRING=Internet Explorer
USER=SVEN
!::=::\
SYSTEMDRIVE=C:
USERPROFILE=C:\Users\SVEN
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
SYSTEMROOT=C:\WINDOWS
USERDOMAIN_ROAMINGPROFILE=GERMANIUM
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 142 Stepping 9, GenuineIntel
MINGW_PACKAGE_PREFIX=mingw-w64-x86_64
PWD=/home/SVEN/kinesis/kinesis-video-native-build
HOME=/home/SVEN
TMP=/tmp
MSYSTEM_PREFIX=/mingw64
OneDrive=C:\Users\SVEN\OneDrive
PROCESSOR_REVISION=8e09
FPS_BROWSER_USER_PROFILE_STRING=Default
NUMBER_OF_PROCESSORS=4
ProgramW6432=C:\Program Files
COMSPEC=C:\WINDOWS\system32\cmd.exe
APPDATA=C:\Users\SVEN\AppData\Roaming
SHELL=/usr/bin/bash
TERM=xterm
WINDIR=C:\WINDOWS
MINGW_CHOST=x86_64-w64-mingw32
AWS_ACCESS_KEY_ID=ACCESS_KEY
ProgramData=C:\ProgramData
SHLVL=1
PRINTER=Brother DCP-J100 Printer (Copy 1)
ACLOCAL_PATH=/mingw64/share/aclocal:/usr/share/aclocal
PROGRAMFILES=C:\Program Files
MANPATH=/mingw64/local/man:/mingw64/share/man:/usr/local/man:/usr/share/man:/usr/man:/share/man
ORIGINAL_TEMP=/c/Users/SVEN/AppData/Local/Temp
ORIGINAL_TMP=/c/Users/SVEN/AppData/Local/Temp
ALLUSERSPROFILE=C:\ProgramData
TEMP=/tmp
temp=C:\Users\SVEN\AppData\Local\Temp
DriverData=C:\Windows\System32\Drivers\DriverData
MSYSTEM=MINGW64
MINGW_PREFIX=/mingw64
SESSIONNAME=Console
[... 224 lines cut ...]
AWS_ACCESS_KEY_ID=ACCESS_KEY
ProgramData=C:\ProgramData
SHLVL=1
PRINTER=Brother DCP-J100 Printer (Copy 1)
ACLOCAL_PATH=/mingw64/share/aclocal:/usr/share/aclocal
PROGRAMFILES=C:\Program Files
MANPATH=/mingw64/local/man:/mingw64/share/man:/usr/local/man:/usr/share/man:/usr/man:/share/man
ORIGINAL_TEMP=/c/Users/SVEN/AppData/Local/Temp
ORIGINAL_TMP=/c/Users/SVEN/AppData/Local/Temp
ALLUSERSPROFILE=C:\ProgramData
TEMP=/tmp
temp=C:\Users\SVEN\AppData\Local\Temp
DriverData=C:\Windows\System32\Drivers\DriverData
MSYSTEM=MINGW64
MINGW_PREFIX=/mingw64
SESSIONNAME=Console
ProgramFiles(x86)=C:\Program Files (x86)
PATH=/mingw64/bin:/usr/local/bin:/usr/bin:/bin:/c/Windows/System32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0/:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl
PS1=\[\e]0;\w\a\]\n\[\e[32m\]\u@\h \[\e[35m\]$MSYSTEM\[\e[0m\] \[\e[33m\]\w\[\e[0m\]\n\$
HOMEDRIVE=C:
PKG_CONFIG_PATH=/mingw64/lib/pkgconfig:/mingw64/share/pkgconfig
INFOPATH=/usr/local/info:/usr/share/info:/usr/info:/share/info
HOMEPATH=\Users\SVEN
ORIGINAL_PATH=/c/Windows/System32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0/
_=/usr/bin/env

chehefen, 2019-07-09:
can you try if this pipeline works?
```
gst-launch-1.0 -v videotestsrc is-live=TRUE ! fakesink silent=FALSE
```

chehefen, 2019-07-09:
can you also try restarting the mingw shell? Thanks

claw68, 2019-07-16:
Hi @chehefen

> 
> 
> can you try if this pipeline works?
> 
> ```
> gst-launch-1.0 -v videotestsrc is-live=TRUE ! fakesink silent=FALSE
> ```

```
(gst-launch-1.0:16844): GStreamer-WARNING **: 09:08:41.654: Failed to load plugin 'C:\msys64\mingw64\lib\gstreamer-1.0\libgstopencv.dll': 'C:\msys64\mingw64\lib\gstreamer-1.0\libgstopencv.dll': The specified module could not be found.
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Segmentation fault
```

> can you also try restarting the mingw shell? Thanks

It produces the same error.

chehefen, 2019-08-09:
It seems like GST_PLUGIN_PATH is causing the issue. You can work around that by copying libcproducer.dll and libproducer.dll to C:\msys64\mingw64\bin and copy libgstkvssink.dll to C:\msys64\mingw64\lib\gstreamer-1.0 and make sure you are not exporting GST_PLUGIN_PATH.

claw68, 2019-08-14:
Hi @chehefen this solution is working! Thanks!

closed 2019-08-14 by claw68