awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #222: Pass parameters to gst-launch via kinesis_video_gstreamer_sample_app

claw68, 2019-06-24:
Hi,

Is there a way to pass gst-launch-1.0 parameters via the sample app?

I run gst-launch with these parameters

> gst-launch-1.0 rtspsrc location="rtsp://sample_url" ! rtph264depay ! h264parse ! kvssink stream-name="sample_stream_name" storage-size=512 access-key="SAMPLEKEY" secret-key="SAMPLESECRET" aws-region="sample_region"

But i want to do something like:

> kinesis_video_gstreamer_sample_app ! rtph264depay ! h264parse ! kvssink stream-name rtsp://url

I want to pass the parameters "rtph264depay ! h264parse ! kvssink" via kinesis_video_gstreamer_sample_app. Is there a way to do this?

chehefen, 2019-06-28:
Sorry there is no way to do this, this is not how gstreamer works. Can you explain why you want to do this? If you want to use the kvssink plugin in a gstreamer application you can refer to this sample: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gstreamer-plugin/demo-src/kvs_producer_plugin_rtsp_demo.cpp

claw68, 2019-07-01:
Hi @chehefen thank you for the reply.

I encountered this issue:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/193#issuecomment-474093882

I have to add h264parse to the pipeline for it to work. Is there a way append/modify the pipeline of kinesis_video_gstreamer_sample_app? Or is kvs_producer_plugin_rtsp_demo.cpp is the file i have to tinker to build a custom sample app?

unicornss, 2019-07-03:
`kvs_producer_plugin_rtsp_demo.cpp` uses the kvssink element.  You may find references on how to use h264parse [kinesis_video_gstreamer_sample_app](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp) which uses the appsink element to send frames to kvs using putFrame API.  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#running-the-gstreamer-rtsp-sample-application

closed 2019-07-04 by claw68