awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #224: Gstreamer Metadata

pasawaya, 2019-06-26:
Is it possible to attach metadata when using the Gstreamer command line to produce an HLS stream? I see that [it's possible when using the C++ API](http://lifestyletransfer.com/how-to-add-metadata-to-gstreamer-buffer-in-python/).

chehefen, 2019-06-26:
Can you tell us about what you are envisioning for attaching metadata using the gstreamer command line? Can you give us more details to help us understand your use case better.

pasawaya, 2019-06-26:
I haven't fleshed out the exact specifics of my use case but I suppose I'm looking to attach some of the standard metadata the PutMedia API allows for such as producer timestamp/timecode along with some custom metadata such as a camera ID, a latitude-longitude coordinates, etc.

At a high level, I'm essentially trying to decide between the ease of use of a Gstreamer producer and the fine-grained control of a producer implemented with the C++/Java SDK.

unicornss, 2019-11-05:
The GStreamer plugin (kvssink) uses the SDK underneath. Based on your use case description,either one of the samples should be fine for use case. 

Closing the issue for now. Please reopen or create a new one if needed.

closed 2019-11-05 by unicornss