awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #226: Error in streaming through RPi using kinesis aws

UmangBhatt09, 2019-07-08:
I have successfully completed the whole process of streaming at https://www.hackster.io/fabolandry/raspberry-pi-ip-cam-aws-and-hls-streaming-b449e7
now when I run ./kinesis_video_gstreamer_sample_app Stream Name it gives me no output and the error is 
./kinesis_video_gstreamer_sample_app MyStream2
[INFO ][2019-07-08 12:46:06] Using region: us-west-2
[INFO ][2019-07-08 12:46:06] Using aws credentials for Kinesis Video Streams
[INFO ][2019-07-08 12:46:06] No session token was detected.
[INFO ][2019-07-08 12:46:06] Initializing curl.
[INFO ][2019-07-08 12:46:06] 
createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2019-07-08 12:46:06] 
heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ][2019-07-08 12:46:06] 
heapInitialize(): Creating AIV heap.
[INFO ][2019-07-08 12:46:06] 
heapInitialize(): Heap is initialized OK
[DEBUG][2019-07-08 12:46:06] Refreshing credentials. Force refreshing: 0 Now time is: 1562589966089698707 Expiration: 0
[INFO ][2019-07-08 12:46:06] New credentials expiration is 1562592366
[DEBUG][2019-07-08 12:46:06] createDeviceHandler invoked
[INFO ][2019-07-08 12:46:06] 
createDeviceResultEvent(): Create device result event.
[DEBUG][2019-07-08 12:46:06] clientReadyHandler invoked
[DEBUG][2019-07-08 12:46:06] Awaiting for the producer to become ready...
[DEBUG][2019-07-08 12:46:06] Kinesis Video producer is Ready.
[DEBUG][2019-07-08 12:46:06] Client is ready
[INFO ][2019-07-08 12:46:06] Creating Kinesis Video Stream MyStream2
[INFO ][2019-07-08 12:46:06] 
createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2019-07-08 12:46:06] describeStreamHandler invoked
[DEBUG][2019-07-08 12:46:06] Awaiting for the stream to become ready...
[DEBUG][2019-07-08 12:46:06] Refreshing credentials. Force refreshing: 0 Now time is: 1562589966092288773 Expiration: 0
[DEBUG][2019-07-08 12:46:07] describeStream response: {"StreamInfo":{"CreationTime":1.562589645061E9,"DataRetentionInHours":24,"DeviceName":null,"KmsKeyId":"arn:aws:kms:us-west-2:175499281365:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:175499281365:stream/MyStream2/1562589645061","StreamName":"MyStream2","Version":"6CRnafBoUOwmTbDTPS8U"}}
[DEBUG][2019-07-08 12:46:07] Discovered existing Kinesis Video stream: arn:aws:kinesisvideo:us-west-2:175499281365:stream/MyStream2/1562589645061
[DEBUG][2019-07-08 12:46:07] stream arn in stream_info struct: arn:aws:kinesisvideo:us-west-2:175499281365:stream/MyStream2/1562589645061
[INFO ][2019-07-08 12:46:07] 
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2019-07-08 12:46:07] streamingEndpointHandler invoked
[DEBUG][2019-07-08 12:46:07] Refreshing credentials. Force refreshing: 0 Now time is: 1562589967283212799 Expiration: 0
[DEBUG][2019-07-08 12:46:08] getStreamingEndpoint response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ][2019-07-08 12:46:08] streaming to endpoint: https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com
[INFO ][2019-07-08 12:46:08] 
getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2019-07-08 12:46:08] streamingTokenHandler invoked
[DEBUG][2019-07-08 12:46:08] Refreshing credentials. Force refreshing: 1 Now time is: 1562589968534595936 Expiration: 1562592366
[INFO ][2019-07-08 12:46:08] New credentials expiration is 1562592368
[INFO ][2019-07-08 12:46:08] 
getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2019-07-08 12:46:08] streamReadyHandler invoked
[DEBUG][2019-07-08 12:46:08] Kinesis Video stream MyStream2 is Ready.
[DEBUG][2019-07-08 12:46:08] Stream is ready
[INFO ][2019-07-08 12:46:08] Streaming from live source
[DEBUG][2019-07-08 12:46:08] Streaming with live source and width: 0, height: 0, fps: 25, bitrateInKBPS512
[INFO ][2019-07-08 12:46:09] 
kinesisVideoStreamFormatChanged(): Stream format changed.
[DEBUG][2019-07-08 12:46:09] putStreamHandler invoked
[DEBUG][2019-07-08 12:46:09] Created a new PutMedia request with stream upload handle: 0
[INFO ][2019-07-08 12:46:09] 
putStreamResultEvent(): Put stream result event.
[INFO ][2019-07-08 12:46:09] Creating new connection for Kinesis Video stream: MyStream2
[DEBUG][2019-07-08 12:46:09] Refreshing credentials. Force refreshing: 0 Now time is: 1562589969114529168 Expiration: 0
[INFO ][2019-07-08 12:46:09] 
getKinesisVideoStreamMetrics(): Get stream metrics for Stream 000000000048b579.
[INFO ][2019-07-08 12:46:09] 
getKinesisVideoMetrics(): Get the memory metrics size.
[DEBUG][2019-07-08 12:46:09] Kinesis Video client and stream metrics
	>> Overall storage byte size: 134217728
	>> Available storage byte size: 134205323
	>> Allocated storage byte size: 12405
	>> Total view allocation byte size: 144064
	>> Total streams frame rate (fps): 25
	>> Total streams transfer rate (bps): 33554432 (32768 Kbps)
[... 313 lines cut ...]
[INFO ][2019-07-08 12:46:39] 
putStreamResultEvent(): Put stream result event.
[INFO ][2019-07-08 12:46:39] Creating new connection for Kinesis Video stream: MyStream2
[DEBUG][2019-07-08 12:46:39] Refreshing credentials. Force refreshing: 0 Now time is: 1562589999847282601 Expiration: 0
[DEBUG][2019-07-08 12:46:42] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 7
[INFO ][2019-07-08 12:46:42] Curl post header write function returned:x-amzn-RequestId: ff9b1520-b577-ccf5-aa06-285115d38d60

[DEBUG][2019-07-08 12:46:42] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 7
[DEBUG][2019-07-08 12:46:43] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 7
[DEBUG][2019-07-08 12:46:43] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 7
[DEBUG][2019-07-08 12:46:43] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 7
[DEBUG][2019-07-08 12:46:44] Wrote 16372 bytes to Kinesis Video. Upload stream handle: 7
[DEBUG][2019-07-08 12:46:44] Curl post body write function for stream: MyStream2 and upload handle: 7 returned: {"EventType":"ERROR","FragmentTimecode":0,"FragmentNumber":"91343852333181605709239322731743776950322689167","ErrorCode":"FRAGMENT_TIMECODE_LESSER_THAN_PREVIOUS","ErrorId":4004}

[DEBUG][2019-07-08 12:46:44] describeStreamHandler invoked
[DEBUG][2019-07-08 12:46:44] streamErrorHandler invoked
[ERROR][2019-07-08 12:46:44] Reporting stream error. Errored timecode: 4060000 Status: 1375731808
[DEBUG][2019-07-08 12:46:44] fragmentAckReceivedHandler invoked
[DEBUG][2019-07-08 12:46:44] Refreshing credentials. Force refreshing: 0 Now time is: 1562590004133773604 Expiration: 0
[WARN ][2019-07-08 12:46:44] Reporting fragment ack received. Ack timecode 0
[INFO ][2019-07-08 12:46:44] Closing connection for upload stream handle: 7
[DEBUG][2019-07-08 12:46:44] Wrote 0 bytes to Kinesis Video. Upload stream handle: 7
[DEBUG][2019-07-08 12:46:44] terminate curl for handle 7
[DEBUG][2019-07-08 12:46:44] Connection for Kinesis Video stream: MyStream2 closed.
[DEBUG][2019-07-08 12:46:44] Network thread for Kinesis Video stream: MyStream2 with upload handle: 7 exited. http status: 200

unicornss, 2019-07-08:
eporting stream error. Errored timecode: 4060000 Status: 1375731808 (Hex 0x52000060) Suggests [STATUS_ACK_ERR_FRAGMENT_TIMECODE_NOT_MONOTONIC](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html) .

Can you provide us the following details:
 - Operating system (Ubuntu/MacOS/Pi)
 - Webcam or USB cam ( if it is an USB camera, model is helpful)

Also, please  send us the logs with the frame debug environment variable set 

   ` export DEBUG_DUMP_FRAME_INFO=TRUE `

unicornss, 2019-10-31:
Closing the issue for now; please reopen if it needs support. Thanks.

closed 2019-10-31 by unicornss