awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #227: Streams stops after a while when using Docker for raspberry Pi (GST-STREAMER C++)

luisRubiera, 2019-07-08:
Hi team,
I'm currently working in a project that uses both **Docker for raspberry Pi** and **Docker for debian/Ubuntu** image. if i use the ubuntu image, it works like a charm, but when i try using the one for Raspberry, the streams stops after a few minutes.

my script for raspberry:
```
docker run -e AWS_DEFAULT_REGION=$MY_AWS_REGION -it 46150905175.dkr.ecr.us-west-2.amazonaws.com/kinesis-video-producer-sdk-cpp-raspberry-pi:latest   gst-launch-1.0  rtspsrc location=$MY_RTSP_URL short-header=TRUE ! rtph264depay ! h264parse! video/x-h264, format=avc,alignment=au ! kvssink stream-name=$STREAM_NAME storage-size=128 access-key=$MY_ACCESS_KEY secret-key=$MY_SECRET_KEY
```

I contacted the AWS support and they told me:

> The service team have potentially identified the source of your issue and would like you to make a small change to the code on line 1052 of the following sample [1], the change they would like test is replacing "||" with "&&". This change from OR to AND should hopefully restrict the terms under which the stream can be stopped due to lack of content. currently the line be changed line reads as:
> 
>  if (buf == NULL || track_data == NULL) {
> 
> and the service team would like you to replace this with:
> 
>  if (buf == NULL && track_data == NULL) {

But i still have the problem ( i changed the Dockerfile to use the modified version instead of cloning the project )

Just to check, this is my logs for a working stream using the **same camera** and **same network** :
[image]

This is the one from the Raspberry:
[image]

and these are the logs of the Raspberry:
```
[DEBUG][2019-07-08 13:47:41] fragmentAckReceivedHandler invoked
[DEBUG][2019-07-08 13:47:41] Wrote 1398 bytes to Kinesis Video. Upload stream handle: 3
[DEBUG][2019-07-08 13:47:41] Pausing CURL read for upload handle: 3
[DEBUG][2019-07-08 13:47:42] Wrote 8670 bytes to Kinesis Video. Upload stream handle: 3
[DEBUG][2019-07-08 13:47:42] Pausing CURL read for upload handle: 3
[DEBUG][2019-07-08 13:47:43] Wrote 7667 bytes to Kinesis Video. Upload stream handle: 3
[DEBUG][2019-07-08 13:47:43] Pausing CURL read for upload handle: 3
[DEBUG][2019-07-08 13:47:44] Wrote 6713 bytes to Kinesis Video. Upload stream handle: 3
[DEBUG][2019-07-08 13:47:44] Pausing CURL read for upload handle: 3
[DEBUG][2019-07-08 13:47:45] Wrote 1949 bytes to Kinesis Video. Upload stream handle: 3
[DEBUG][2019-07-08 13:47:45] Pausing CURL read for upload handle: 3
[INFO ][2019-07-08 13:48:13] 
stopKinesisVideoStream(): Stopping Kinesis Video Stream 00000000018c24e1.
[DEBUG][2019-07-08 13:48:13] Awaiting for the stream 1x5dmbug1edd5wmp44ziktdh0qs8e3znwdiap2wdopgecnoqlh4 to stop...
[INFO ][2019-07-08 13:48:13] 
getStreamData(): Handle 3 waiting for last persisted ack with ts 71837160000
[DEBUG][2019-07-08 13:48:13] Wrote 84 bytes to Kinesis Video. Upload stream handle: 3
[DEBUG][2019-07-08 13:48:13] Pausing CURL read for upload handle: 3 waiting for last ack.
[ERROR][2019-07-08 13:48:20] curl perform failed for url https://s-96607c7a.kinesisvideo.eu-central-1.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
[WARN ][2019-07-08 13:48:20] HTTP Error 0: Response: 
Request URL: https://s-96607c7a.kinesisvideo.eu-central-1.amazonaws.com/putMedia
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIASGHSSNSCLB7LENHZ/20190708/eu-central-1/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowledgment-required;x-amzn-fragment-timecode-type;x-amzn-producer-start-timestamp;x-amzn-stream-name, Signature=9775c0de27f66f8252c7cd03647458495b182e526c8b0e02042b3bb98cf01a07
    connection: keep-alive
    host: s-96607c7a.kinesisvideo.eu-central-1.amazonaws.com
    transfer-encoding: chunked
    user-agent: AWS-SDK-KVS-CLIENT/1.7.9 GCC/6.3.0 Linux/4.14.98-v7+ armv7l
    X-Amz-Date: 20190708T134147Z
    x-amzn-fragment-acknowledgment-required: 1
    x-amzn-fragment-timecode-type: RELATIVE
    x-amzn-producer-start-timestamp: 1562593307.299
    x-amzn-stream-name: 1x5dmbug1edd5wmp44ziktdh0qs8e3znwdiap2wdopgecnoqlh4
[DEBUG][2019-07-08 13:48:20] Connection for Kinesis Video stream: 1x5dmbug1edd5wmp44ziktdh0qs8e3znwdiap2wdopgecnoqlh4 closed.
[DEBUG][2019-07-08 13:48:20] Network thread for Kinesis Video stream: 1x5dmbug1edd5wmp44ziktdh0qs8e3znwdiap2wdopgecnoqlh4 with upload handle: 3 exited. http status: 0
[WARN ][2019-07-08 13:48:20] Stream for 1x5dmbug1edd5wmp44ziktdh0qs8e3znwdiap2wdopgecnoqlh4 has exited without triggering end-of-stream. Service call result: 599
[INFO ][2019-07-08 13:48:20] 
kinesisVideoStreamTerminated(): Stream terminated event.
[DEBUG][2019-07-08 13:48:20] describeStreamHandler invoked
[DEBUG][2019-07-08 13:48:20] Refreshing credentials. Force refreshing: 0 Now time is: 1562593700927773880 Expiration: 0
[DEBUG][2019-07-08 13:48:21] describeStream response: {"StreamInfo":{"CreationTime":1.561714002027E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:eu-central-1:150832573572:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:eu-central-1:150832573572:stream/1x5dmbug1edd5wmp44ziktdh0qs8e3znwdiap2wdopgecnoqlh4/1561714002027","StreamName":"1x5dmbug1edd5wmp44ziktdh0qs8e3znwdiap2wdopgecnoqlh4","Version":"xQHFDFXItqP9OAUxpDbN"}}
[DEBUG][2019-07-08 13:48:21] Discovered existing Kinesis Video stream: arn:aws:kinesisvideo:eu-central-1:150832573572:stream/1x5dmbug1edd5wmp44ziktdh0qs8e3znwdiap2wdopgecnoqlh4/1561714002027
[DEBUG][2019-07-08 13:48:21] stream arn in stream_info struct: arn:aws:kinesisvideo:eu-central-1:150832573572:stream/1x5dmbug1edd5wmp44ziktdh0qs8e3znwdiap2wdopgecnoqlh4/1561714002027
[... 35 lines cut ...]
[DEBUG][2019-07-08 13:48:21] Curl post body write function for stream: 1x5dmbug1edd5wmp44ziktdh0qs8e3znwdiap2wdopgecnoqlh4 and upload handle: 4 returned: {"EventType":"RECEIVED","FragmentTimecode":350007,"FragmentNumber":"91343852333181486862043791178096858207627834317"}

[DEBUG][2019-07-08 13:48:21] fragmentAckReceivedHandler invoked
[DEBUG][2019-07-08 13:48:21] Curl post body write function for stream: 1x5dmbug1edd5wmp44ziktdh0qs8e3znwdiap2wdopgecnoqlh4 and upload handle: 4 returned: {"EventType":"PERSISTED","FragmentTimecode":350007,"FragmentNumber":"91343852333181486862043791178096858207627834317"}

[DEBUG][2019-07-08 13:48:21] fragmentAckReceivedHandler invoked
[INFO ][2019-07-08 13:48:21] 
getStreamData(): Indicating an EOS after last persisted ACK is received for stream upload handle 4
[INFO ][2019-07-08 13:48:21] Reported end-of-stream for 1x5dmbug1edd5wmp44ziktdh0qs8e3znwdiap2wdopgecnoqlh4. Upload handle: 4
[DEBUG][2019-07-08 13:48:21] Wrote 0 bytes to Kinesis Video. Upload stream handle: 4
[DEBUG][2019-07-08 13:48:21] terminate curl for handle 4
[DEBUG][2019-07-08 13:48:21] streamClosedHandler invoked for upload handle: 4
[DEBUG][2019-07-08 13:48:21] Close connection: 1x5dmbug1edd5wmp44ziktdh0qs8e3znwdiap2wdopgecnoqlh4, upload handle: 4, is EOS: 1, is in Shutdown: 0
[DEBUG][2019-07-08 13:48:21] Kinesis Video stream 1x5dmbug1edd5wmp44ziktdh0qs8e3znwdiap2wdopgecnoqlh4 is Closed.
[INFO ][2019-07-08 13:48:22] Sending eos
Got EOS from element "pipeline0".
Execution ended after 2:00:24.952496328
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
[INFO ][2019-07-08 13:48:22] Freeing Kinesis Video Stream 1x5dmbug1edd5wmp44ziktdh0qs8e3znwdiap2wdopgecnoqlh4
[INFO ][2019-07-08 13:48:22] 
freeKinesisVideoStream(): Freeing Kinesis Video stream.
```

chehefen, 2019-07-09:
Hi, can you try replace kvssink with fakesink and see if the same thing still happens? The pipeline will look like this:
```
gst-launch-1.0  rtspsrc location=$MY_RTSP_URL short-header=TRUE ! rtph264depay ! h264parse! video/x-h264, format=avc,alignment=au ! fakesink silent=FALSE
```

luisRubiera, 2019-07-11:
Hi @chehefen , today i did the test, i dont know what should be the result, im doing :
```
docker run -e AWS_DEFAULT_REGION=eu-central-1 -it kinesis-raspberry   gst-launch-1.0  rtspsrc location=$MY_RTSP_URL short-header=TRUE ! rtph264depay ! h264parse! video/x-h264, format=avc,alignment=au ! fakesink silent=FALSE
```
and the result is :
```

(gst-plugin-scanner:17): GStreamer-WARNING **: Failed to load plugin '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libfl.so': /opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libfl.so: undefined symbol: yylex
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
Progress: (connect) Connecting to rtsp://luis:vivamexico1@192.168.88.177
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
Progress: (request) SETUP stream 0
Progress: (open) Opened Stream
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
Progress: (request) Sent PLAY request
```
it has not stopped since this morning ( more or less 5h of stream).
Note: the docker image used is the same given by amazon with the modification asked by your team, i could try with the official one if you want me to.

MushMal, 2019-07-22:
@luisRubiera sorry for the delay.

We have attempted to reproduce this scenario with and without docker on RPi with no luck so far. Could you please run the same command with GST_DEBUG=4 on the unmodified docker?

docker run -e GST_DEBUG=4 -e AWS_DEFAULT_REGION=$MY_AWS_REGION -it 46150905175.dkr.ecr.us-west-2.amazonaws.com/kinesis-video-producer-sdk-cpp-raspberry-pi:latest   gst-launch-1.0  rtspsrc location=$MY_RTSP_URL short-header=TRUE ! rtph264depay ! h264parse! video/x-h264, format=avc,alignment=au ! kvssink stream-name=$STREAM_NAME storage-size=128 access-key=$MY_ACCESS_KEY secret-key=$MY_SECRET_KEY

Please send the log file to kinesis-video-support@amazon.com so we can take a look and determine what's going on.

MushMal, 2019-12-04:
I am going to resolve this issue due to lack of activity. Please try the latest release

closed 2019-12-04 by MushMal