awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #230: Parent gstreamer process can't determine when the kvssink is in error

ecolgan, 2019-07-10:
We need to understand the state of the kvssink so we can address errors properly and stream relaiably

zhiyua-git, 2019-07-11:
Hi @ecolgan ,

kvssink will return GST_FLOW_ERROR when it hit fatal error and cannot retry, but for retriable error, the internal retry logic will try to resend what is in the buffer and continue streaming.

Can you update to latest SDK version and see if the retry logic works for you?

Thanks,
-ZH

ecolgan, 2019-07-16:
Thanks.  I rebuilt and installed.   Using this build as my only change and after an idle period of about 35 seconds, all of the streams wind up with the error "[ERROR][DefaultCallbackProvider.cpp:318] postWriteCallback(): Failed to submit ACK: {"EventType":"ERROR","ErrorCode":"INVALID_MKV_DATA","ErrorId":4006}" .  After thi
[INVALID_MKV.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3398458/INVALID_MKV.log)
s error, it  won't send any more data.  Restarting the stream will cause another cycle of this error to occur.  This seems like a regression, since I have not seen it before and downgrading the plugin causes it to go away.  I followed advice on https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html, but to no avail.   Again, this  error happens when the pipeline becomes idle.

MushMal, 2019-07-16:
@ecolgan I see the logs are from the end of the streaming session. It seems that you are properly streaming with 0 delay (buffer is empty) with 1.8Mbps and you are receiving proper ACKs from the backend. After this, I see the following line in the log:

[GST_INFO_LOG][2019-07-16 16:40:05][gstminutefilepushsrc.cpp:1178][pid=25908] Dropping Frame -  EXPECTED END OF MOTION EVENT OCCURRED @ segment time 835.052, Sample Time: 1563295196.998, Finish Target: 1563295196.000

After which the putMedia session times out as there are no frames being produced into the stream.

Could you please describe your scenario? Are you streaming a file? Why is the pipeline iddling instead of stopping and re-starting?

ecolgan, 2019-07-16:
Thanks for the quick response.  We send video in response to motion events and the video is file based.  Most of the time the pipeline is idle.  We don't want to stop and restart the process or even the pipeline for many reasons - latency, efficiency, reliability.  This part was working well in 1.7.9

MushMal, 2019-07-16:
@ecolgan, sounds really interesting. However, file based transmission should use the "offline" mode. This is triggered by specifying streamingtype=offline as in:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#running-the-gst-launch-10-command-to-upload-mkv-file-that-contains-both-audio-and-video-in-raspberry-pi-note-that-video-should-be-h264-encoded-and-audio-should-be-aac-encoded

Could you please try the offline mode and attach the full log so we can see what's going on.

ecolgan, 2019-07-16:
We currently use offline mode(streaming-type=2) and we set the file-start-time.  That also works pretty well.
Here is a log:
[UChannel_logs.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3398704/UChannel_logs.log)

MushMal, 2019-07-16:
Could you please specify 

export KVS_DEBUG_DUMP_DATA_FILE_DIR=</path/to/dump/directory>

environment variable while building the producer sources so we can capture the dump and analyze whats wrong with the produced stream?

You can email the results to kinesis-video-support@amazon.com <kinesis-video-support@amazon.com>

MushMal, 2019-07-30:
Resolving this issue. Please re-open if this issue happens with Release v 2.0.1

closed 2019-07-30 by MushMal

indoorrobot, 2019-12-26:
Hi

I am suffering from the same issue, however i am constantly streaming (should be realtime), but after a few seconds / minutes I am getting the same errror.

See logs:

```
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - defaultDroppedFrameReportCallback(): Reported droppedFrame callback for stream handle 27000953. Dropped frame timecode in 100ns: 10950330000
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - droppedFrameReportHandler invoked
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] WARN - Reported droppedFrame callback for stream handle 27000953. Dropped frame timecode in 100ns: 10950330000
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback for stream handle 27000953. Current buffer duration in 100ns: 927670000
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - streamLatencyPressureHandler invoked
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - Reported streamLatencyPressure callback for stream handle 27000953. Current buffer duration in 100ns: 927670000
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] WARN - viewItemRemoved(): Reporting a dropped frame/fragment.
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - defaultDroppedFrameReportCallback(): Reported droppedFrame callback for stream handle 27000953. Dropped frame timecode in 100ns: 10950660000
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - droppedFrameReportHandler invoked
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] WARN - Reported droppedFrame callback for stream handle 27000953. Dropped frame timecode in 100ns: 10950660000
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback for stream handle 27000953. Current buffer duration in 100ns: 927710000
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - streamLatencyPressureHandler invoked
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - Reported streamLatencyPressure callback for stream handle 27000953. Current buffer duration in 100ns: 927710000
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] WARN - viewItemRemoved(): Reporting a dropped frame/fragment.
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - defaultDroppedFrameReportCallback(): Reported droppedFrame callback for stream handle 27000953. Dropped frame timecode in 100ns: 10950990000
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - droppedFrameReportHandler invoked
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] WARN - Reported droppedFrame callback for stream handle 27000953. Dropped frame timecode in 100ns: 10950990000
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback for stream handle 27000953. Current buffer duration in 100ns: 927700000
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - streamLatencyPressureHandler invoked
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - Reported streamLatencyPressure callback for stream handle 27000953. Current buffer duration in 100ns: 927700000
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] WARN - viewItemRemoved(): Reporting a dropped frame/fragment.
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - defaultDroppedFrameReportCallback(): Reported droppedFrame callback for stream handle 27000953. Dropped frame timecode in 100ns: 10951330000
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - droppedFrameReportHandler invoked
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] WARN - Reported droppedFrame callback for stream handle 27000953. Dropped frame timecode in 100ns: 10951330000
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback for stream handle 27000953. Current buffer duration in 100ns: 927700000
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:459913431840512] DEBUG - Reported streamLatencyPressure callback for stream handle 10741817. Current buffer duration in 100ns: 924580000
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139913431840512] DEBUG - postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 205
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139913320244992] DEBUG - postWriteCallback(): Curl post body write function for stream with handle: Device05_rc and upload handle: 205 returned: {"EventType":"ERROR","ErrorCode":"INVALID_MKV_DATA","ErrorId":4006}
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139913320244992] DEBUG - defaultStreamErrorReportCallback(): Reported streamError callback for stream handle 10741817. Upload handle 205. Fragment timecode in 100ns: 10620330000. Error status: 0x52000062
[... cut ...]
Dec 26 10:50:45 Device05 robot[1875]: 2019-12-26 10:50:45 [139807149221632] DEBUG - Reported streamLatencyPressure callback for stream handle 27000953. Current buffer duration in 100ns: 927700000

```
The way I am running the stream is:
gst-launch-1.0 -v v4l2src device=/dev/video3 ! h264parse ! video/x-h264,stream-frmat=avc,alignment=au,width=1920,height=1080,framerate=30/1 ! kvssink name=sink stream-name=${Device_ID}_fc access-key=$AWS_ACCESS_KEY_ID secret-key=$AWS_SECRET_ACCESS_KEY

What am I missing?

ecolgan, 2019-12-26:
@indoorrobot - My reading of your log tells me the fragment of video you are streaming is more than ten seconds in length.  Verify that the time between IFrames(e.g. GOP length) is less than ten seconds.

indoorrobot, 2019-12-26:
@ecolgan thanks for assisting with this... According to the aws params documentation the default fragment-duration is 2 secs, even though tried to set it to more or less did not change much... How can this be controlled?
What is GOP length?

ecolgan, 2019-12-26:
The GOP length is amount of time that elapses between IFrames.  It is almost always a configurable value for encoders and therefore streaming cameras.  The needs to be controlled by your encoder - for you, whatever is producing /dev/video3 for v4l2src.  I haven't worked with KVS for a few months, so I don't recall the documentation. I do recall it won't make a fragment shorter than one GOP.

zhiyua-git, 2019-12-26:
@indoorrobot Please also check if your network bandwidth is enough, your log shows that you have cumulated a large buffer of data that cannot be sent. Could be due to network bandwidth issue.