awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #232: Network Camera

NichelleBot, 2019-07-17:
I have a network camera that does not have an rstp address, but a website endpoint attached to it. When I try to run the demo, I put in the web address instead of the rtsp address. This always results in my webcam being turned on.

In short, I guess my question is...does this repo work with networked cameras with IP addresses?

If so, how have people gone about doing this?

MushMal, 2019-07-17:
@NichelleBot it would depend on the type of camera. Could you please include the make/model so we can try to look it up?

I assume you can't modify the camera firmware...?

NichelleBot, 2019-07-17:
Correct, I cannot. The camera is an AXIS P5635-E Mk II PTZ Dome Network Camera.

unicornss, 2019-07-17:
Hi @NichelleBot,
Typical RTSP URL for Axis IP camera is as follows: 
rtsp://**username**:**password**@**ipaddress**/axis-media/media.amp
You can fill in the IP address in the following along with the user name and password. 

You  should be able to play it in the VLC as well.

NichelleBot, 2019-07-17:
So if this is a public streaming camera but I don't know the username and password to the specific device, there's no way around it?

Also, this may be a dumb question, but what is the VLC?

I was thinking that a way around this is to take a screen grab of the screen and then use that as the stream I read from. Do you think that would work?

NichelleBot, 2019-07-17:
I just tried out the script again by adding everything BUT the username and password, but it still turns on my webcam...

Why does my webcam still turn on if I specify a stream and rstp?

MushMal, 2019-07-17:
@NichelleBot are you running a GStreamer gstlaunch command? We have a sample command line in the readme

gst-launch-1.0 rtspsrc location=rtsp://YourCameraRtspUrl short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! h264parse ! kvssink stream-name=YourStreamName storage-size=128 access-key="YourAccessKey" secret-key="YourSecretKey"

The rtsp uri encodes the used name and password as @unicornss mentioned. 

Please try that.

NichelleBot, 2019-07-18:
I have tried that but am getting the error `WARNING: erroneous pipeline: no element "kvssink"` I looked at the readme and see that it is important to install `libasound2-dev`, but there doesn't seem to be a mac-friendly version? 

I have looked around online and have installed `libsndfile` and ran `./install_script` after deleting the necessary files. That said, I am still getting the same error.

MushMal, 2019-07-18:
@NichelleBot the error is about the gst-launch not finding the kvssink element which is built after running the install script. You might be running a different installation of the GStreamer or the paths are not set.

Can you please follow the instructions highlighted here to set the path correctly?
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-macos.md#setting-the-environment-variables-for-library-path

NichelleBot, 2019-07-18:
Yay! It works. Thanks @MushMal  and @unicornss for all of you're help. I'm gonna close the issue.

closed 2019-07-18 by NichelleBot