awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #233: The camera integration with AWS

Hughxuhui, 2019-07-19:
Hi AWS team,

We met the following issues when we tried to integarte your SDK:

[WARN ][2019-07-15 11:31:20] HTTP Error 403: Response: {"message":"The security token included in the request is invalid."}
 
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=/20190715/us-west-2/kinesisvideo/aws4_request, SignedHeaders=content-type;host;user-agent;x-amz-date, Signature=b67ddd2a8fa5b2e2d1c0ae23c00f22536002905f0657d567df54e68f8d317562
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS/1.5.0 GCC/6.3.0 Linux/4.9.37 armv7l
    X-Amz-Date: 20190715T113115Z
[DEBUG][2019-07-15 11:31:20] describeStream response: {"message":"The security token included in the request is invalid."}
[INFO ][2019-07-15 11:31:20] Describe stream did not find the stream MyKvsCameraTest in Kinesis Video (stream will be created)

unicornss, 2019-07-19:
Hi @Hughxuhui 

> {"message":"The security token included in the request is invalid."}

Please check the following required parameters (credentials) are provided to run the sample application.

$ export AWS_ACCESS_KEY_ID=YourAccessKeyId
$ export AWS_SECRET_ACCESS_KEY=YourSecretAccessKey

and for the region other than us-west-2

$export AWS_DEFAULT_REGION=YourAwsRegion

Reference: [Installation instructions](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/424580b800087f05f80000d30b43b61e5a964c29/install-instructions-linux.md#how-to-run-sample-applications-for-sending-media-to-kvs-using-gstreamer)

MushMal, 2019-07-19:
@Hughxuhui please let us know if you have further issues. Please re-open the issue if you still have any problems with the particular issue or open a new one.

closed 2019-07-19 by MushMal

zhiyua-git, 2019-07-19:
The log shows your access key is not set. Can you debug and see why it is missing?

Hughxuhui, 2019-07-25:
Hi guys,

Thanks for your kind support.
Our RD found the issue is:
No matter use AWS demo or run the demo into the camera, the camera will crash by accident after AWS SDK send stream for a while.
Please check the following yellow part, would you please help to check? Thanks so much.

[TRACE][2019-07-24 12:21:25] postHeaderReadFunc (curl callback) invoked
[DEBUG][2019-07-24 12:21:25] Curl post header write function returned:Content-Type: application/json

[TRACE][2019-07-24 12:21:25] postHeaderReadFunc (curl callback) invoked
[DEBUG][2019-07-24 12:21:25] Curl post header write function returned:Transfer-Encoding: chunked

[TRACE][2019-07-24 12:21:25] postHeaderReadFunc (curl callback) invoked
[DEBUG][2019-07-24 12:21:25] Curl post header write function returned:

[TRACE][2019-07-24 12:21:25] postBodyStreamingReadFunc (curl callback) invoked
[TRACE][2019-07-24 12:21:25] Awaiting data...
[TRACE][2019-07-24 12:21:25] Available bytes to read: 25284669 buffer size: 65524 written bytes: 65524 for client stream handle: 13 current stream handle: 13
[DEBUG][2019-07-24 12:21:25] Wrote 65524 bytes to Kinesis Video. Upload stream handle: 13
[TRACE][2019-07-24 12:21:25] postBodyStreamingReadFunc (curl callback) invoked
[TRACE][2019-07-24 12:21:25] Awaiting data...
[TRACE][2019-07-24 12:21:25] Available bytes to read: 25219145 buffer size: 65524 written bytes: 65524 for client stream handle: 13 current stream handle: 13
[DEBUG][2019-07-24 12:21:25] Wrote 65524 bytes to Kinesis Video. Upload stream handle: 13
[TRACE][2019-07-24 12:21:25] postBodyStreamingReadFunc (curl callback) invoked
[TRACE][2019-07-24 12:21:25] Awaiting data...
[TRACE][2019-07-24 12:21:25] Available bytes to read: 25153621 buffer size: 65524 written bytes: 65524 for client stream handle: 13 current stream handle: 13
[DEBUG][2019-07-24 12:21:25] Wrote 65524 bytes to Kinesis Video. Upload stream handle: 13
[TRACE][2019-07-24 12:21:25] postBodyStreamingReadFunc (curl callback) invoked
[TRACE][2019-07-24 12:21:25] Awaiting data...
[TRACE][2019-07-24 12:21:25] Available bytes to read: 25088097 buffer size: 65524 written bytes: 65524 for client stream handle: 13 current stream handle: 13
[DEBUG][2019-07-24 12:21:25] Wrote 65524 bytes to Kinesis Video. Upload stream handle: 13
[TRACE][]TRACE][2019-07-24 12:21:25] 9-07-24 12:21:25] ] streamLatencyPressureHandler invoked
*** Error in `/usr/bin/sonia': double free or corruption (fasttop): 0x02f0bb88 ***
[TRACE][2019-07-24 12:21:25] streamDataAvailableHandler invoked for stream: MyKvsCameraTest and stream upload handle: 13
[TRACE][2019-07-24 12:21:25] Note data received: duration(100ns): 480900000 bytes: 25021691 for stream handle: 13
Time : Wed Jul 24 12:21:25 2019

=========================== TRACE START ===================================
Tid:1706, Exception type : SIGABRT
[DEBUG][2019-07-24 12:21:25] streamLatencyPressureHandler invoked
[TRACE][2019-07-24 12:21:25] streamDataAvailableHandler invoked for stream: MyKvsCameraTest and stream upload handle: 13
[TRACE][2019-07-24 12:21:25] Note data received: duration(100ns): 481300000 bytes: 25039555 for stream handle: 13
[DEBUG][2019-07-24 12:21:25] streamLatencyPressureHandler invoked
[TRACE][2019-07-24 12:21:25] streamDataAvailableHandler invoked for stream: MyKvsCameraTest and stream upload handle: 13
[TRACE][2019-07-24 12:21:25] Note data received: duration(100ns): 481700000 bytes: 25051167 for stream handle: 13
[DEBUG][2019-07-24 12:21:25] streamLatencyPressureHandler invoked
[TRACE][2019-07-24 12:21:25] streamDataAvailableHandler invoked for stream: MyKvsCameraTest and stream upload handle: 13
[TRACE][2019-07-24 12:21:25] Note data received: duration(100ns): 482100000 bytes: 25068305 for stream handle: 13
[DEBUG][2019-07-24 12:21:25] streamLatencyPressureHandler invoked
[TRACE][2019-07-24 12:21:25] streamDataAvailableHandler invoked for stream: MyKvsCameraTest and stream upload handle: 13
[TRACE][2019-07-24 12:21:25] Note data received: duration(100ns): 482500000 bytes: 25087633 for stream handle: 13
[DEBUG][2019-07-24 12:21:25] streamLatencyPressureHandler invoked
[TRACE][2019-07-24 12:21:25] streamDataAvailableHandler invoked for stream: MyKvsCameraTest and stream upload handle: 13
[TRACE][2019-07-24 12:21:25] Note data received: duration(100ns): 482900000 bytes: 25099217 for stream handle: 13
PC:[0x005f3190](Failed to locate address)=========================== TRACE   END ===================================
=========================== New TRACE   START ===================================
=========================== New TRACE   END =====================================
Aborted

Hughxuhui, 2019-07-25:
Please pay attention to the " Error in `/usr/bin/sonia': double free or corruption (fasttop): 0x02f0bb88 ***"

MushMal, 2019-07-25:
@Hughxuhui this is an unrelated problem to the original issue reported. Could you please open a new issue so we can track this properly.

Could you please provide some more information about the platform/OS, version of the SDK you are using and most importantly whether you have provided any latency pressure callback function at all.

Hughxuhui, 2019-08-08:
Dear Mushmal,

Thanks, but I think this is related, the camera will crash even we used the following demo:
http://wikisend.com/download/186082/kinesis_video_native_sample_app.cpp

We already updated the newest SDK that Zhiyuan help to share at the Github, please help to check.
The OS info is following:
1.	platform : arm-himix200v002-linux 
2.	os: linux 4.9.37 armv7l GNU/Linux
3.	SDK version: 2018/10/30 从master 分支上下载的

MushMal, 2019-08-08:
@Hughxuhui could you please open a separate issue to track this.

Please include the details about the platform including whether the platform supports unaligned memory access. Please use the min install script to clean compile everything and run the sample from the SDK itself.