awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #234: Segmentation fault on mingw with amazon-kinesis-video-streams-producer-sdk-cpp

VenkatasivaTF, 2019-07-22:
Segmentation Fault on running the below commands on mingw:
Proper input is provided for access-key/secret-key and stream name

./kinesis_video_gstreamer_audio_video_sample_app <my-stream-name>

gst-launch-1.0 -v ksvideosrc ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=512 tune=zerolatency ! video/x-h264,profile=baseline,stream-format=avc,alignment=au ! kvssink name=sink stream-name=amar-sample-video-stream access-key=access-key secret-key=secret-key wasapisrc device="\{0.0.1.00000000\}.\{f6d4a51c-e8eb-4db9-b85f-ecdb37cbcb63\}" ! audioconvert ! avenc_aac ! queue ! sink.

MushMal, 2019-07-23:
@VenkatasivaTF could you please include the crash stack which would help understanding what's wrong? Some of the pipeline elements can be unstable. We also observed crashes if the parameter names are misspelled. Please run with GST_DEBUG=4 before launching. Please collect the logs and the stack dump and send it to 'kinesis-video-support@amazon.com'.

Thanks in advance.

claw68, 2019-08-14:
@VenkatasivaTF check out this [solution](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/221#issuecomment-520076310). I have a similar problem.

unicornss, 2019-10-31:
Closing the issue for now. Please reopen or send email to kinesis-video-support@amazon.com as @MushMal  has mentioned.

closed 2019-10-31 by unicornss