awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #235: "Green Screen" in Kinesis video console while sending RTSP video stream via Raspberry PI

Aashikms, 2019-07-23:
I have tried to send RTSP camera stream to Kinesis video stream via C++ producer sdk built on Raspberry Pi. I followed the step as mentioned in this link https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md

I ran the below command in Raspberry pi
```$ gst-launch-1.0 -v rtspsrc location=rtsp://YourCameraRtspUrl short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! h264parse ! kvssink stream-name=YourStreamName storage-size=128```

Above command is streaming the data to Kenisis video stream but i am only getting green screen in the console. Attached the screenshot. 
Note: I have checked the RTSP url and it is working fine. Also i have tried streaming with USB camera on Raspberry Pi and its streams perfectly.
[image]

MushMal, 2019-07-24:
@Aashikms in most cases, a green screen is an artifact of a decoder failing to parse the elementary stream.

Could you please specify KVS_DEBUG_DUMP_DATA_FILE_DIR variable to point to a temporary directory to store the media files and attempt to reproduce this?

'export KVS_DEBUG_DUMP_DATA_FILE_DIR=/tmp/mediadump/' (or something like this)

Also, please enable GStreamer verbose logging by specifying GST_DEBUG variable like so

'export GST_DEBUG=4'

Please collect a few minutes worth of media. Attempt to play it back using VLC or other media program.

Please send the logs to kinesis-video-support@amazon.com

Aashikms, 2019-07-24:
I have tried above but i am not finding any media file in the folder location i have exported in KVS_DEBUG_DUMP_DATA_FILE_DIR variable.
Below are the command i ran
```$ export KVS_DEBUG_DUMP_DATA_FILE_DIR=/tmp/mediadump/
$ export GST_DEBUG=4
$ gst-launch-1.0 -v rtspsrc location=rtsp://YourCameraRtspUrl short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! h264parse ! kvssink stream-name=YourStreamName storage-size=512```

Aashikms, 2019-07-24:
However i have sent the log file to kinesis-video-support@amazon.com.

MushMal, 2019-07-24:
Replied via email:

I have looked through the file but can’t spot any issues at all – you are streaming just fine to a stream called “HarshaUSBCam” with 0 delay, ~900Kbps. It’s a little strange that you have 5 FPS frame rate but doesn’t in itself constitute anything.
 
Could you please let us know the make/model of the camera?

Aashikms, 2019-07-25:
The camera details are as below:
Make: Acti
Model: E53

Now the problem is resolved as there was some extension missing in the RTSP url
As per Acti Documents to stream RTSP URL below format was recommended
```rtsp://<username:pw@ip:port>```
I used the above url and was able to stream the RTSP video feed in VLC and Apple Quicktime player. Hence i used the same url in the Gstream command.

This Model has 2 video channels video1 and video2. So after referring other models in acti, i experimented by adding the respective channel extension into my RTSP url and i was able to stream to Kinesis video stream.
```rtsp://<username:pw@ip:port/videox> (where x is channel no)```

Because of the above difference to stream in video player and Gstream, i guess i was getting the green screen. I am closing this issue and thanks for the support.

closed 2019-07-25 by Aashikms

MushMal, 2019-07-25:
Just wow! Thanks for letting us know :)