awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #236: Kinesis Video Streaming I want to do in Raspberry Pi and Jetson Nano but nothing is getting in the error

ashishbharadwaj100, 2019-07-24:
root@ashish:/home/ashish/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build# ./kinesis_video_gstreamer_sample_app my_streaming
[INFO ][2019-07-24 04:43:43] Using region: us-east-1
[INFO ][2019-07-24 04:43:43] Using aws credentials for Kinesis Video Streams
[INFO ][2019-07-24 04:43:43] No session token was detected.
[INFO ][2019-07-24 04:43:43] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2019-07-24 04:43:43] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ][2019-07-24 04:43:43] heapInitialize(): Creating AIV heap.
[INFO ][2019-07-24 04:43:43] heapInitialize(): Heap is initialized OK
[DEBUG][2019-07-24 04:43:43] stepStateMachine(): PIC Client State Machine - Current state: 0x1, Next state: 0x2
[DEBUG][2019-07-24 04:43:43] getSecurityTokenHandler invoked
[DEBUG][2019-07-24 04:43:43] Refreshing credentials. Force refreshing: 0 Now time is: 1563943423364141403 Expiration: 0
[INFO ][2019-07-24 04:43:43] New credentials expiration is 1563945823
[DEBUG][2019-07-24 04:43:43] stepStateMachine(): PIC Client State Machine - Current state: 0x2, Next state: 0x10
[INFO ][2019-07-24 04:43:43] createDeviceResultEvent(): Create device result event.
[DEBUG][2019-07-24 04:43:43] stepStateMachine(): PIC Client State Machine - Current state: 0x10, Next state: 0x40
[DEBUG][2019-07-24 04:43:43] clientReadyHandler invoked
[DEBUG][2019-07-24 04:43:43] Client is ready
[INFO ][2019-07-24 04:43:43] Creating Kinesis Video Stream my_streaming
[INFO ][2019-07-24 04:43:43] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2019-07-24 04:43:43] stepStateMachine(): PIC Stream State Machine - Current state: 0x1, Next state: 0x2
[DEBUG][2019-07-24 04:43:43] stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
[INFO ][2019-07-24 04:43:44] describeStreamResultEvent(): Describe stream result event.
[DEBUG][2019-07-24 04:43:44] stepStateMachine(): PIC Stream State Machine - Current state: 0x2, Next state: 0x20
[DEBUG][2019-07-24 04:43:44] stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
[INFO ][2019-07-24 04:43:46] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2019-07-24 04:43:46] stepStateMachine(): PIC Stream State Machine - Current state: 0x20, Next state: 0x10
[DEBUG][2019-07-24 04:43:46] stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
[DEBUG][2019-07-24 04:43:46] getStreamingTokenHandler invoked
[DEBUG][2019-07-24 04:43:46] Refreshing credentials. Force refreshing: 1 Now time is: 1563943426466739880 Expiration: 1563945823
[INFO ][2019-07-24 04:43:46] New credentials expiration is 1563945826
[INFO ][2019-07-24 04:43:46] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2019-07-24 04:43:46] stepStateMachine(): PIC Stream State Machine - Current state: 0x10, Next state: 0x40
[DEBUG][2019-07-24 04:43:46] defaultStreamReadyCallback(): Reported streamReady callback for stream handle 366778919817
[DEBUG][2019-07-24 04:43:46] streamReadyHandler invoked
[DEBUG][2019-07-24 04:43:46] Stream is ready
0:00:00.001084224 12143   0x5565b7fc00 INFO                GST_INIT gst.c:507:init_pre: Initializing GStreamer Core Library version 1.12.3
0:00:00.002445740 12143   0x5565b7fc00 INFO                GST_INIT gst.c:508:init_pre: Using library installed in /home/ashish/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib
0:00:00.003098607 12143   0x5565b7fc00 INFO                GST_INIT gst.c:528:init_pre: Linux ashish 4.9.140-tegra #1 SMP PREEMPT Wed Mar 13 00:32:22 PDT 2019 aarch64
0:00:00.006899664 12143   0x5565b7fc00 INFO                GST_INIT gstmessage.c:127:_priv_gst_message_initialize: init messages
0:00:00.011365204 12143   0x5565b7fc00 INFO                GST_INIT gstcontext.c:84:_priv_gst_context_initialize: init contexts
0:00:00.012355416 12143   0x5565b7fc00 INFO      GST_PLUGIN_LOADING gstplugin.c:317:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.012900471 12143   0x5565b7fc00 INFO      GST_PLUGIN_LOADING gstplugin.c:225:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.012966512 12143   0x5565b7fc00 INFO      GST_PLUGIN_LOADING gstplugin.c:227:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.013074065 12143   0x5565b7fc00 INFO            GST_REGISTRY gstregistry.c:1721:ensure_current_registry: reading registry cache: /root/.cache/gstreamer-1.0/registry.aarch64.bin
0:00:00.060120770 12143   0x5565b7fc00 INFO            GST_REGISTRY gstregistrybinary.c:621:priv_gst_registry_binary_read_cache: loaded /root/.cache/gstreamer-1.0/registry.aarch64.bin in 0.046912 seconds
0:00:00.060356292 12143   0x5565b7fc00 INFO            GST_REGISTRY gstregistry.c:1577:scan_and_update_registry: Validating plugins from registry cache: /root/.cache/gstreamer-1.0/registry.aarch64.bin
0:00:00.067390019 12143   0x5565b7fc00 INFO            GST_REGISTRY gstregistry.c:1679:scan_and_update_registry: Registry cache has not changed
0:00:00.067425280 12143   0x5565b7fc00 INFO            GST_REGISTRY gstregistry.c:1756:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.067452311 12143   0x5565b7fc00 INFO                GST_INIT gst.c:727:init_post: GLib runtime version: 2.54.2
0:00:00.067478874 12143   0x5565b7fc00 INFO                GST_INIT gst.c:729:init_post: GLib headers version: 2.54.2
0:00:00.067503770 12143   0x5565b7fc00 INFO                GST_INIT gst.c:730:init_post: initialized GStreamer successfully
[INFO ][2019-07-24 04:43:46] Streaming from live source
0:00:00.067680854 12143   0x5565b7fc00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "pipeline" named "live-kinesis-pipeline"
[DEBUG][2019-07-24 04:43:46] Streaming with live source and width: 0, height: 0, fps: 25, bitrateInKBPS512
0:00:00.069528049 12143   0x5565b7fc00 INFO      GST_PLUGIN_LOADING gstplugin.c:843:_priv_gst_plugin_load_file_for_registry: plugin "/home/ashish/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/gstreamer-1.0/libgstcoreelements.so" loaded
0:00:00.069575393 12143   0x5565b7fc00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "capsfilter" named "source_filter"
0:00:00.069877946 12143   0x5565b7fc00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@0x5565c4a100> adding pad 'sink'
0:00:00.069940238 12143   0x5565b7fc00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@0x5565c4a100> adding pad 'src'
0:00:00.070000655 12143   0x5565b7fc00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "capsfilter" named "encoder_filter"
0:00:00.070052634 12143   0x5565b7fc00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@0x5565c4a440> adding pad 'sink'
0:00:00.070097530 12143   0x5565b7fc00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@0x5565c4a440> adding pad 'src'
0:00:00.070623730 12143   0x5565b7fc00 INFO      GST_PLUGIN_LOADING gstplugin.c:843:_priv_gst_plugin_load_file_for_registry: plugin "/home/ashish/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/gstreamer-1.0/libgstapp.so" loaded
0:00:00.070658939 12143   0x5565b7fc00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "appsink" named "appsink"
0:00:00.070936440 12143   0x5565b7fc00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseSink@0x5565d8b390> adding pad 'sink'
0:00:00.073797910 12143   0x5565b7fc00 INFO      GST_PLUGIN_LOADING gstplugin.c:843:_priv_gst_plugin_load_file_for_registry: plugin "/home/ashish/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/gstreamer-1.0/libgstvideoparsersbad.so" loaded
0:00:00.073855098 12143   0x5565b7fc00 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:359:gst_element_factory_create: creating element "h264parse" named "h264parse"
0:00:00.074151714 12143   0x5565b7fc00 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseParse@0x5565d8f550> adding pad 'sink'
[... cut ...]
[DEBUG][2019-07-24 04:43:46] streamClosedHandler invoked for upload handle: 18446744073709551615
[INFO ][2019-07-24 04:43:46] Freeing Kinesis Video Stream my_streaming
[INFO ][2019-07-24 04:43:46] freeKinesisVideoStream(): Freeing Kinesis Video stream.
[DEBUG][2019-07-24 04:43:47] defaultStreamShutdownCallback(): Reported streamShutdown callback for stream handle 366778919817
[INFO ][2019-07-24 04:43:47] Freeing Kinesis Video Stream my_video_streaming
[INFO ][2019-07-24 04:43:47] freeKinesisVideoClient(): Freeing Kinesis Video Client

MushMal, 2019-07-24:
The sample application should have loaded 'omxh264enc' encoder on RPi. Looking at the log it fails to load it and also fails fallback to the software encoder.

Could you please confirm you are using the following steps correctly? Check out https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md for the troubleshooting. Specifically, 

If you are using Raspberry pi, you can also install the gstreamer1.0-omx package to get the omxh264enc hardware encoder

sudo apt-get install gstreamer1.0-omx

unicornss, 2019-07-31:
Hi @ashishbharadwaj100 ,

Can you please run the [gst-device-monitor-1.0](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#discovering-audio-and-video-devices-available-in-your-system) and send us the output?

That should help us understand what devices are available/supported in your device environment...

Thanks

closed 2019-08-12 by ashishbharadwaj100