awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #237: jsmn lib not found.. Installing

pankeshpatel, 2019-07-24:
I have been trying to run  `./mini-install-script` of `Kinesis-video-native-build` folder. I am getting a message `jsmn lib not found. Installing`.

MushMal, 2019-07-24:
min-install-script is detecting that you don't have jsmn installed so it tries to pull down and install it. Does the installation result in an error?

pankeshpatel, 2019-07-24:
No, there is no error in installation. 

I  just realize that the raspberry PI's internet connection was lost some how. Therefore, the `./mini-install-script` was not able to pull the repository from github. 

Won't be nice, if we could write few lines in script that check the the internet connection and print a message, if the internet is not available.

MushMal, 2019-07-24:
Perhaps adding some check at the start of the script makes sense. Though, checking for the general internet availability is a hard thing to crack. The "pulling down" step itself can be the indicator of the network availability.

closed 2019-07-24 by MushMal