awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #239: Memory leak for multiple start stop of kinesis video stream

pateltithi, 2019-07-30:
[KinesisVideoStreamer.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3448478/KinesisVideoStreamer.txt)

Attached is my class definition. In this , as per my use  case I need to call StartStreaming/StopStreaming multiple times. 

What I see is memory leak of n* 368 bytes where 'n' is the number of start/stops. valgrind complains of the leak to be in gstreamer.

Can you please guide as to what I have done wrong ?

unicornss, 2019-07-30:
Hi @pateltithi 

You can use Gstreamer debug tools or tracing by setting the environment variables f.

`GST_DEBUG="GST_TRACER:7" GST_TRACERS="leaks(GstTaskPool,GstCaps)`

Also refer to the https://github.com/GStreamer/common/blob/master/gst.supp

unicornss, 2019-07-30:
Hi @pateltithi 

There is a g_free missing for cpd in the video streaming sample. This issue in the video streaming sample will be fixed in the upcoming release.    For now, can you also add [g_free(cpd);](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_audio_video_sample_app.cpp#L384)
 

```
  gchar *cpd = gst_value_serialize(gstStreamFormat);
  data.kinesis_video_stream->start(std::string(cpd));
  g_free(cpd);
```

Please let us know if that fixes the  issue.  Thanks

pateltithi, 2019-07-30:
368 bytes in 1 blocks are definitely lost in loss record 92 of 133
==5557==    at 0x4845178: malloc (vg_replace_malloc.c:309)
==5557==    by 0x11DAF58B: ??? (in /usr/lib/aarch64-linux-gnu/tegra/libnvomx.so)
==5557==    by 0x11DAB19B: ??? (in /usr/lib/aarch64-linux-gnu/tegra/libnvomx.so)
==5557==    by 0x1135448B: ??? (in /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstomx.so)
==5557==    by 0x113713E3: ??? (in /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstomx.so)
==5557==    by 0x111DB78F: gst_video_encoder_setcaps (gstvideoencoder.c:611)
==5557==    by 0x111DB78F: gst_video_encoder_sink_event_default (gstvideoencoder.c:963)
==5557==    by 0x48F1363: ??? (in /usr/lib/aarch64-linux-gnu/libgstreamer-1.0.so.0.803.0)
This is the recurring leak.

g_free (cpd) reduces the fixed leak by 50 bytes, but doen't help for these 368 bytes.

unicornss, 2019-07-30:
Is this happening on nvidia encoder `libnvomx.so`?

pateltithi, 2019-07-30:
I'm runnign the code on Nvidia Tx2.

unicornss, 2019-07-31:
Can you please run `export GST_DEBUG=GST_REFCOUNTING:5` and start your application? This should help you to see which objects are not freed at the end of each run.

For example, if you run the application _two_ times and you notice one object is not freed at the end of the run (count shows _2_ ), then that could be a potential root cause for the memory leak.

unicornss, 2019-08-08:
Hi @pateltithi , were you able to check the debug output with `GST_REFCOUNTING:` ?

pateltithi, 2019-08-08:
I was able to check but It did not help me. I still have the recurring leak.

MushMal, 2019-08-08:
@pateltithi what version of the SDK are you using? The latest version does have some cumulative fixes. Could you include the valgrind full report

pateltithi, 2019-08-12:
Attaching logs for Jun 12 version of SDK as well as 30 July verson. The 368 byte leak for 1 start/stop and 736 (368*2) bytes leak for 2 start/stops is present for both versions.

Any suggestions here will be of great help, Thanks[
[valg_1_startstop_30Jul.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3493981/valg_1_startstop_30Jul.txt)
[valg_1_startstop.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3493982/valg_1_startstop.txt)
[valg_2_startstop.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3493983/valg_2_startstop.txt)
[valg_2_startstop_30July.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3493984/valg_2_startstop_30July.txt)

](url)

unicornss, 2019-08-12:
Hi @pateltithi ,

Please check if you are running the [valgrind](http://valgrind.org/docs/manual/mc-manual.html) with the [GStreamer suppression files](https://github.com/GStreamer/common/blob/master/gst.supp).

Given that the log shows  nvidia h264 - were you able to check if similar issue has been reported or any uninit steps are required for that element in  [nvdia dev forums](https://devtalk.nvidia.com/default/board/188/jetson-tx2/) or in [GStreamer devel forums.](https://gstreamer.freedesktop.org/lists/)
```
==4125== 368 bytes in 1 blocks are definitely lost in loss record 93 of 134
==4125==    at 0x4845178: malloc (vg_replace_malloc.c:309)
==4125==    by 0x11DAF58B: ??? (in /usr/lib/aarch64-linux-gnu/tegra/libnvomx.so)
==4125==    by 0x11DAB19B: ??? (in /usr/lib/aarch64-linux-gnu/tegra/libnvomx.so)
==4125==    by 0x1135448B: ??? (in /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstomx.so)
==4125==    by 0x113713E3: ??? (in /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstomx.so)
==4125==    by 0x111DB78F: gst_video_encoder_setcaps (gstvideoencoder.c:611)
==4125==    by 0x111DB78F: gst_video_encoder_sink_event_default (gstvideoencoder.c:963)
==4125==    by 0x48F1363: ??? (in /usr/lib/aarch64-linux-gnu/libgstreamer-1.0.so.0.803.0)
==4125== 
==4125== 368 bytes in 1 blocks are definitely lost in loss record 94 of 134
==4125==    at 0x4845178: malloc (vg_replace_malloc.c:309)
==4125==    by 0x11DAF58B: ??? (in /usr/lib/aarch64-linux-gnu/tegra/libnvomx.so)
==4125==    by 0x11DAB19B: ??? (in /usr/lib/aarch64-linux-gnu/tegra/libnvomx.so)
==4125==    by 0x1135448B: ??? (in /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstomx.so)
==4125==    by 0x113715D7: ??? (in /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstomx.so)
==4125==    by 0x111DB78F: gst_video_encoder_setcaps (gstvideoencoder.c:611)
==4125==    by 0x111DB78F: gst_video_encoder_sink_event_default (gstvideoencoder.c:963)
==4125==    by 0x48F1363: ??? (in /usr/lib/aarch64-linux-gnu/libgstreamer-1.0.so.0.803.0)
```

unicornss, 2019-10-31:
Closing the issue for now. Please reopen or send email to kinesis-video-support@amazon.com along with your use case if it requires support.

closed 2019-10-31 by unicornss